openbmc/linux #11: I2C errors in console

nkskjames, 2015-10-30:
oot@palmetto:~# i2cdetect 6
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-6.
I will probe address range 0x03-0x77.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 -- -- -- -- -- -- -- -- -- -- -- -- 
10: 10 systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- -- -- -- -- -- -- -- systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- -- -- -- -- -- systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- 
20: systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
20 -- -- -- -- -- -- -- -- systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- -- -- -- 2d systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
2e systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- 
30: -- -- -- -- -- -- -- systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- -- -- -- -- -- systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- -- -- 
40: -- -- -- -- systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- -- -- -- 48 systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- -- -- -- -- -- -- 
50: -- systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- -- -- -- systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
55 -- -- -- -- -- systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- -- -- -- -- 
60: systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- -- -- -- -- -- -- systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- -- -- -- -- -- systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- -- -- 
70: -- -- -- systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.
-- -- -- -- --                 systemd-journald[402]: /dev/kmsg buffer overrun, some messages lost.

shenki, 2015-10-31:
These look like systemd-journald errors, not i2c errors

williamspatrick, 2015-10-31:
It seems like #12 is the culprit.  If we are flooding out i2c logs to the console then systemd might not be able to keep up.  Seems like two different symptoms of the same problem.

jk-ozlabs, 2015-11-04:
Should be fixed by 3cd273e, but there seems to be other issues with i2cdetect.

closed 2015-11-24 by shenki