openbmc/linux #12: I2C is flooding journald log

nkskjames, 2015-10-30:
only thing in journald in 30000 lines is i2c stuff

shenki, 2015-10-31:
What does the journald log look like?

What is your kernel command line?

shenki, 2015-11-01:
```
Booting Linux on physical CPU 0x0
Linux version 4.2.0-openbmc-20151028-1 (causten@causten-VirtualBox) (gcc version 4.9.3 (GCC) ) #1 Sun Nov 1 14:07:15 CST 2015
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine model: Palmetto BMC
Memory policy: Data cache writeback
SOC Rev: 02010303
On node 0 totalpages: 114176
free_area_init_node: node 0, pgdat c0502680, node_mem_map dba79000
  Normal zone: 892 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 114176 pages, LIFO batch:31
UART IO MUX...
UART PIN MUX...
SCU LOCK: 00000001
DONE, MUX=cb000000 00fff0c0
CLOCK_CTRL=00000001
WDT0C=00000000
WDT2C=00000082
DLL=1
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0 
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 113284
Kernel command line: console=ttyS4,38400
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Memory: 431416K/456704K available (3536K kernel code, 171K rwdata, 1260K rodata, 160K init, 108K bss, 25288K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xdc000000 - 0xff000000   ( 560 MB)
    lowmem  : 0xc0000000 - 0xdbe00000   ( 446 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc04b7720   (4798 kB)
      .init : 0xc04b8000 - 0xc04e0000   ( 160 kB)
      .data : 0xc04e0000 - 0xc050ac68   ( 172 kB)
       .bss : 0xc050ac68 - 0xc0526028   ( 109 kB)
NR_IRQS:16 nr_irqs:16 16
Aspeed VIC Initiallized
clocksource: moxart_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 39817925974 ns
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
Calibrating delay loop... 190.87 BogoMIPS (lpj=954368)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x40008400 - 0x40008458
devtmpfs: initialized
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
i2c-core: driver [dummy] registered
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
clocksource: Switched to clocksource moxart_timer
NET: Registered protocol family 2
TCP established hash table entries: 4096 (order: 2, 16384 bytes)
TCP bind hash table entries: 4096 (order: 2, 16384 bytes)
TCP: Hash tables configured (established 4096 bind 4096)
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
RPC: Registered named UNIX socket transport module.
RPC: Registered udp transport module.
RPC: Registered tcp transport module.
[... 83 lines cut ...]
systemd[1]: Listening on Journal Socket.
systemd[1]: Starting Remount Root and Kernel File Systems...
systemd[1]: Starting Apply Kernel Variables...
systemd[1]: Created slice User and Session Slice.
systemd[1]: Mounting Configuration File System...
systemd[1]: Listening on Journal Socket (/dev/log).
systemd[1]: Created slice system-getty.slice.
systemd[1]: Reached target Slices.
systemd[1]: Listening on udev Kernel Socket.
systemd[1]: Listening on udev Control Socket.
systemd[1]: Mounting Debug File System...
systemd[1]: Reached target Remote File Systems.
systemd[1]: Listening on Syslog Socket.
systemd[1]: Starting Journal Service...
systemd[1]: Mounting Temporary Directory...
systemd[1]: Mounted Configuration File System.
systemd[1]: Mounted Debug File System.
systemd[1]: Mounted Temporary Directory.
systemd[1]: Started Remount Root and Kernel File Systems.
systemd[1]: Started Apply Kernel Variables.
systemd[1]: Started Journal Service.
systemd-journald[388]: Received request to flush runtime journal from PID 1
ftgmac100: NCSI dev is up
random: nonblocking pool is initialized
```

shenki, 2015-11-01:
```
cat /proc/cmdline 
console=ttyS4,38400
```

nkskjames, 2015-11-02:
## instantiate a hwmon (this is barreleye)
echo nct7904 0x2e > /sys/bus/i2c/devices/i2c-6/new_device

## the tmp421 temperature sensor will cause same behavior on palmetto.

## set pwm to max
echo 255 > /sys/class/hwmon/hwmon0/pwm1

I get this in console (issue #11)
ystemd-journald[388]: /dev/kmsg buffer overrun, some messages lost.
root@palmetto:/ssystemd-journald[388]: /dev/kmsg buffer overrun, some messages lost.

I need to recreate to get log entries.

nkskjames, 2015-11-02:
This is on Palmetto
echo tmp423 0x4c > /sys/bus/i2c/devices/i2c-2/new_device

lots of these lines in journald log
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                             
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                             
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                             
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                             
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                             
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                             
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                             
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                             
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                             
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent               
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                             
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                             
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                          
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent         
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                 
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent          
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                             
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                      
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent             
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent                        
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent
Oct 31 23:13:18 palmetto kernel: i2c 2-004c: uevent

shenki, 2015-11-02:
I am not familiar with journald. Does it show us all of the messages from the kernel, or is some kind of filtering performed?

Can you paste the output of `dmesg | tail` and /proc/cmdline? 

On the specifics of your issue, this message comes from the core i2c code (https://github.com/torvalds/linux/blob/master/drivers/i2c/i2c-core.c#L492). It will take some more digging to work out why we're seeing this message, but it is harmless.

nkskjames, 2015-11-02:
root@palmetto:~# dmesg|tail
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent

dmesg totally rolled
root@palmetto:~# dmesg|head
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent
i2c 2-004c: uevent

root@palmetto:~# cat /proc/cmdline
console=ttyS4,38400

jk-ozlabs, 2015-11-04:
Looks like we're hitting a systemd issue:

 https://bugs.freedesktop.org/show_bug.cgi?id=76886

so perhaps we should disable `CONFIG_I2C_DEBUG_CORE`

jk-ozlabs, 2015-11-04:
Should be fixed by cff17c45f and 3cd273e4.

closed 2015-11-04 by jk-ozlabs