openbmc/linux #17: Kernel crash caused by network setting.

Kenthliu, 2015-11-05:
Reproduce step:
	1. Do not plug in LAN cable
	2. AC on
	3. Log in BMC 
	4. plug in LAN cable	
	5. Set ifconfig IP
	6. ifconfig eth0 down
	7. ifconfig eth0 up
	8. kernel crash

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2015.11.05 17:15:45 =~=~=~=~=~=~=~=~=~=~=~=

DRAM Init-DDR3
CBR0-1357135670123456701234567
CBR134Done

U-Boot 2013.07 (Nov 02 2015 - 10:34:04)

I2C:   ready
DRAM:  256 MiB
WARNING: Caches not enabled
Flash: SPI Flash ID: 1940ef 
32 MiB
In:    serial
Out:   serial
Err:   serial
H/W:   AST2400 series chip Rev. 01 
Watchdog: 300s
Net:   Found NCSI Network Controller at (0, 0)
Found NCSI Network Controller at (0, 1)
aspeednic#0
Warning: aspeednic#0 using MAC address from net device

Hit any key to stop autoboot:  3  2  1  0 
## Booting kernel from Legacy Image at 20080000 ...
   Image Name:   Phosphor OpenBMC (Phosphor OpenB
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1903084 Bytes = 1.8 MiB
   Load Address: 40008000
   Entry Point:  40008000
## Loading init Ramdisk from Legacy Image at 20300000 ...
   Image Name:   obmc-phosphor-image-barreleye-20
   Image Type:   ARM Linux RAMDisk Image (lzma compressed)
   Data Size:    16572752 Bytes = 15.8 MiB
   Load Address: 40800000
   Entry Point:  40800000
   Loading Kernel Image ... OK

Starting kernel ...

atags_to_fdt
atags=40000100
fdt=401d7540
cmdline=console=ttyS4,38400n8 root=/dev/ram rw
atags_to_fdt
atags=401d7540
fdt=406dabc0
Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0x0
Linux version 4.2.0-openbmc-20151104-1 (openpower@openpower-VirtualBox) (gcc version 4.9.3 (GCC) ) #1 Thu Nov 5 15:31:35 CST 2015
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
CPU: VIVT data cache, VIVT instruction cache
Machine model: Barrelye BMC
Memory policy: Data cache writeback
SOC Rev: 02010303
UART IO MUX...
UART PIN MUX...
SCU LOCK: 00000001
DONE, MUX=cb000000 00fff0c0
CLOCK_CTRL=00000001
[... 420 lines cut ...]
5ec0: c04e67b0 c034ff64 cf970030 cf970000 cf97000c c034ff50 cf955efc cf970000
5ee0: 1f004101 c04e0004 00000000 c04e70b0 cebac001 cf86b100 cf973030 c04e67b0
5f00: 00000000 c00263e4 cf86b100 cf973030 cfdedb00 cf86b100 cf86b118 cf86b118
5f20: cf954000 c04e67c0 c04e67b0 00000000 00000008 c0026544 cf973034 cf86b100
5f40: c04e67b0 c0026aa4 00000000 cf90a2a0 00000000 cf86b100 c00267cc 00000000
5f60: 00000000 00000000 00000000 c002a650 cf955f70 00000000 00000000 cf86b100
5f80: 00000000 cf955f84 cf955f84 00000000 cf955f90 cf955f90 cf955fac cf90a2a0
5fa0: c002a58c 00000000 00000000 c000a2e0 00000000 00000000 00000000 00000000
5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 1da40b2b 96260215
[<c022efe4>] (ftgmac100_hard_start_xmit) from [<c02c72b0>] (dev_hard_start_xmit+0x254/0x2ec)
[<c02c72b0>] (dev_hard_start_xmit) from [<c02dd848>] (sch_direct_xmit+0x78/0x180)
[<c02dd848>] (sch_direct_xmit) from [<c02c76c4>] (__dev_queue_xmit+0x28c/0x4b4)
[<c02c76c4>] (__dev_queue_xmit) from [<c034d9a8>] (ncsi_xmit_cmd+0x1a8/0x204)
[<c034d9a8>] (ncsi_xmit_cmd) from [<c034f554>] (ncsi_dev_start+0x84/0x3a8)
[<c034f554>] (ncsi_dev_start) from [<c034ff64>] (ncsi_dev_work+0x1a0/0x1ec)
[<c034ff64>] (ncsi_dev_work) from [<c00263e4>] (process_one_work+0x1bc/0x2f8)
[<c00263e4>] (process_one_work) from [<c0026544>] (process_scheduled_works+0x24/0x2c)
[<c0026544>] (process_scheduled_works) from [<c0026aa4>] (worker_thread+0x2d8/0x3f0)
[<c0026aa4>] (worker_thread) from [<c002a650>] (kthread+0xc4/0xd8)
[<c002a650>] (kthread) from [<c000a2e0>] (ret_from_fork+0x14/0x34)
Code: e1a03b83 e081220c e1a03ba3 e584347c (e582000c) 
---[ end trace b79e5d9832c7fe8d ]---
Kernel panic - not syncing: Fatal exception in interrupt
---[ end Kernel panic - not syncing: Fatal exception in interrupt

jk-ozlabs, 2015-11-05:
Looks similar to issue #8; I suspect we have a similar cause here.

gwshan, 2015-11-13:
It would be caused by a abnormal situation. From the beginning, we try to bring up the interface, but the NCSI interface is down for the attempt. The NCSI stack notifies the ftgmac100 driver about that and the driver shutdown itself down by freeing all BDs/buffers. However, the interface is still up. After that, the proceeding "ifconfig eth0 down" tries to close the NCSI dev. For that, the NCSI stack tries to send packets. Unfortunately, there're no available BDs/buffers any more.

The solution would be: (1) Return error when there're no available BDs/buffer or (2) Don't send NCSI packets when closing NCSI dev, or (3) The NIC and NCSI interface should have consistent state ("down" for the case).

gwshan, 2015-11-16:
The problem was reproduced successfully with private code: Force returning errcode from ncsi_select_active_channel(). The root cause (as last update) is still applied. I'll figure out the fix.

causten, 2015-12-08:
Any news on the fix?  I tried it today and no crash, but the network doesn't come back either

shenki, 2016-02-04:
We think that this is fixed with the lock fix. Please re-open if you see it again

closed 2016-02-04 by shenki