openbmc/linux #23: GPIO interrupt crashes kernel

nkskjames, 2015-11-30:
Kernel crashes when power button is pushed which is on gpio352.  This probably was introduced with 4.3 kernel.

root@barreleye:~# ls /sys/class/gpio
export       gpio326      gpio343      gpio353      gpiochip320
gpio324      gpio333      gpio344      gpio378      unexport
gpio325      gpio334      gpio352      gpio461

 
root@barreleye:~# cat /sys/class/gpio/gpio324/edge
none
root@barreleye:~# cat /sys/class/gpio/gpio324/edge
none
root@barreleye:~# cat /sys/class/gpio/gpio325/edge
none
root@barreleye:~# cat /sys/class/gpio/gpio326/edge
none
root@barreleye:~# cat /sys/class/gpio/gpio333/edge
none
root@barreleye:~# cat /sys/class/gpio/gpio334/edge
none
root@barreleye:~# cat /sys/class/gpio/gpio343/edge
none
root@barreleye:~# cat /sys/class/gpio/gpio344/edge
none
root@barreleye:~# cat /sys/class/gpio/gpio352/edge
both
root@barreleye:~# cat /sys/class/gpio/gpio353/edge
none
root@barreleye:~# cat /sys/class/gpio/gpio378/edge
none
root@barreleye:~# cat /sys/class/gpio/gpio461/edge
none

root@barreleye:~# cat /sys/class/gpio/gpio352/direction
in

root@barreleye:~# Unable to handle kernel paging request at virtual address 0f0e0d24
pgd = c0004000
[0f0e0d24] *pgd=00000000
Internal error: Oops: 5 [#1] ARM
Modules linked in:
CPU: 0 PID: 0 Comm: swapper Not tainted 4.3.0-openbmc-20151123-1 #1
Hardware name: ASpeed SoC
task: c04f5fb8 ti: c04f2000 task.ti: c04f2000
PC is at aspeed_gpio_irq_handler+0x70/0xe8
LR is at 0xf00c00a8
pc : [<c01bdb20>]    lr : [<f00c00a8>]    psr: 60000093
sp : c04f3f08  ip : cf8030c0  fp : 00000000
r10: 404e47f4  r9 : c0383a58  r8 : cf8030c0
r7 : cf802200  r6 : 00000000  r5 : c04ff960  r4 : cf86baa0
r3 : 0f0e0d24  r2 : 00000018  r1 : f00c0080  r0 : 01870000
Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 0005317f  Table: 48b08000  DAC: 00000053
Process swapper (pid: 0, stack limit = 0xc04f2190)
Stack: (0xc04f3f08 to 0xc04f4000)
3f00:                   cf86baa0 00000016 00000016 c04ff728 00000000 cf802200
3f20: 00000001 cfffc680 404e47f4 c003ad48 000000ea c003af8c cf8030c0 c04f3f60
3f40: ffffffff c04f3f94 c04e5a28 c00093cc c000ab98 c000aba4 60000013 c000d970
3f60: 00000000 0005317f 0005217f 60000013 c04f2000 00000001 c04f4068 c04f4000
3f80: c04e5a28 cfffc680 404e47f4 00000000 600000d3 c04f3fb0 c000ab98 c000aba4
3fa0: 60000013 ffffffff 00000053 c03647c0 00000000 c0034d24 00000001 c04c5bf4
3fc0: ffffffff ffffffff 00000000 c04c5684 00000000 c04e5a28 00000000 c051cd34
3fe0: c04f4018 c04e5a24 c04f6d30 40004000 41069265 40008048 00000000 00000000
[<c01bdb20>] (aspeed_gpio_irq_handler) from [<c003ad48>] (generic_handle_irq+0x18/0x28)
[<c003ad48>] (generic_handle_irq) from [<c003af8c>] (__handle_domain_irq+0x7c/0xa0)
[<c003af8c>] (__handle_domain_irq) from [<c00093cc>] (avic_handle_irq+0x5c/0x64)
[<c00093cc>] (avic_handle_irq) from [<c000d970>] (__irq_svc+0x50/0x64)
Exception stack(0xc04f3f60 to 0xc04f3fa8)
3f60: 00000000 0005317f 0005217f 60000013 c04f2000 00000001 c04f4068 c04f4000
3f80: c04e5a28 cfffc680 404e47f4 00000000 600000d3 c04f3fb0 c000ab98 c000aba4
3fa0: 60000013 ffffffff
[<c000d970>] (__irq_svc) from [<c000aba4>] (arch_cpu_idle+0x2c/0x34)
[<c000aba4>] (arch_cpu_idle) from [<c0034d24>] (cpu_startup_entry+0x7c/0xd4)
[<c0034d24>] (cpu_startup_entry) from [<c04c5bf4>] (start_kernel+0x32c/0x3a0)
Code: e1d330b2 e2832010 e598306c e0833002 (e5933000) 
---[ end trace c800b08928a1dd92 ]---
Kernel panic - not syncing: Fatal exception in interrupt
---[ end Kernel panic - not syncing: Fatal exception in interrupt

jk-ozlabs, 2015-12-01:
Taking a look.

jk-ozlabs, 2015-12-01:
@nkskjames : could you send me the vmlinux used in that build, or add the disassembly of that compile of the `aspeed_gpio_irq_handler` function, as a comment?

jk-ozlabs, 2015-12-01:
@nkskjames is testing with https://github.com/jk-ozlabs/linux/commit/87c10ddd748e0a45508ea1cc8a7504dcff103987

nkskjames, 2015-12-01:
jk-ozlabs@87c10dd fixes it

jk-ozlabs, 2015-12-01:
@nkskjames thanks for testing. Merged as https://github.com/openbmc/linux/commit/6041e18256bd64692c576f7393073c2d14e778eb

closed 2015-12-01 by jk-ozlabs