openbmc/linux #25: Provide a multiplatform kernel for ast & qemu

jk-ozlabs, 2015-12-03:
Would be nice to use a single build & source tree for hardware and qemu.

If we don't have a suitable qemu model available, a single source tree (but separate build) would be ok.

williamspatrick, 2016-01-09:
@causten - You might be interested in this.

amboar, 2016-01-09:
Related to this, I've got some patches to qemu that boot the aspeed ast2400 kernel. They're still a work in progress (I.e. likely to get rebased),  but if you're keen to test out review them they are available in my ast2400 branch:

https://github.com/amboar/qemu/tree/ast2400

shenki, 2016-01-14:
Once @amboar has his model then we will only need the one kernel.

In the mean time we can look at doing a multiplatform kernel. What machine type does poky currently use to boot in qemu?

causten, 2016-01-14:
using the kernel we boot our barreleyes with instead of the yocto default one would be a great step forward.  Currently our qemu image runs on the versatilepb machine.  When do I get this feature?

/home/causten/openbmc-barreleye/build/tmp/sysroots/x86_64-linux/usr/bin/qemu-system-arm -kernel /home/causten/openbmc-barreleye/build/tmp/deploy/images/qemuarm/zImage-qemuarm.bin -net nic,vlan=0 -net tap,vlan=0,ifname=tap0,script=no,downscript=no -M versatilepb -drive file=/home/causten/openbmc-barreleye/build/tmp/deploy/images/qemuarm/obmc-phosphor-image-qemuarm-20160114150324.rootfs.ext4,format=raw -no-reboot -show-cursor -usb -usbdevice wacom-tablet -no-reboot -m 128 --append "root=/dev/sda rw console=ttyAMA0,115200 console=tty ip=192.168.7.2::192.168.7.1:255.255.255.0 mem=128M highres=off rootfstype=ext4 "

Chris Austen
POWER Systems Enablement Manager 
(512) 286-5184 (T/L: 363-5184)

-----

amboar, 2016-01-14:
@causten you can try the code now ( see https://github.com/openbmc/linux/issues/25#issuecomment-170186246 ), but it's still very much a WIP. The critical bits for qemu invocation are:

```
qemu -machine ast2400 -m 512M -dtb aspeed-bmc-opp-palmetto.dtb -append 'console=ttyS4' -nographic -serial pty -kernel ...
```

legoater, 2017-02-13:
now that qemu can boot flash images for the palmetto and romulus platforms, may be we can close this issue ?

closed 2018-07-05 by shenki