openbmc/linux #26: ftgmac crashing

causten, 2015-12-03:
I am on palm303.  I am testing with  Sprint 3 Candidate #114  

root@palmetto:~# ------------[ cut here ]------------
WARNING: CPU: 0 PID: 0 at /home/causten/openbmc-barreleye/build/tmp/work-shared/palmetto/kernel-source/kernel/softirq.c:150 __local_bh_enable_ip+0x48/0xbc()
Modules linked in:
CPU: 0 PID: 0 Comm: swapper Not tainted 4.3.0-openbmc-20151202-1 #1
Hardware name: ASpeed SoC
[<c000f28c>] (unwind_backtrace) from [<c000cf0c>] (show_stack+0x10/0x14)
[<c000cf0c>] (show_stack) from [<c0016734>] (warn_slowpath_common+0x84/0xac)
[<c0016734>] (warn_slowpath_common) from [<c00167ec>] (warn_slowpath_null+0x18/0x20)
[<c00167ec>] (warn_slowpath_null) from [<c0018eb4>] (__local_bh_enable_ip+0x48/0xbc)
[<c0018eb4>] (__local_bh_enable_ip) from [<c02d9fac>] (dst_alloc+0x50/0x84)
[<c02d9fac>] (dst_alloc) from [<c02f3848>] (rt_dst_alloc+0x4c/0x58)
[<c02f3848>] (rt_dst_alloc) from [<c02f5118>] (ip_route_input_noref+0x60c/0x7dc)
[<c02f5118>] (ip_route_input_noref) from [<c0321394>] (arp_process+0x22c/0x668)
[<c0321394>] (arp_process) from [<c0321b04>] (arp_rcv+0x108/0x17c)
[<c0321b04>] (arp_rcv) from [<c02cf064>] (__netif_receive_skb_core+0x674/0x7a8)
[<c02cf064>] (__netif_receive_skb_core) from [<c02d0658>] (netif_receive_skb_internal+0x4c/0x58)
[<c02d0658>] (netif_receive_skb_internal) from [<c02d0df0>] (napi_gro_receive+0x48/0x9c)
[<c02d0df0>] (napi_gro_receive) from [<c0239084>] (ftgmac100_poll+0x398/0x594)
[<c0239084>] (ftgmac100_poll) from [<c02d129c>] (net_rx_action+0xe0/0x290)
[<c02d129c>] (net_rx_action) from [<c0018ca0>] (__do_softirq+0xc0/0x1ec)
[<c0018ca0>] (__do_softirq) from [<c003af50>] (__handle_domain_irq+0x84/0xa0)
[<c003af50>] (__handle_domain_irq) from [<c00093cc>] (avic_handle_irq+0x5c/0x64)
[<c00093cc>] (avic_handle_irq) from [<c000d970>] (__irq_svc+0x50/0x64)
Exception stack(0xc04f3f60 to 0xc04f3fa8)
3f60: 00000000 0005317f 0005217f 60000013 c04f2000 00000001 c04f4068 c04f4000
3f80: c04e6a28 dbdfcec0 404e5768 00000000 600000d3 c04f3fb0 c000ab98 c000aba4
3fa0: 60000013 ffffffff
[<c000d970>] (__irq_svc) from [<c000aba4>] (arch_cpu_idle+0x2c/0x34)
[<c000aba4>] (arch_cpu_idle) from [<c0034ce0>] (cpu_startup_entry+0x7c/0xd4)
[<c0034ce0>] (cpu_startup_entry) from [<c04c6bf4>] (start_kernel+0x32c/0x3a0)
---[ end trace 9f44e726bf4b59cf ]---

shenki, 2015-12-03:
@gwshan can you please take a look?

gwshan, 2015-12-03:
It is same as https://github.com/openbmc/openbmc/issues/107

causten, 2015-12-07:
I agree this is the same as #107, I'll use it for tracking

closed 2015-12-07 by causten