openbmc/linux #27: tmp423 probe is failing

nkskjames, 2015-12-07:
On palmetto, the tmp423 that is in device tree is not being instantiated:

[   67.340000] tmp421 2-004c: Could not read configuration register (-5)
[   67.350000] tmp421: probe of 2-004c failed with error -5

but, I can read the mfg and device id reg manually and they look ok:

root@palmetto:~# i2cget 2 0x4c 0xff b
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will read from device file /dev/i2c-2, chip address 0x4c, data address
0xff, using read byte data.
Continue? [Y/n] 
0x23

I also tried to manually instantiate the occ driver, and it gives an error too on Palmetto only.  Barreleye is fine.  the lm75 works fine on Barreleye.

jk-ozlabs, 2015-12-07:
Could you try:

    dmesg -c > /dev/null
    echo "module i2c_aspeed +p" > /sys/kernel/debug/dynamic_debug

Then attempt the probe, then paste the resulting `dmesg`?

nkskjames, 2015-12-07:
if i remove from device tree and manually instantiate, it works.  so this must be another case of the SCU's not being setup properly in the kernel for i2c-2.   i will get those bits into aspeed.c as referenced in https://github.com/openbmc/skeleton/issues/20

closed 2016-01-11 by nkskjames