openbmc/linux #33: hwclock displaying error on empty date with kernel openbmc-20151210-1

anoo1, 2015-12-18:
When AC cycle is applied, the RTC date is empty (issue been tracked separately to find out why the RTC is not persisting the date).
With kernel 201521202, issuing "hwclock" from the BMC command line displays the epoch time.
After updating to kernel openbmc-20151210-1, hwclock displays an error:

root@barreleye:~# hwclock
hwclock: RTC_RD_TIME: Invalid argument

After updating the RTC via "hwclock -w", the "hwclock" command will display the date, so the issue is only when the RTC is empty.

shenki, 2016-02-04:
This should be resolved now that we disable the internal rtc in the device tree. Please re-open if you still see this issue.

closed 2016-02-04 by shenki

mdmillerii, 2016-02-04:
This could easly be  due to the month off-by-one that we fixed too.  (especially in December or January).