openbmc/linux #35: BMC kernel panic cause by AC power cycle

Kenthliu, 2015-12-23:
We could see kernel panic after AC cycle few times. Below is the BMC console information.
[bmc_kernalpanic.txt](https://github.com/openbmc/linux/files/70547/bmc_kernalpanic.txt)

shenki, 2016-02-04:
Here is the oops:

```
bt-host 1e789140.ibt: Found bt host device
bt-host 1e789140.ibt: Using IRQ 18
Unable to handle kernel paging request at virtual address 003045c0
pgd = c0004000
[003045c0] *pgd=00000000
Internal error: Oops: 805 [#1] ARM
Modules linked in:
CPU: 0 PID: 1 Comm: swapper Not tainted 4.3.0-openbmc-20151202-1 #1
Hardware name: ASpeed SoC
task: cf810aa0 ti: cf826000 task.ti: cf826000
PC is at __memzero+0x24/0x80
LR is at 0x0
pc : [<c01938e4>]    lr : [<00000000>]    psr: 20000053
sp : cf827e6c  ip : 00000000  fp : 00000000
r10: c051e560  r9 : 00000000  r8 : 00000000
r7 : a0000053  r6 : cf8003a0  r5 : 000080d0  r4 : 003045c0
r3 : 00000000  r2 : 00000000  r1 : 00000fc0  r0 : 003045c0
Flags: nzCv  IRQs on  FIQs off  Mode SVC_32  ISA ARM  Segment none
Control: 0005317f  Table: 40004000  DAC: 00000053
Process swapper (pid: 1, stack limit = 0xcf826190)
Stack: (0xcf827e6c to 0xcf828000)
7e60:                            c0088454 c052ae94 cfa823a0 00000000 c0517fc8
7e80: cf918e00 c0196d50 c0518760 c0508418 00000000 cfa823a0 c05074ac cf8fdae0
7ea0: 00000000 c0433180 00000000 c0508418 00000000 c05080c0 0000000c 000000b8
7ec0: c04e6838 c051e560 00000000 c01e33d8 c02341f8 c05083a0 00000000 c023536c
7ee0: 00000000 00000004 00000000 c02353f4 cc9e2e40 c04dbb8c 00000000 c04f5de0
7f00: c04f5de0 c00096d8 cf80bbe0 cf805d00 cf805d00 c0369054 c0526700 00000000
7f20: 0000006c cfffc6e4 00000000 c002b718 cf80bbe0 cf804c20 c04a37e0 00000006
7f40: 00000006 c04a4064 0000006c c04a4064 00000000 00000006 c04e682c 00000006
7f60: c04e6830 0000006d c04f0fb0 c051e560 c04e6838 c04c6d68 00000006 00000006
7f80: 00000000 c04c65a0 00000000 c03644d4 00000000 00000000 00000000 00000000
7fa0: 00000000 c03644dc 00000000 c000a330 00000000 00000000 00000000 00000000
7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
7fe0: 00000000 00000000 00000000 00000000 00000013 00000000 951f3100 d5b8026e
[<c01938e4>] (__memzero) from [<c0088454>] (kmem_cache_alloc+0xa8/0xd0)
[<c0088454>] (kmem_cache_alloc) from [<c0196d50>] (kobject_uevent_env+0x170/0x5a4)
[<c0196d50>] (kobject_uevent_env) from [<c01e33d8>] (driver_register+0xd0/0xe0)
[<c01e33d8>] (driver_register) from [<c023536c>] (phy_driver_register+0x30/0x94)
[<c023536c>] (phy_driver_register) from [<c02353f4>] (phy_drivers_register+0x24/0x60)
[<c02353f4>] (phy_drivers_register) from [<c00096d8>] (do_one_initcall+0x114/0x1d4)
[<c00096d8>] (do_one_initcall) from [<c04c6d68>] (kernel_init_freeable+0x100/0x1c4)
[<c04c6d68>] (kernel_init_freeable) from [<c03644dc>] (kernel_init+0x8/0xe4)
[<c03644dc>] (kernel_init) from [<c000a330>] (ret_from_fork+0x14/0x24)
Code: e52de004 e1a0c002 e1a0e002 e2511040 (a8a0500trace 23c433632fc2923a ]---
```

shenki, 2016-02-04:
@Kenthliu have you seen this with newer versions of the firmware?

Kenthliu, 2016-02-04:
@shenki In which version or commit? I saw this issue in 0.2. If this is fixed, I will verify in 0.3.

shenki, 2016-03-21:
This looks related to our u-boot issue. Please reopen if you can reproduce with the fixed u-boot

closed 2016-03-21 by shenki