openbmc/linux #44: Clean up power8 occ hwmon driver

shenki, 2016-01-27:
```  CC      drivers/hwmon/power8_occ_i2c.o
drivers/hwmon/power8_occ_i2c.c: In function 'get_occdata_length':
drivers/hwmon/power8_occ_i2c.c:351:22: warning: passing argument 1 of '__be16_to_cpup' from incompatible pointer type
  return be16_to_cpup(&data[RESP_DATA_LENGTH]);
                      ^
In file included from include/linux/byteorder/little_endian.h:4:0,
                 from ./arch/arm/include/uapi/asm/byteorder.h:21,
                 from include/asm-generic/bitops/le.h:5,
                 from ./arch/arm/include/asm/bitops.h:340,
                 from include/linux/bitops.h:36,
                 from include/linux/kernel.h:10,
                 from include/linux/list.h:8,
                 from include/linux/module.h:9,
                 from drivers/hwmon/power8_occ_i2c.c:18:
include/uapi/linux/byteorder/little_endian.h:87:21: note: expected 'const __be16 *' but argument is of type 'uint8_t *'
 static inline __u16 __be16_to_cpup(const __be16 *p)
                     ^
drivers/hwmon/power8_occ_i2c.c: In function 'parse_occ_response':
drivers/hwmon/power8_occ_i2c.c:424:40: warning: passing argument 1 of '__be16_to_cpup' from incompatible pointer type
     f_sensor->sensor_id = be16_to_cpup(&data[dnum]);
                                        ^
In file included from include/linux/byteorder/little_endian.h:4:0,
                 from ./arch/arm/include/uapi/asm/byteorder.h:21,
                 from include/asm-generic/bitops/le.h:5,
                 from ./arch/arm/include/asm/bitops.h:340,
                 from include/linux/bitops.h:36,
                 from include/linux/kernel.h:10,
                 from include/linux/list.h:8,
                 from include/linux/module.h:9,
                 from drivers/hwmon/power8_occ_i2c.c:18:
include/uapi/linux/byteorder/little_endian.h:87:21: note: expected 'const __be16 *' but argument is of type 'uint8_t *'
 static inline __u16 __be16_to_cpup(const __be16 *p)
                     ^
drivers/hwmon/power8_occ_i2c.c:425:36: warning: passing argument 1 of '__be16_to_cpup' from incompatible pointer type
     f_sensor->value = be16_to_cpup(&data[dnum+2]);
                                    ^
In file included from include/linux/byteorder/little_endian.h:4:0,
                 from ./arch/arm/include/uapi/asm/byteorder.h:21,
                 from include/asm-generic/bitops/le.h:5,
                 from ./arch/arm/include/asm/bitops.h:340,
                 from include/linux/bitops.h:36,
                 from include/linux/kernel.h:10,
                 from include/linux/list.h:8,
                 from include/linux/module.h:9,
                 from drivers/hwmon/power8_occ_i2c.c:18:
include/uapi/linux/byteorder/little_endian.h:87:21: note: expected 'const __be16 *' but argument is of type 'uint8_t *'
 static inline __u16 __be16_to_cpup(const __be16 *p)
                     ^
drivers/hwmon/power8_occ_i2c.c:441:40: warning: passing argument 1 of '__be16_to_cpup' from incompatible pointer type
     t_sensor->sensor_id = be16_to_cpup(&data[dnum]);
                                        ^
In file included from include/linux/byteorder/little_endian.h:4:0,
                 from ./arch/arm/include/uapi/asm/byteorder.h:21,
                 from include/asm-generic/bitops/le.h:5,
                 from ./arch/arm/include/asm/bitops.h:340,
                 from include/linux/bitops.h:36,
                 from include/linux/kernel.h:10,
                 from include/linux/list.h:8,
                 from include/linux/module.h:9,
                 from drivers/hwmon/power8_occ_i2c.c:18:
include/uapi/linux/byteorder/little_endian.h:87:21: note: expected 'const __be16 *' but argument is of type 'uint8_t *'
 static inline __u16 __be16_to_cpup(const __be16 *p)
                     ^
drivers/hwmon/power8_occ_i2c.c:442:36: warning: passing argument 1 of '__be16_to_cpup' from incompatible pointer type
     t_sensor->value = be16_to_cpup(&data[dnum+2]);
                                    ^
In file included from include/linux/byteorder/little_endian.h:4:0,
                 from ./arch/arm/include/uapi/asm/byteorder.h:21,
                 from include/asm-generic/bitops/le.h:5,
                 from ./arch/arm/include/asm/bitops.h:340,
[... 161 lines cut ...]
                 from include/linux/module.h:9,
                 from drivers/hwmon/power8_occ_i2c.c:18:
drivers/hwmon/power8_occ_i2c.c: In function 'set_user_powercap':
include/linux/dynamic_debug.h:64:16: warning: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'int' [-Wformat=]
  static struct _ddebug  __aligned(8)   \
                ^
include/linux/dynamic_debug.h:84:2: note: in expansion of macro 'DEFINE_DYNAMIC_DEBUG_METADATA'
  DEFINE_DYNAMIC_DEBUG_METADATA(descriptor, fmt);  \
  ^
include/linux/device.h:1171:2: note: in expansion of macro 'dynamic_dev_dbg'
  dynamic_dev_dbg(dev, format, ##__VA_ARGS__); \
  ^
drivers/hwmon/power8_occ_i2c.c:1141:2: note: in expansion of macro 'dev_dbg'
  dev_dbg(dev, "set user powercap to: %lu\n", val);
  ^
drivers/hwmon/power8_occ_i2c.c:1142:41: warning: passing argument 5 of 'occ_send_cmd' from incompatible pointer type
  err = occ_send_cmd(client, 0, 0x22, 2, &val, resp);
                                         ^
drivers/hwmon/power8_occ_i2c.c:535:16: note: expected 'uint8_t *' but argument is of type 'uint16_t *'
 static uint8_t occ_send_cmd(struct i2c_client *client, uint8_t seq,
                ^
drivers/hwmon/power8_occ_i2c.c:1133:11: warning: unused variable 'powercap' [-Wunused-variable]
  uint32_t powercap;
           ^
```

shenki, 2016-02-10:
@adamliyi can you please take a look at these warnings. Although the driver is in the tree, they must be fixed before I will review again.

adamliyi, 2016-02-22:
OK. I will fix the warning.

adamliyi, 2016-02-29:
@shenki , Joel, I just fixed the warning message issue. in: https://github.com/openbmc/linux/pull/56
Could you please have a look?

cyrilbur-ibm, 2016-03-01:
I've commented on the patch, it looks ok. Is __beXX what we should be using?

shenki, 2019-01-16:
Driver was reworked and is now upstream

closed 2019-01-16 by shenki