openbmc/linux #61: ftgmac is still not detecting link if cable plugged in after BMC is booted

nkskjames, 2016-03-15:
Here are steps to fail:
1. LAN cable plugged
2. BMC booted
3. Setup network
[network works fine]
4. turn off AC
5. unplug LAN cable
6. turn on AC
7. BMC boots
8. Plug in cable
9. LAN is not enabled (no link detected)

gwshan, 2016-03-17:
Did we issue below commands after step (8) ? By the way, who from IRC channel I can get more info?

ifconfig eth0 down && ifconfig eth0 up

nkskjames, 2016-03-17:
ifconfig eth0 down && ifconfig eth0 up does make network work.  The expectation from end user is that when the cable is plugged the link should come up on its own though.  Is that not correct?  I will see if we can get person who reported this on IRC.

gwshan, 2016-03-17:
No, when the cable is plugged, the network interface won't be up automatically. When MAC has NCSI is enabled, we don't have a PHY attached. That means we don't have a thread monitoring the link status and wake up the MAC when link is ready. With NCSI, we have to issue "ifconifg ethx down && ifconfig eth0 up" after the cable is plugged.

The problem here is the above command doesn't help after cable is plugged.

nkskjames, 2016-03-17:
"The problem here is the above command doesn't help after cable is plugged."
not sure what you mean. ifconfig down/up does work after cable is plugged.

NC-SI specifies an async event notification for things like link state change.  Please see section 6.7 in: http://www.dmtf.org/sites/default/files/standards/documents/DSP0222_1.0.1.pdf.   If that doesn't work, it is not unusual to poll for link status from what I understand.

gwshan, 2016-03-21:
Yeah, you're correct that the AEN packet will be sent upon link-down event. The NCSI stack will try to choose another available channel if existing. Otherwise, the NCSI interface is down and I guess it's the case. When NCSI stack receives AEN on link-up event, it tries to choose an available channel for communication if we don't have one yet.

I'll try to reproduce this issue on local BMC tomorrow or a bit later...

nkskjames, 2016-04-11:
What is the status of this?

gwshan, 2016-04-12:
I don't have a chance to reproduce it locally. I will try it later of this week or early of next week.

anoo1, 2016-04-15:
Customer reported same issue today. Any progress on recreating?

gwshan, 2016-04-16:
Not yet, I'll continue on this next Monday when sitting around the BMC :-)

gwshan, 2016-04-18:
I'm using "dev-4.3" branch as the base source code to debug the issue. Also the hardware platform is Palmetto. The experiment I did is as below and it should match with the experiment that James did.

- Unplug the cable;
- Load the kernel and boot it;
- ifconfig eth0 xxx.xxx.xxx.xxx, and we should see the interface is down;
- Plug the cable and issue command "ifcofig eth0 down && ifconfig eth0 up". The network is back to service.

One thing I observed from James saw is: Reconfiguring the network interface after plugging the network cable will bring the network back. I also added some code to print the AEN packet received on plugging or unplugging the network cable. However, I never see a AEN packet received.

nkskjames, 2016-04-20:
So what next?

apopple, 2016-04-20:
@gwshan it sounds like you are expecting to receive an AEN packet but aren't? Does the device driver rely on getting this packet? If so we should try and debug why you aren't getting them.

gwshan, 2016-04-21:
Yes, I expect to receive AEN indicating link change on the active channel and I don't receive it when unplugging the ethernet cable. I added some printk() in the NCSI stack and print the received AEN packet. I don't see it.

mdmillerii, 2016-04-26:
Do you get an AEN when the link drops?
Do you have a poller to make sure the NCSI selection is still active? (see 6.8.3)

Kenthliu, 2016-04-27:
@gwshan Did you try wireshark to check the packet?

gwshan, 2016-04-27:
@Kenliu when the NCSI interface is down, no frames will be received on the
interface. So there're no packets seen on wireshark. As I mentioned before,
I add some printk() in the source code of the NCSI stack that dumps the
packets on receiving AEN packet. Nothing was dumped.

In current design, the NCSI and other (like ARP/IP) packets are transmitted
or received through same channel. When opening the NCSI, the channel is
enabled and NCSI packets exchanged with remote end to setup the NCSI
interface. the channel is going to be disabled if that fails. Otherwise,
the channel is enabled and monitor on AEN packets. In order to support
polling mechanism, we need separate channels for NCSI and other packets. It
will introduce huge changes to current farady driver implementation and I
need evaluate after finishing work at hand. On the other hand, there is one
additional question: when the cable is unplugged, can we probe the
situation by NCSI (Get Link Status packet) response without issue? I need
check it out later as well.

Another thing I'm not satisfied with current NCSI stack is: all available
package/channel are probed when opening up the ethernet interface. However,
once the NCSI topology is finalized for once and it's not going to change
dynamically. I need improve it later.

jk-ozlabs, 2016-04-28:
@nkskjames If we can get some info from Broadcom about the absence of that AEN message, that'd be helpful for us.

Kenthliu, 2016-04-28:
Yes. I already sent an issue to Broadcom. Waiting for the response.
[image]

Kenthliu, 2016-05-06:
Hi @gwshan and @jk-ozlabs 
Response from Broadcom. Do you agree this?
[image]

gwshan, 2016-05-06:
Hi Ken, Thanks for checking with broadcom on this. It sounds the case. I
will enable that explicitly with AEN enable
command packet. Prior to that, I have to introduce some changes to farady
driver so that we can always receive
ingress NCSI packets.

However, the statement you had isn't always true according to the
experiment I did. Initially, I never send a AEN enable
packet to target NCSI channel. With a PCI slot reset through GPIO PINs, I
still can receive AEN packet from far end.

gwshan, 2016-05-07:
Thanks again for Ken's feedback from Broadcom. I added some code in the
NCSI stack to enable AEN explicitly by transmitting a AEN enablement packet
on the active channel. I can receive AEN packet from far end (broadcom)
when unplugging the cable. However the AEN packet is still missed when
plugging the cable. I assume something is missed in the code. I will
continue the debugging.

2016-05-07 0:12 GMT+10:00 Shan Gavin <shan.gavin@gmail.com>:

gwshan, 2016-05-09:
A pachset sent to maillist for review. The issue is fixed by it. I had below testing with the patchset on palm4-bmc. Everything looks good:

   * From uboot, download the kernel image;
   * Unplug the ethernet cable;
   * From uboot, boot the kernel image;
   * Configure the ethernet interface when the kernel boots up completely.
     The NCSI interface should be down;
   * Plug the cable, the NCSI interface is up automatically. The interface
     is pingable from external.

shenki, 2016-05-24:
I think this is resolved. Please re-open if it happens again.

closed 2016-05-24 by shenki