openbmc/linux #66: pr_warn from fs/jffs2/scan.c line 671 in new 4.4 kernel

mdmillerii, 2016-03-22:
The jffs2 file system over the ast2400 fmc driver is working fine on the dev-4.3 branch 35d2cfb27392c73096b9cbdb0e250a900474301f .  But the file system is spewing pr_warn in the dev-4.4 branch 1500e269432548059f34907990465cbba34437c5 . 

The spew is from fs/jffs2/scan.c line 671.  The whole scan.c file has not changed between revisions.  A quick web search shows this is likely due to either file system corruption or a bad driver.  After scanning file system for tens of minutes the file system typically reports an error.  Returning to a previous 4.3 kernel build the flash is able to be read.

The error could be caused by flash partition alignment, by flash driver, or by something elsewhere in the kernel.

I did a very quick web search and found plenty of hits the first few I looked at were recovered with erasing and formating the whole partition.  I did a git diff on the file and there have been zero changes since Febuary 2012 in kernel v3.4-rc1.   I did a git diff between drivers/mtd/spi-nor/ and see changes including moving the struct mtd around, but no obvious errors on first scan.

It has been suggested a bisect be attempted, but that will require cherry picking each aspeed support patch stack for each rev (outside the add support options).  I feel an attempt to recreate on an existing mtd platform in qemu or similar be attempted first, and/or logging the IO to the flash to compare the pattern.

mdmillerii, 2016-03-24:
Doing a git bisect run with the test being the rofs could be the root file system yields:
git bisect start
git bisect good 35d2cfb27392c73096b9cbdb0e250a900474301f
git bisect bad 1500e269432548059f34907990465cbba34437c5
git bisect good 6a13feb9c82803e2b815eca72fa7a9f5561d7861
git bisect good ab1228e42e71f5cb687c740c4c304f1d48bcf68a
git bisect good ea5c58e70c3a148ada0d3061a8f529589bb766ba
git bisect bad 9b1b61cd8e31d9beba871333d7a798b3adb89288
git bisect bad bd4f203e433387d39be404b67ad02acf6f76b7bc
git bisect good 2d49dcb9e48f65a69281fe4c698c8f1a20215daf
git bisect bad 713009809681e5a7871e96e6992692c805b4480b
git bisect bad 02f0d3f758ab456c50199b723a53f2443fa4f684
git bisect skip 0df415598ef6551070267b1d8c9dc63f4ac00117
git bisect bad 8e2c992b59fcb5e56e3667f5c30c7d26fbbf14a2
git bisect bad 271707b1d817f5104e02b2bd1bab43f0c8759418
git bisect bad 0791a5f8ab5082d2726d99f3b21036d438c11424
git bisect bad 0791a5f8ab5082d2726d99f3b21036d438c11424
git bisect good 6ff33f3902c3b1c5d0db6b1e2c70b6d76fba357f
git bisect bad 11bff0b70cc003b995cf4c2acec4adab90957b02
git bisect good e747dbe75e83345379455a78bb208ab7202229df
git bisect good 7eadd47fd708df2028877558d085f47dc9086459
git bisect good f9b97fe61402f7f18da71e034b1e72f68df82c1a
git bisect good f9b97fe61402f7f18da71e034b1e72f68df82c1a

Showing commit 11bff0b70cc003b995cf4c2acec4adab90957b02
as the first bad commit.  I reverifed I could mount on f9b97fe61402f7f18da71e034b1e72f68df82c1a
and failed on 11bff0b70cc003b995cf4c2acec4adab90957b02 , even after adding
`chip->nor.flash_node = child;` and removing the assignment to `chip->mtd.priv`.

## prep:
    git diff v4.4.6 openbmc/dev-4.4 > aspeed.patch
    git diff openbmc/dev-4.4 openbmc/dev-4.3 -- drivers/mtd/spi-nor/aspeed_smc.c > aspeed-spi-nor-4.3

## u-boot boot prep
    setenv test 'setenv tftp cuImage_mdm ; bootm ${loadaddr}'
    setenv justro 'setenv bootargs ${bootargs} root=/dev/mtdblock5 rootfstype=squashfs ro saveenv'

## each step
    git clean -e obj -d -f
    git reset --hard
    git bisect <blah>
    git apply --whitespace nowarn --reject aspeed.patch
    git checkout openbmc/dev-4.3 arch/arm/configs/aspeed_defconfig
    git checkout openbmc/dev-4.4 drivers/clocksource/moxart_timer.c drivers/irqchip/Makefile
    make O=obj/ -k -j15  ARCH=arm CROSS_COMPILE=${ARM_CROSS_COMPILE} aspeed_defconfig arch/arm/boot/dts/aspeed-bmc-opp-palmetto.dtb
    # remove CONFIG_I2C_ASPEED  as it will encounter a NULL pointer from rejects
    vi arch/arm/mach-aspeed/Kconfig obj/.config
    make O=obj/ -k -j15  ARCH=arm CROSS_COMPILE=${ARM_CROSS_COMPILE} oldconfig 
    make O=obj/ -k -j15  ARCH=arm CROSS_COMPILE=${ARM_CROSS_COMPILE} zImage aspeed-bmc-opp-palmetto.dtb

## in u-boot
    run justro test

mdmillerii, 2016-03-24:
It turns out I identified the fix (filling out `flash_node`) but missed a step copying the image over for test.

shenki, 2016-03-29:
This is fixed in https://github.com/openbmc/linux/commit/aeb4718beca0d07ff232341cdd544008e17f1fdc

closed 2016-03-29 by shenki