openbmc/linux #70: lockdep warning with 4.6-rc1

shenki, 2016-03-30:
```
# umount /mnt
[  159.060000] INFO: trying to register non-static key.
[  159.060000] the code is fine but needs lockdep annotation.
[  159.060000] turning off the locking correctness validator.
[  159.060000] CPU: 0 PID: 318 Comm: umount Not tainted 4.6.0-rc1-00031-gd725a4eaaf03 #119
[  159.060000] Hardware name: ASpeed SoC
[  159.060000] [<c0107550>] (unwind_backtrace) from [<c0105128>] (show_stack+0x10/0x14)
[  159.060000] [<c0105128>] (show_stack) from [<c01357c4>] (__lock_acquire+0x67c/0xb6c)
[  159.060000] [<c01357c4>] (__lock_acquire) from [<c0136424>] (lock_acquire+0x70/0x90)
[  159.060000] [<c0136424>] (lock_acquire) from [<c01210e8>] (flush_work+0x48/0x244)
[  159.060000] [<c01210e8>] (flush_work) from [<c0123848>] (__cancel_work_timer+0x10c/0x1a8)
[  159.060000] [<c0123848>] (__cancel_work_timer) from [<c02248c8>] (jffs2_sync_fs+0x14/0x38)
[  159.060000] [<c02248c8>] (jffs2_sync_fs) from [<c01ce2f4>] (sync_filesystem+0x54/0x84)
[  159.060000] [<c01ce2f4>] (sync_filesystem) from [<c01a7c18>] (generic_shutdown_super+0x24/0xdc)
[  159.060000] [<c01a7c18>] (generic_shutdown_super) from [<c029ef18>] (kill_mtd_super+0xc/0x20)
[  159.060000] [<c029ef18>] (kill_mtd_super) from [<c02247bc>] (jffs2_kill_sb+0x28/0x34)
[  159.060000] [<c02247bc>] (jffs2_kill_sb) from [<c01a801c>] (deactivate_locked_super+0x48/0x7c)
[  159.060000] [<c01a801c>] (deactivate_locked_super) from [<c01c1494>] (cleanup_mnt+0x4c/0x70)
[  159.060000] [<c01c1494>] (cleanup_mnt) from [<c01264e0>] (task_work_run+0x7c/0x90)
[  159.060000] [<c01264e0>] (task_work_run) from [<c0104b14>] (do_work_pending+0xa4/0xc0)
[  159.060000] [<c0104b14>] (do_work_pending) from [<c01021b0>] (slow_work_pending+0xc/0x20)
```

shenki, 2016-08-15:
Have not seen this for some time. We have moved away from this tree.

mdmillerii, 2016-08-19:
Actually, taking a look, this occurs when` CONFIG_JFFS2_FS_WRITEBUFFER` is enabled but the flash does not require a write buffer (because its not UBI, NAND, or DATAFLASH, and the NOR was flagged BIT_WRITABLE).  In that case `jffs2_sync_fs` calls `cancel_delayed_work_sync` on the `wbuf_dwork` to clear the write buffer but the work was never setup in `wbuf.c` because `jffs2_flash_setup` in fs.c never called any write buffer allocation.

justinylin, 2018-07-17:
@mdmillerii Hi, same backtrace printed on Linux 3.10.62, we used Nor Flash. I want to figure out, is this an issue for Nor Flash or not? 
If CONFIG_JFFS2_FS_WRITEBUFFER is enabled for JFFS2 on Nor Flash, it seems sync command will always crash and print such backtrace, while store a large file into Nor Flash，some content will stay in Write Buffer and never flushed.

mdmillerii, 2018-07-17:
The issue appears when the write buffer is configured but not required.  I don't remember if its only a lockdep issue or only an issue when lockdep is enabled.  If you only have NOR flash you can turn off the config, otherwise some code paths need to be altered to properly classify the lock (but that is complicated by conditional compilation and unit separation).

justinylin, 2018-10-09:
I see, thanks