openbmc/linux #71: adm1278 driver misses "in2_input" (Vout) sensor

adamliyi, 2016-04-22:
There might be some issue with the upstream adm1278 hwmon driver. I haven't dig into the issue yet.
Some sensors are missing, e.g, "in2_input". This stands for "Vout" register in adm1278. Customer will need this sensor.

Here is the sensors currently available:
```
root@barreleye:~# cat  /sys/class/hwmon/hwmon3/name 
adm1278

root@barreleye:~# cat /sys/class/hwmon/hwmon3/
curr1_highest         in1_label             power1_alarm
curr1_input           in1_max               power1_input
curr1_label           in1_max_alarm         power1_input_highest
curr1_max             in1_min               power1_label
curr1_max_alarm       in1_min_alarm         power1_max
curr1_reset_history   in1_reset_history     power1_reset_history
device/               name                  subsystem/
in1_highest           of_node/              uevent
in1_input             power/                
root@barreleye:~# cat /sys/class/hwmon/hwmon3/in1_input 
11990
root@barreleye:~# cat /sys/class/hwmon/hwmon3/in1_label  
vin
```
We will need 
```
/sys/class/hwmon/hwmon3/in2_input
```
Kernel version:
```
root@barreleye:~# uname -a
Linux barreleye 4.4.6-openbmc-20160329-2 #1 Wed Apr 20 23:06:07 CST 2016 armv5tejl GNU/Linux
root@barreleye:~# cat /proc/version  
Linux version 4.4.6-openbmc-20160329-2 (adam@ubuntu) (gcc version 4.9.3 (GCC) ) #1 Wed Apr 20 23:06:07 CST 2016
```

adamliyi, 2016-05-11:
The bug is fixed by PR: https://github.com/openbmc/linux/pull/80
After fixing:
```
root@barreleye:~# cat /sys/class/hwmon/hwmon3/in2_label 
vout1
root@barreleye:~# cat /sys/class/hwmon/hwmon3/in2_input 
10
```

shenki, 2016-05-11:
This fix is in kernel tag https://github.com/openbmc/linux/releases/tag/openbmc-20160511-1

closed 2016-05-12 by nkskjames