openbmc/linux #74: Kernel panic after AC cycle

Kenthliu, 2016-04-25:
We use 0.6.1 stable pure code to run AC cycle via RMC.
Attachment are the log.
[kernelpanic1.txt.txt](https://github.com/openbmc/linux/files/233770/kernelpanic1.txt.txt)
[ACfail.txt](https://github.com/openbmc/linux/files/233771/ACfail.txt)

shenki, 2016-04-26:
From kernelpanic1.txt.txt:
```
Unable to handle kernel NULL pointer dereference at virtual address 0000001c
pgd = cfbac000 [0000001c] *pgd=4fb25831, *pte=00000000, *ppte=00000000
Internal error: Oops: 17 [#1] ARM
Modules linked in:
CPU: 0 PID: 968 Comm: agetty Not tainted 4.3.6-openbmc-20160222-1 #1
Hardware name: ASpeed SoC
task: cfb1c120 ti: cfb26000 task.ti: cfb26000
PC is at do_select+0x270/0x514
LR is at 0x1
pc : [<c009c220>]    lr : [<00000001>]    psr: a0000013
sp : cfb27b50  ip : 00004000  fp : 00000000
r10: 00000000  r9 : 00000000  r8 : 00000000
r7 : 00000001  r6 : 00000000  r5 : 00000000  r4 : 00000005
r3 : cfbdc200  r2 : cfbdc200  r1 : 00000000  r0 : cfbdc200
Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 0005317f  Table: 4fbac000  DAC: 00000051
Process agetty (pid: 968, stack limit = 0xcfb26190)
Stack: (0xcfb27b50 to 0xcfb28000)
7b40:                                     cfb27e38 00000000 00000000 00000000
7b60: 00000000 cfb26000 00000000 cfb27e5c cfb27e60 cfb27e64 00000011 00000000
7b80: 00000000 cfb27e54 cfb27e58 cfb27e5c 00000000 00000000 00000020 00000011
```

```
Unable to handle kernel NULL pointer dereference at virtual address 00000000
pgd = ce468000
[00000000] *pgd=00000000
Internal error: Oops: 80000005 [#2] ARM
Modules linked in:
CPU: 0 PID: 531 Comm: systemd-journal Tainted: G      D         4.3.6-openbmc-20160222-1 #1
Hardware name: ASpeed SoC
task: cf941440 ti: ce49a000 task.ti: ce49a000
PC is at 0x0
LR is at __wake_up_common+0x48/0x74
pc : [<00000000>]    lr : [<c0034578>]    psr: 80000093
sp : ce49bce8  ip : 00000000  fp : ce4ffc00
r10: c0529e3d  r9 : 00000000  r8 : 00000000
r7 : 00000001  r6 : ce4ffd44  r5 : 00000001  r4 : fffffff4
r3 : 00000004  r2 : 00000000  r1 : 00000001  r0 : cfb27bf8
Flags: Nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 0005317f  Table: 4e468000  DAC: 00000051
Process systemd-journal (pid: 531, stack limit = 0xce49a190)
Stack: (0xce49bce8 to 0xce49c000)
bce0:                   00000000 60000093 cdc4e300 0000000b 00000060 cdd39400
bd00: 0000001b c00345c0 00000004 00000001 c0545354 c01e0ab4 c0545354 000000c2
bd20: 60000093 c01e14cc e8784000 c0545354 00000000 00000000 00000000 c01e1508
```

From ACFail.txt:

```
Unable to handle kernel paging request at virtual address 0a374230
pgd = ce4dc000
[0a374230] *pgd=00000000
Internal error: Oops: 805 [#1] ARM
Modules linked in:
CPU: 0 PID: 527 Comm: systemd-journal Not tainted 4.3.6-openbmc-20160222-1 #1
Hardware name: ASpeed SoC
task: cf93a080 ti: ce476000 task.ti: ce476000
PC is at kernfs_path_locked+0x1c/0x80
LR is at kernfs_path+0x14/0x1c
pc : [<c00da89c>]    lr : [<c00da914>]    psr: a0000093
sp : ce477e68  ip : cdc9e000  fp : 00000001
r10: c053f150  r9 : c053d160  r8 : 0a373231
r7 : 0000002f  r6 : cfba79e8  r5 : cde57760  r4 : 0a374230
r3 : 00000000  r2 : 0a374231  r1 : 0a373231  r0 : cfba79e8
Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 0005317f  Table: 4e4dc000  DAC: 00000051
[... 33 lines cut ...]
Hardware name: ASpeed SoC
task: cf96c840 ti: cdfba000 task.ti: cdfba000
PC is at putname+0x10/0x64
LR is at SyS_renameat2+0x3ac/0x3e4
pc : [<c00986b8>]    lr : [<c0099b6c>]    psr: 60000013
sp : cdfbbf00  ip : c050816c  fp : cfb31000
r10: cfb31000  r9 : 00000000  r8 : 00000000
r7 : fffffffe  r6 : 00000000  r5 : 00000000  r4 : 00000000
r3 : 00000000  r2 : 00000000  r1 : cfb31000  r0 : cfb31000
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 0005317f  Table: 4dfcc000  DAC: 00000051
Process fan_control.py (pid: 1027, stack limit = 0xcdfba190)
Stack: (0xcdfbbf00 to 0xcdfbc000)
bf00: cdfbbf60 cdfbbf38 b6f3d000 00000000 00000800 cdfbbf50 cdfbbf60 ffffff9c
bf20: 000897b0 ffffff9c 000d3310 00000000 00000000 00000000 00000000 00000000
bf40: cfaa31b0 cf4f0330 cfaa31b0 cf4f0330 7f1251bf 0000000f cfb31015 00000000
bf60: 75c9a23b 00000010 cfb31015 00000000 cfaf48e0 00000014 00000000 000897b0
bf80: 00000026 c000a424 cdfba000 00000000 b68f8030 c0099be0 00000000 000d3310
bfa0: 00000000 c000a280 00000014 00000000 000897b0 000d3310 00000000 000d3324
bfc0: 00000014 00000000 000897b0 00000026 b6dd4968 00000001 000d3310 b68f8030
bfe0: b6dd4cb8 be98b87c b6dbfc74 486e037c 80000010 000897b0 868da94b 8e9b0ac4
[<c00986b8>] (putname) from [<cdfbbf38>] (0xcdfbbf38)
Code: e590300c e1a01000 e3530000 ca000000 (e7f001f2) 
---[ end trace a5368d6ae9e2e337 ]---
```

shenki, 2016-04-26:
Milton mentioned on IRC:

```
10:21 < mdmii> I looked at #74 oops in Acfail, it's a pointer to 4k kmalloc of  
               ascii "127\n" in proc cgroup show  
10:23 < mdmii> Followed by a putname assert in another context, which is 0 ref  
               count when freeing a 4k buffer descriptor  
10:23 < mdmii> I didn't look at the other log yet.  
10:25 < mdmii> Happens when reading i2c roms (sysfs i thonk) and then fan   
               control doing raw i2c ioctls  
10:31 < mdmii> Should we suggest some slab debugging and try to recreate?  
10:33 < mdmii> It's a repeatedly ac cycle the box test after a failure to boot  
               was observed  
```

shenki, 2016-04-28:
@Kenthliu We were having a chat about this issue and #77. It would be helpful to attempt to recreate this issue with some kernel debugging turned on.

The options are:
```
CONFIG_UBSAN=y
CONFIG_UBSAN_ALIGNMENT=y
CONFIG_DEBUG_OBJECTS=y
CONFIG_DEBUG_OBJECTS_FREE=y
CONFIG_DEBUG_SLAB=y
CONFIG_DEBUG_SLAB_LEAK=y
CONFIG_DEBUG_KMEMLEAK=y
CONFIG_DEBUG_MEMORY_INIT=y
```

You can grab a patch that enables that from https://github.com/shenki/openbmc/commit/879b5dc318a4cc47b21cb2115d5e30f102fb92a5

shenki, 2016-04-28:
There are also images built here: 

https://openpower.xyz/job/openbmc-build-pull-request/150/distro=ubuntu,target=barreleye/artifact/images/barreleye/

shenki, 2016-06-06:
I think this is the same as https://github.com/openbmc/linux/issues/77

closed 2016-06-06 by shenki