openbmc/linux #75: SCU and GPIO register modify

Kenthliu, 2016-04-25:
Please help to change the below setting since there was a GPIO debouce issue. 
1E6E20A8 bit24 =1
1E780048 bit0 = 1
1E78004C bit0 = 1
1E780058 = 75300h

Kenthliu, 2016-04-25:
One more register. GPIOQ7 need to pull low when power on that make identify LED can be control. 
1E780084 bit7 = 1
1E780080 bit7 = 0

nkskjames, 2016-04-25:
@Kenthliu, you could make the change yourself so you could test as well.  Change goes here:  https://github.com/openbmc/linux/blob/dev-4.4/arch/arm/mach-aspeed/aspeed.c#L128

Kenthliu, 2016-04-26:
Ok sure

Kenthliu, 2016-04-29:
Done.
https://github.com/openbmc/linux/pull/78

shenki, 2016-05-12:
https://lists.ozlabs.org/pipermail/openbmc/2016-May/002853.html

amboar, 2016-05-13:
@Kenthliu I've replied to @shenki 's review email. In summary, I think we need to configure the debounce in the way that you have in #78, but configuring the directions should be done through skeleton's `bin/Barreleye.py` or @anoo1 's LED driver.

Kenthliu, 2016-05-13:
Yes. For GPIOQ7, I can do that. But for debounce issue, we have to change SCU register to enable. Can Barreleye.py do that?

amboar, 2016-05-13:
@Kenthliu personally I'd leave any debounce configuration in the board file. When I've completed the pinctrl driver we can configure it that way. That's a while off though

Kenthliu, 2016-06-06:
Both commit fix this issue. It can be closed.
https://github.com/openbmc/linux/commit/4dbfab2f3c9167e8f40df6a4805b2b4eab673c47
https://github.com/openbmc/skeleton/commit/584747e83666e5661fea8e445956c4f4ff0c9cac

closed 2016-06-06 by Kenthliu