openbmc/linux #77: Bad page map after continue AC cycle

johnhcwang, 2016-04-27:
We use 0.6.1 stable pure code to run AC cycle by power cycle fixture.
This fixture can automatic AC on the system and then AC off after 3 minutes, and AC on again after 1 minutes.
Do this test continually and get the bad page map issue as attachment.
[0426_104918_061_FAIL](https://app.box.com/files/0/f/7654657062/1/f_62929584978)

shenki, 2016-04-27:
Can you please paste the relevant output into the bug. That link requires a sign in

johnhcwang, 2016-04-27:
Seems that github filters some characters and can not show here.
Attachment is partial context.
[0426_104918_061_cap.txt](https://github.com/openbmc/linux/files/237889/0426_104918_061_cap.txt)

shenki, 2016-04-27:
```
system_manager.py[958]: Running System State: BMC_READY
system_manager.py[958]: Starting process: /usr/sbin/discover_system_state.py: restore
BUG: Bad page map in process system_manager.  pte:4a44e100 pmd:4dfc8831
addr:b6a00000 vm_flags:00100077 anon_vma:cde093a0 mapping:  (null) index:b6a00
file:  (null) fault:  (null) mmap:  (null) readpage:  (null)
CPU: 0 PID: 1041 Comm: system_manager. Not tainted 4.3.6-openbmc-20160222-1 #1
Hardware name: ASpeed SoC
[<c000f28c>] (unwind_backtrace) from [<c000cf0c>] (show_stack+0x10/0x14)
[<c000cf0c>] (show_stack) from [<c00787c8>] (print_bad_pte+0x158/0x190)
[<c00787c8>] (print_bad_pte) from [<c0079cec>] (unmap_single_vma+0x4ec/0x500)
[<c0079cec>] (unmap_single_vma) from [<c007a438>] (unmap_vmas+0x44/0x54)
[<c007a438>] (unmap_vmas) from [<c007e97c>] (exit_mmap+0xc8/0x1dc)
[<c007e97c>] (exit_mmap) from [<c0014770>] (mmput+0x38/0xb8)
[<c0014770>] (mmput) from [<c0091c48>] (flush_old_exec+0x464/0x518)
[<c0091c48>] (flush_old_exec) from [<c00c91e4>] (load_elf_binary+0x294/0x101c)
[<c00c91e4>] (load_elf_binary) from [<c00915ac>] (search_binary_handler+0x58/0x160)
[<c00915ac>] (search_binary_handler) from [<c00c8ad0>] (load_script+0x1f0/0x210)
[<c00c8ad0>] (load_script) from [<c00915ac>] (search_binary_handler+0x58/0x160)
[<c00915ac>] (search_binary_handler) from [<c00926f0>] (do_execveat_common+0x3dc/0x588)
[<c00926f0>] (do_execveat_common) from [<c00928c4>] (do_execve+0x28/0x30)
[<c00928c4>] (do_execve) from [<c000a280>] (ret_fast_syscall+0x0/0x38)
Disabling lock debugging due to kernel taint
```

```
BUG: Bad page map in process system_manager.  pte:4a44e100 pmd:4dfc8831
addr:b6a00000 vm_flags:00100077 anon_vma:cde093a0 mapping:  (null) index:b6a00
file:  (null) fault:  (null) mmap:  (null) readpage:  (null)
CPU: 0 PID: 1041 Comm: system_manager. Not tainted 4.3.6-openbmc-20160222-1 #1
Hardware name: ASpeed SoC
[<c000f28c>] (unwind_backtrace) from [<c000cf0c>] (show_stack+0x10/0x14)
[<c000cf0c>] (show_stack) from [<c00787c8>] (print_bad_pte+0x158/0x190)
[<c00787c8>] (print_bad_pte) from [<c0079cec>] (unmap_single_vma+0x4ec/0x500)
[<c0079cec>] (unmap_single_vma) from [<c007a438>] (unmap_vmas+0x44/0x54)
[<c007a438>] (unmap_vmas) from [<c007e97c>] (exit_mmap+0xc8/0x1dc)
[<c007e97c>] (exit_mmap) from [<c0014770>] (mmput+0x38/0xb8)
[<c0014770>] (mmput) from [<c0091c48>] (flush_old_exec+0x464/0x518)
[<c0091c48>] (flush_old_exec) from [<c00c91e4>] (load_elf_binary+0x294/0x101c)
[<c00c91e4>] (load_elf_binary) from [<c00915ac>] (search_binary_handler+0x58/0x160)
[<c00915ac>] (search_binary_handler) from [<c00c8ad0>] (load_script+0x1f0/0x210)
[<c00c8ad0>] (load_script) from [<c00915ac>] (search_binary_handler+0x58/0x160)
[<c00915ac>] (search_binary_handler) from [<c00926f0>] (do_execveat_common+0x3dc/0x588)
[<c00926f0>] (do_execveat_common) from [<c00928c4>] (do_execve+0x28/0x30)
[<c00928c4>] (do_execve) from [<c000a280>] (ret_fast_syscall+0x0/0x38)
Disabling lock debugging due to kernel taint
```

johnhcwang, 2016-04-28:
After running with the image from https://github.com/openbmc/linux/issues/74#issuecomment-215328524
```
system_manager.py[903]: Last System State: ;  Power Policy: RESTORE_LAST_STATE
Slab corruption (Not tainted): kmalloc-256 start=ce813900, len=256
000: 00 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  .kkkkkkkkkkkkkkk
Prev obj: start=ce813800, len=256
000: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
010: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
Next obj: start=ce813a00, len=256
000: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
010: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
kernel[902]: Slab corruption (Not tainted): kmalloc-256 start=ce813900, len=256
kernel[902]: 000: 00 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  .kkkkkkkkkkkkkkk
kernel[902]: Prev obj: start=ce813800, len=256
kernel[902]: 000: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
kernel[902]: 010: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
kernel[902]: Next obj: start=ce813a00, len=256
kernel[902]: 000: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
kernel[902]: 010: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
```

```
system_manager.py[907]: New object: /org/openbmc/control/fans (:1.28)
system_manager.py[907]: Starting fan control
system_manager.py[907]: Setting fans to max
Slab corruption (Not tainted): kmalloc-256 start=cdfae100, len=256
000: 00 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  .kkkkkkkkkkkkkkk
Prev obj: start=cdfae000, len=256
000: 0a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a  .ZZZZZZZZZZZZZZZ
010: 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a  ZZZZZZZZZZZZZZZZ
Next obj: start=cdfae200, len=256
000: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
010: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
kernel[906]: Slab corruption (Not tainted): kmalloc-256 start=cdfae100, len=256
kernel[906]: 000: 00 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  .kkkkkkkkkkkkkkk
kernel[906]: Prev obj: start=cdfae000, len=256
kernel[906]: 000: 0a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a  .ZZZZZZZZZZZZZZZ
kernel[906]: 010: 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a  ZZZZZZZZZZZZZZZZ
kernel[906]: Next obj: start=cdfae200, len=256
kernel[906]: 000: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
kernel[906]: 010: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
system_manager.py[907]: New object: /org/openbmc/control/chassis0 (:1.29)
system_manager.py[907]: All required objects started for BMC_STARTING2
system_manager.py[907]: SystemManager Goto System State: BMC_READY
```

shenki, 2016-04-28:
@johnhcwang can you please attach the output of dmesg as well?

johnhcwang, 2016-04-29:
```
root@barreleye:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.4.6-openbmc-20160329-2 (jenkins@hudson) (gcc version 4.9.3 (GCC) ) #1 Thu Apr 28 06:27:34 UTC 2016
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine model: Barrelye BMC
[    0.000000] Memory policy: Data cache writeback
[    0.000000] SOC Rev: 02010303
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat c053f908, node_mem_map cfdf9000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: console=ttyS4,38400n8 root=/dev/ram rw
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 252596K/262144K available (3650K kernel code, 176K rwdata, 1324K rodata, 224K init, 173K bss, 9548K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xd0800000 - 0xff800000   ( 752 MB)
                   lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc04e3f5c   (4976 kB)
                     .init : 0xc04e4000 - 0xc051c000   ( 224 kB)
                     .data : 0xc051c000 - 0xc0548170   ( 177 kB)
                      .bss : 0xc0548170 - 0xc05735dc   ( 174 kB)
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Aspeed VIC Initiallized
[    0.000000] clocksource: moxart_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 39817925974 ns
[    0.000000] sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
[    0.000000] kmemleak: Kernel memory leak detector disabled
[    0.000000] kmemleak: Early log buffer exceeded (772), please increase DEBUG_KMEMLEAK_EARLY_LOG_SIZE
[    0.070000] Calibrating delay loop... 190.87 BogoMIPS (lpj=954368)
[    0.070000] pid_max: default: 32768 minimum: 301
[    0.070000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070000] CPU: Testing write buffer coherency: ok
[    0.070000] Setting up static identity map for 0x40008400 - 0x40008458
[    0.090000] devtmpfs: initialized
[    0.130000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.140000] NET: Registered protocol family 16
[    0.140000] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.370000] SCSI subsystem initialized
[    0.380000] usbcore: registered new interface driver usbfs
[    0.380000] usbcore: registered new interface driver hub
[    0.380000] usbcore: registered new device driver usb
[    0.390000] clocksource: Switched to clocksource moxart_timer
[    0.440000] NET: Registered protocol family 2
[    0.450000] TCP established hash table entries: 2048 (order: 1, 8192 bytes)
[    0.450000] TCP bind hash table entries: 2048 (order: 1, 8192 bytes)
[    0.450000] TCP: Hash tables configured (established 2048 bind 2048)
[    0.450000] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.450000] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.450000] NET: Registered protocol family 1
[    0.460000] RPC: Registered named UNIX socket transport module.
[    0.460000] RPC: Registered udp transport module.
[    0.460000] RPC: Registered tcp transport module.
[    0.460000] RPC: Registered tcp NFSv4.1 backchannel transport module.
[    0.470000] Trying to unpack rootfs image as initramfs...
[    5.320000] Freeing initrd memory: 1632K (c0800000 - c0998000)
[    5.320000] futex hash table entries: 256 (order: -1, 3072 bytes)
[    5.340000] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    5.340000] jffs2: version 2.2. (NAND) (SUMMARY)  \xffffffc2\xffffffa9 2001-2006 Red Hat, Inc.
[    5.360000] io scheduler noop registered (default)
[... 94 lines cut ...]
[   19.360000] systemd[1]: Mounted Debug File System.
[   19.410000] systemd[1]: Mounted Configuration File System.
[   19.470000] systemd[1]: Mounted Temporary Directory.
[   19.540000] systemd[1]: Started (null).
[   19.590000] systemd[1]: Started Apply Kernel Variables.
[   19.670000] systemd[1]: Started Remount Root and Kernel File Systems.
[   20.260000] systemd[1]: Started Journal Service.
[   23.150000] systemd-journald[531]: Received request to flush runtime journal from PID 1
[   42.850000] ftgmac100: NCSI dev is up
[   63.950000] random: nonblocking pool is initialized
[  133.430000] Slab corruption (Not tainted): kmalloc-256 start=ce812f00, len=256
[  133.440000] 000: 00 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  .kkkkkkkkkkkkkkk
[  133.450000] Prev obj: start=ce812e00, len=256
[  133.460000] 000: 53 74 61 72 74 69 6e 67 20 66 61 6e 20 63 6f 6e  Starting fan con
[... cut ...]
[  147.050000] 000: 00 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  .kkkkkkkkkkkkkkk
[  147.070000] Prev obj: start=cdfcc900, len=256
[  147.080000] 000: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
[  147.090000] 010: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
[  147.100000] Next obj: start=cdfccb00, len=256
[  147.110000] 000: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
[  147.120000] 010: 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  kkkkkkkkkkkkkkkk
root@barreleye:~#
```

Kenthliu, 2016-05-06:
@shenki Any update about this issue? Is there fix in 0.7? Thanks.

mdmillerii, 2016-05-06:
The log shows there was an write to the first byte of a freed object.  The previous and next objects were free in one case and a message that did not run to the end in the other case, so not the likely cause.   I am going to change the code to force on some additional debugging and attempt to recreate on one of our systems.

nkskjames, 2016-05-10:
@mdmillerii did you reproduce this issue?

mdmillerii, 2016-05-10:
I've hit this a few times, its not near 100% maybe 5-10% of the time.  I'm proceeding to figure out which path through the code is turning off the additional debug after the initial lines turn it on, and then will try in earnest to get a hit with the additional debug flags.

mdmillerii, 2016-05-11:
I was able to force slab red zone on the kmalloc caches. This showed phosphor-read-eeprom was causing a buffer freed in kernfs_fop_read to be overrun 1 byte.  Investigation shows 3 calls to that program in Barelleye.py reading eeprom using the at24 driver.  A simple cat failed to reproduce but after looking at strace output the dd command with a buffer size of a kmalloc slab will fail (128, 256, 192 tried, 191 etc ok).  On my system the extra byte was 0xff.
The at24 driver appears to setup a i2c xfer directly to its passed in buffer so apparently it's the i2c driver.  Test base was dev-4.4 from Friday.

Kenthliu, 2016-05-11:
@mdmillerii We use 0.7 and SOCflah to update. The system will hang in booting the kernel sometime, and also found kernel panic.
[ACHang.txt](https://github.com/openbmc/linux/files/258931/ACHang.txt)
[ACHang2.txt](https://github.com/openbmc/linux/files/258932/ACHang2.txt)

mdmillerii, 2016-05-16:
Booted  with `log_buf_len=16M` and after the boot `echo module i2c_aspeed +p > /sys/kernel/debug/dynamic_debug/control` and `dd bs=4 count=1 if=/sys/bus/i2c/devices/0-0050/eeprom of=/dev/null` then `dmesg | grep i2c-0' showed:
    [  670.670000] i2c_aspeed i2c-0: state    [0], SCL    [1], SDA    [1], BUS    [0]
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_do_byte_xfer W: addr 50 start, len 2
    [  670.670000] i2c_aspeed i2c-0: issuing cmd: 3
    [  670.670000] i2c_aspeed i2c-0: irq! status 0x00000001, cmd 0x0c430000
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_master_xfer_done xfer 0S
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_do_byte_xfer W-: addr 50 xfer 0, len 2
    [  670.670000] i2c_aspeed i2c-0: issuing cmd: 2
    [  670.670000] i2c_aspeed i2c-0: irq! status 0x00000001, cmd 0x0c430000
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_master_xfer_done xfer 0 
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_do_byte_xfer W-: addr 50 xfer 1, len 2
    [  670.670000] i2c_aspeed i2c-0: issuing cmd: 2
    [  670.670000] i2c_aspeed i2c-0: irq! status 0x00000001, cmd 0x0c430000
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_master_xfer_done xfer 1 
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_do_byte_xfer R: addr 50 start, len 4
    [  670.670000] i2c_aspeed i2c-0: issuing cmd: 3
    [  670.670000] i2c_aspeed i2c-0: irq! status 0x00000001, cmd 0x0c410000
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_master_xfer_done xfer 0S
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_do_byte_xfer R-: addr 50 xfer 0, len 4
    [  670.670000] i2c_aspeed i2c-0: issuing cmd: 8
    [  670.670000] i2c_aspeed i2c-0: irq! status 0x00000004, cmd 0x0c410000
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_master_xfer_done xfer 0 
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_do_byte_xfer R-: addr 50 xfer 1, len 4
    [  670.670000] i2c_aspeed i2c-0: issuing cmd: 8
    [  670.670000] i2c_aspeed i2c-0: irq! status 0x00000004, cmd 0x0c410000
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_master_xfer_done xfer 1 
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_do_byte_xfer R-: addr 50 xfer 2, len 4
    [  670.670000] i2c_aspeed i2c-0: issuing cmd: 8
    [  670.670000] i2c_aspeed i2c-0: irq! status 0x00000004, cmd 0x0c410000
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_master_xfer_done xfer 2 
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_do_byte_xfer RT: addr 50 xfer 3, len 4
    [  670.670000] i2c_aspeed i2c-0: issuing cmd: 38
    [  670.670000] i2c_aspeed i2c-0: irq! status 0x00000004, cmd 0x135d0000
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_master_xfer_done xfer 3 
    [  670.670000] i2c_aspeed i2c-0: irq! status 0x00000010, cmd 0x0a060000
    [  670.670000] i2c_aspeed i2c-0: ast_i2c_master_xfer_done xfer 4 

So it looks like we get the data rx ack (status 4) for the 4 bytes, then get a normal stop (status 0x10) and call xfer_done a 5th time which if the message is still read will write the 5th byte of the buffer.

mdmillerii, 2016-05-20:
@mdmillerii/linux@e1079d1470b58d822815d170ff729d9480aed43b shows that the overflow occurs on each read.  @shenki can someone who has worked on the driver propose a proper fix?  I don't know what the variables are for in the state machine.

Also, if we ever retry a query_len transaction it looks like we will extend the length each retry.

jk-ozlabs, 2016-05-20:
@mdmillerii : good catch there.

What's unexpected is this:

    [ 670.670000] i2c_aspeed i2c-0: irq! status 0x00000004, cmd 0x135d0000
    [ 670.670000] i2c_aspeed i2c-0: ast_i2c_master_xfer_done xfer 3 
    [ 670.670000] i2c_aspeed i2c-0: irq! status 0x00000010, cmd 0x0a060000

Here, the command register value (`0x135d0000`) tells us that the normal stop and read commands that we issued is complete, so when we mask the pending values out in `ast_i2c_bus_irq`, we end up with zero, and so expect that all is done (and so we increment the read data pointer).

However, we then get another interrupt to tell us that the stop is complete.

I'd say the fix here is to use the IRQ status value to clear bits out of the pending mask (cmd_pending), rather than using the command register value directly.

jk-ozlabs, 2016-05-25:
Working on a fix here.

jk-ozlabs, 2016-05-25:
OK, I think this should fix the problem: https://github.com/jk-ozlabs/linux/commit/a33fa2a2e6bbb668b2e4d5c44644d42e2d116df7

I've push a new branch with that fix on top of the dev-4.4 code, to help with testing: https://github.com/jk-ozlabs/linux/commits/openbmc/i2c-completion-fix

Kenthliu, 2016-05-26:
OK Jeremy. We will try the patch do the AC cycle for weekend. Thanks.

shenki, 2016-05-26:
This looks good @jk-ozlabs, thank you for spending the time on it.

@Kenthliu please report back on how your tests go

Kenthliu, 2016-05-30:
We still see kernel panic during 782 times AC cycle over weekend.
Please take a look to below log.
[accycle.txt](https://github.com/openbmc/linux/files/288912/accycle.txt)

jk-ozlabs, 2016-05-30:
OK, we'll need to keep investigating here.

amboar, 2016-05-31:
Here's a mostly scripted extraction of interesting boots in @Kenthliu's accycle.txt. It probably hasn't caught all the failures. boot-003.log looked boring enough to use as a reference. Scripts for extraction are included.

[interesting-logs.zip](https://github.com/openbmc/linux/files/290203/interesting-logs.zip)

mdmillerii, 2016-06-01:
Thanks @amboar.   I looked through the logs.  Many of them happen earlier than the previous round of oopses.  Since I do not see the extra messages from the u-boot patch in the 0.6.1 branch, so  I need to ask what version of `u-boot` was running?   If it was from the 0.8.1 openbmc build it is missing the patches to stop the network, which was just merged via openbmc/openbmc#304 on Thu May 26 09:43:33 2016 -0500 openbmc@265b0d55113d364554102d532177916741c8c379 (Looking at boot-001.log it looks like a new chip was programmed but the u-boot said it was using the hardware mac, which I don't think it should do if it was the new code). 

Here's my initial classification:
* The following logs the initial error was during mount of `rofs`: `018` `071` `120` `217` `253` `271` `315` `471` `490` `543` `581` `636` `657` `741`.
* The following died before user space at phy init: `374` `598` `712`
* The remaining: `175` `433` `435`.

mdmillerii, 2016-06-02:
@Kenthliu please see instructions in previous comment.

Kenthliu, 2016-06-02:
Hi @mdmillerii Not sure what I need to do. Let me summarize, what the patch or commit miss in 0.8.1? I have to build another test version with below config, right?
CONFIG_UBSAN=y
CONFIG_UBSAN_ALIGNMENT=y
CONFIG_DEBUG_OBJECTS=y
CONFIG_DEBUG_OBJECTS_FREE=y
CONFIG_DEBUG_SLAB=y
CONFIG_DEBUG_SLAB_LEAK=y
CONFIG_DEBUG_KMEMLEAK=y
CONFIG_DEBUG_MEMORY_INIT=y

Kenthliu, 2016-06-02:
0.8.1 Uboot version https://github.com/openbmc/openbmc/blob/v0.8.1/meta-phosphor/common/recipes-bsp/u-boot/u-boot-fw-utils_2013.07.bb
Under this commit
https://github.com/openbmc/u-boot/commit/4b44678c11ac82c7c797bb115e276181752ad54d

mdmillerii, 2016-06-02:
@Kenthliu please retest with the current `u-boot` version merged in openbmc/opennmc#304 and the kernel from @jk-ozlabs.  If you still see issues then add the configs you mentioned and post the log.  Note: you can configure with the debug configs if you want to save a step but if it's clean then you will need to retest without them.

Kenthliu, 2016-06-03:
I will add all the patches and retest this weekend.
This is my repository brach.
https://github.com/Kenthliu/linux/commits/dev-4.4

mdmillerii, 2016-06-03:
Since e1079d1470b58d822815d170ff729d9480aed43b emits a warn it's okay to include it but it should not be needed.  Are you including the debug configs?  You will use the new `u-boot` too, right?

Kenthliu, 2016-06-03:
Yes I include the debug config in my local branch. It should use the latest uboot, but I don't know how to tell the version.

Kenthliu, 2016-06-06:
We ran AC cycle about 423 times over night. So far seems good.

shenki, 2016-06-06:
Thanks. I've merged @jk-ozlabs's patch into our kernel trees, and created a new kernel tag for inclusion

 https://github.com/openbmc/openbmc/pull/313

closed 2016-06-06 by shenki