openbmc/linux #87: ftgmac100 can't be instantiated twice

shenki, 2016-05-25:
On the ast2500-evb we have two Ethernet devices in the device tree:

```
[    1.030000] ftgmac100 1e660000.ethernet: Generated random MAC address 8a:26:9d:18:62:09
[    1.460000] libphy: ftgmac100_mdio: probed
[    1.550000] ftgmac100 1e680000.ethernet: Generated random MAC address 1a:db:8d:7d:1b:31
[    1.550000] ------------[ cut here ]------------
[    1.550000] WARNING: CPU: 0 PID: 1 at fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
[    1.550000] sysfs: cannot create duplicate filename '/devices/virtual/mdio_bus/ftgmac100_mii'
[    1.550000] Modules linked in:
[    1.550000] CPU: 0 PID: 1 Comm: swapper Not tainted 4.6.0-10263-g41e19f2632d9 #502
[    1.550000] Hardware name: ASpeed SoC
[    1.550000] [<c000f800>] (unwind_backtrace) from [<c000d360>] (show_stack+0x10/0x14)
[    1.550000] [<c000d360>] (show_stack) from [<c0017dac>] (__warn+0xc8/0xf8)
[    1.550000] [<c0017dac>] (__warn) from [<c0017e20>] (warn_slowpath_fmt+0x44/0x6c)
[    1.550000] [<c0017e20>] (warn_slowpath_fmt) from [<c0101b7c>] (sysfs_warn_dup+0x50/0x70)
[    1.550000] [<c0101b7c>] (sysfs_warn_dup) from [<c0101c58>] (sysfs_create_dir_ns+0x74/0x88)
[    1.550000] [<c0101c58>] (sysfs_create_dir_ns) from [<c0150a38>] (kobject_add_internal+0x124/0x300)
[    1.550000] [<c0150a38>] (kobject_add_internal) from [<c0150c90>] (kobject_add+0x7c/0xb4)
[    1.550000] [<c0150c90>] (kobject_add) from [<c01913c8>] (device_add+0x118/0x554)
[    1.550000] [<c01913c8>] (device_add) from [<c01b8b70>] (__mdiobus_register+0x84/0x194)
[    1.550000] [<c01b8b70>] (__mdiobus_register) from [<c01ba8fc>] (ftgmac100_probe+0x360/0x540)
[    1.550000] [<c01ba8fc>] (ftgmac100_probe) from [<c0195e54>] (platform_drv_probe+0x50/0xa0)
[    1.550000] [<c0195e54>] (platform_drv_probe) from [<c0194284>] (driver_probe_device+0x1c0/0x408)
[    1.550000] [<c0194284>] (driver_probe_device) from [<c019459c>] (__driver_attach+0xd0/0x110)
[    1.550000] [<c019459c>] (__driver_attach) from [<c0192548>] (bus_for_each_dev+0x78/0xb4)
[    1.550000] [<c0192548>] (bus_for_each_dev) from [<c01936a0>] (bus_add_driver+0x110/0x23c)
[    1.550000] [<c01936a0>] (bus_add_driver) from [<c0194eac>] (driver_register+0x9c/0xe0)
[    1.550000] [<c0194eac>] (driver_register) from [<c0009818>] (do_one_initcall+0xb4/0x178)
[    1.550000] [<c0009818>] (do_one_initcall) from [<c0393db4>] (kernel_init_freeable+0x100/0x1bc)
[    1.550000] [<c0393db4>] (kernel_init_freeable) from [<c0291450>] (kernel_init+0x8/0xec)
[    1.550000] [<c0291450>] (kernel_init) from [<c000a298>] (ret_from_fork+0x14/0x3c)
[    1.550000] ---[ end trace 9d18a3c44f80aa8b ]---
[    1.550000] ------------[ cut here ]------------
[    1.550000] WARNING: CPU: 0 PID: 1 at lib/kobject.c:240 kobject_add_internal+0x298/0x300
[    1.550000] kobject_add_internal failed for ftgmac100_mii with -EEXIST, don't try to register things with the same name in the same directory.
[    1.550000] Modules linked in:
[    1.550000] CPU: 0 PID: 1 Comm: swapper Tainted: G        W       4.6.0-10263-g41e19f2632d9 #502
[    1.550000] Hardware name: ASpeed SoC
[    1.550000] [<c000f800>] (unwind_backtrace) from [<c000d360>] (show_stack+0x10/0x14)
[    1.550000] [<c000d360>] (show_stack) from [<c0017dac>] (__warn+0xc8/0xf8)
[    1.550000] [<c0017dac>] (__warn) from [<c0017e20>] (warn_slowpath_fmt+0x44/0x6c)
[    1.550000] [<c0017e20>] (warn_slowpath_fmt) from [<c0150bac>] (kobject_add_internal+0x298/0x300)
[    1.550000] [<c0150bac>] (kobject_add_internal) from [<c0150c90>] (kobject_add+0x7c/0xb4)
[    1.550000] [<c0150c90>] (kobject_add) from [<c01913c8>] (device_add+0x118/0x554)
[    1.550000] [<c01913c8>] (device_add) from [<c01b8b70>] (__mdiobus_register+0x84/0x194)
[    1.570000] [<c01b8b70>] (__mdiobus_register) from [<c01ba8fc>] (ftgmac100_probe+0x360/0x540)
[    1.570000] [<c01ba8fc>] (ftgmac100_probe) from [<c0195e54>] (platform_drv_probe+0x50/0xa0)
[    1.570000] [<c0195e54>] (platform_drv_probe) from [<c0194284>] (driver_probe_device+0x1c0/0x408)
[    1.570000] [<c0194284>] (driver_probe_device) from [<c019459c>] (__driver_attach+0xd0/0x110)
[    1.570000] [<c019459c>] (__driver_attach) from [<c0192548>] (bus_for_each_dev+0x78/0xb4)
[    1.570000] [<c0192548>] (bus_for_each_dev) from [<c01936a0>] (bus_add_driver+0x110/0x23c)
[    1.570000] [<c01936a0>] (bus_add_driver) from [<c0194eac>] (driver_register+0x9c/0xe0)
[    1.570000] [<c0194eac>] (driver_register) from [<c0009818>] (do_one_initcall+0xb4/0x178)
[    1.570000] [<c0009818>] (do_one_initcall) from [<c0393db4>] (kernel_init_freeable+0x100/0x1bc)
[    1.570000] [<c0393db4>] (kernel_init_freeable) from [<c0291450>] (kernel_init+0x8/0xec)
[    1.570000] [<c0291450>] (kernel_init) from [<c000a298>] (ret_from_fork+0x14/0x3c)
[    1.570000] ---[ end trace 9d18a3c44f80aa8c ]---
[    1.570000] libphy: mii_bus ftgmac100_mii failed to register
[    1.570000] ftgmac100 1e680000.ethernet: Cannot register MDIO bus!
[    1.570000] ftgmac100 1e680000.ethernet: Error -22 setting up MDIO
[    1.570000] kmemleak: Cannot insert 0xdf665c00 into the object search tree (overlaps existing)
[    1.570000] CPU: 0 PID: 1 Comm: swapper Tainted: G        W       4.6.0-10263-g41e19f2632d9 #502
[    1.570000] Hardware name: ASpeed SoC
[    1.570000] [<c000f800>] (unwind_backtrace) from [<c000d360>] (show_stack+0x10/0x14)
[    1.570000] [<c000d360>] (show_stack) from [<c00af374>] (create_object+0x178/0x21c)
[    1.570000] [<c00af374>] (create_object) from [<c00ac5f0>] (__kmalloc+0xe0/0xf4)
[    1.570000] [<c00ac5f0>] (__kmalloc) from [<c00fc554>] (__register_sysctl_table+0x44/0x45c)
[    1.570000] [<c00fc554>] (__register_sysctl_table) from [<c024f4d8>] (__devinet_sysctl_register+0x94/0xe8)
[    1.570000] [<c024f4d8>] (__devinet_sysctl_register) from [<c024f6f0>] (devinet_sysctl_register+0x64/0x8c)
[    1.570000] [<c024f6f0>] (devinet_sysctl_register) from [<c024f7e4>] (inetdev_init+0xcc/0x15c)
[    1.570000] [<c024f7e4>] (inetdev_init) from [<c0250390>] (inetdev_event+0x5c/0x4e0)
[    1.570000] [<c0250390>] (inetdev_event) from [<c0031564>] (notifier_call_chain+0x40/0x68)
[    1.570000] [<c0031564>] (notifier_call_chain) from [<c00315e4>] (raw_notifier_call_chain+0x14/0x1c)
[    1.570000] [<c00315e4>] (raw_notifier_call_chain) from [<c01f07c4>] (call_netdevice_notifiers+0x18/0x38)
[    1.570000] [<c01f07c4>] (call_netdevice_notifiers) from [<c01f8690>] (register_netdevice+0x4b4/0x57c)
[    1.570000] [<c01f8690>] (register_netdevice) from [<c01f876c>] (register_netdev+0x14/0x24)
[    1.570000] [<c01f876c>] (register_netdev) from [<c01ba9bc>] (ftgmac100_probe+0x420/0x540)
[    1.570000] [<c01ba9bc>] (ftgmac100_probe) from [<c0195e54>] (platform_drv_probe+0x50/0xa0)
[    1.570000] [<c0195e54>] (platform_drv_probe) from [<c0194284>] (driver_probe_device+0x1c0/0x408)
[    1.570000] [<c0194284>] (driver_probe_device) from [<c019459c>] (__driver_attach+0xd0/0x110)
[    1.570000] [<c019459c>] (__driver_attach) from [<c0192548>] (bus_for_each_dev+0x78/0xb4)
[    1.570000] [<c0192548>] (bus_for_each_dev) from [<c01936a0>] (bus_add_driver+0x110/0x23c)
[    1.570000] [<c01936a0>] (bus_add_driver) from [<c0194eac>] (driver_register+0x9c/0xe0)
[    1.570000] [<c0194eac>] (driver_register) from [<c0009818>] (do_one_initcall+0xb4/0x178)
[    1.570000] [<c0009818>] (do_one_initcall) from [<c0393db4>] (kernel_init_freeable+0x100/0x1bc)
[    1.570000] [<c0393db4>] (kernel_init_freeable) from [<c0291450>] (kernel_init+0x8/0xec)
[    1.570000] [<c0291450>] (kernel_init) from [<c000a298>] (ret_from_fork+0x14/0x3c)
[    1.570000] kmemleak: Kernel memory leak detector disabled
[    1.570000] kmemleak: Object 0xdf665c00 (size 1024):
[    1.570000] kmemleak:   comm "swapper", pid 1, jiffies 4294937453
[    1.570000] kmemleak:   min_count = 1
[    1.570000] kmemleak:   count = 0
[    1.570000] kmemleak:   flags = 0x1
[    1.570000] kmemleak:   checksum = 0
[    1.570000] kmemleak:   backtrace:
[    1.570000]      [<c0091f00>] kmemdup+0x1c/0x38
[    1.570000]      [<c0200a74>] neigh_sysctl_register+0x30/0x180
[    1.570000]      [<c024f6d8>] devinet_sysctl_register+0x4c/0x8c
[    1.570000]      [<c024f7e4>] inetdev_init+0xcc/0x15c
[    1.570000]      [<c0250390>] inetdev_event+0x5c/0x4e0
[    1.570000]      [<c0031564>] notifier_call_chain+0x40/0x68
[    1.570000]      [<c00315e4>] raw_notifier_call_chain+0x14/0x1c
[    1.570000]      [<c01f07c4>] call_netdevice_notifiers+0x18/0x38
[    1.570000]      [<c01f8690>] register_netdevice+0x4b4/0x57c
[    1.570000]      [<c01f876c>] register_netdev+0x14/0x24
[    1.570000]      [<c01ba9bc>] ftgmac100_probe+0x420/0x540
[    1.570000]      [<c0195e54>] platform_drv_probe+0x50/0xa0
[    1.570000]      [<c0194284>] driver_probe_device+0x1c0/0x408
[    1.570000]      [<c019459c>] __driver_attach+0xd0/0x110
[    1.570000]      [<c0192548>] bus_for_each_dev+0x78/0xb4
[    1.570000]      [<c01936a0>] bus_add_driver+0x110/0x23c
```

ya-mouse, 2016-07-25:
I've made a duty patch to fix this issue:
https://github.com/ya-mouse/openbmc-target/blob/master/aspeed/patches-4.4/0060-openmouse-aspeed-fix.patch

shenki, 2017-05-18:
Fixed in 4.10.

closed 2017-05-18 by shenki