openbmc/linux #95: [Barreleye] ftgmac100 1e660000.ethernet eth0

anoo1, 2016-06-24:
@Kenthliu reported the following issue with tag openbmc tag v1.0:

Test AC cycle fails:
We found a lot of "kernel[]: ftgmac100 1e660000.ethernet eth0: rx crc err" and "kernel []: ftgmac100 1e660000.ethernet eth0: rx frame too long" messages in BMC console.
 
Here is the AC cycle step
 
1. AC on
2. BMC power on and enter to host
3. There is a script in Ubuntu system. The script will send command to RMC via LAN.
4. RMC make system AC off, then AC on.
 
We found the system stop at step 3.
At that situation, BMC is still alive, but showed many ftgmac error message. 
So I suspect the LAN has some problems.
[accycle log.txt](https://github.com/openbmc/linux/files/331840/accycle.log.txt)

Attachment is the console and journalctl log.

shenki, 2016-06-25:
What is RMC?

Have you looked at packets on the network using wireshark or similar?

The warnings in the logs are due to the hardware detecting

shenki, 2016-08-19:
Has this been seen since June?

shenki, 2016-08-30:
Please reopen if you can reproduce this issue

closed 2016-08-30 by shenki