openbmc/linux #98: LED sub-node property "default-state" is not working

johnhcwang, 2016-08-04:
Currently the beep LED turns to OFF during kernel loading.
I tried to keep beep LED as solid ON and plan to turn it off by skeleton service once it is available.
Depend on the [LED sub-node properties descriptions](https://www.kernel.org/doc/Documentation/devicetree/bindings/leds/leds-gpio.txt), it can be accomplished by **default-state = "on"**, but it is not working. The LED will be turned to OFF during **create_gpio_led** function.
Here are my rough debug and seems that the value didn't be success write into the gpio.
Sorry that I have no idea how trace **chip->direction_output** for further debug.
Is there any known issue or fix from new kernel regarding to the **gpiod_direction_output** function?

My changes:
````
diff --git a/arch/arm/boot/dts/aspeed-bmc-opp-barreleye.dts b/arch/arm/boot/dts/aspeed-bmc-opp-barreleye.dts
index 0e567e8..5d9e060 100644
--- a/arch/arm/boot/dts/aspeed-bmc-opp-barreleye.dts
+++ b/arch/arm/boot/dts/aspeed-bmc-opp-barreleye.dts
@@ -138,6 +138,7 @@
                };
                beep {
                        gpios = <&gpio 111 GPIO_ACTIVE_HIGH>;
+                       default-state = "on";
                };
        };
 };
diff --git a/drivers/gpio/gpiolib.c b/drivers/gpio/gpiolib.c
index 4e4c308..66c39d1 100644
--- a/drivers/gpio/gpiolib.c
+++ b/drivers/gpio/gpiolib.c
@@ -1154,9 +1154,11 @@ static int _gpiod_direction_output_raw(struct gpio_desc *desc, int value)
                return -EIO;
        }
 
+printk("\t%s\twrite value %d to gpio %d\n",__func__,value,gpio_chip_hwgpio(desc));
        status = chip->direction_output(chip, gpio_chip_hwgpio(desc), value);
        if (status == 0)
                set_bit(FLAG_IS_OUT, &desc->flags);
+printk("\t%s\tread value from gpio %d = %d\n",__func__,gpio_chip_hwgpio(desc),gpiod_get_value(desc));
        trace_gpio_value(desc_to_gpio(desc), 0, value);
        trace_gpio_direction(desc_to_gpio(desc), 0, status);
        return status;
diff --git a/drivers/leds/leds-gpio.c b/drivers/leds/leds-gpio.c
index 5db4515..65e29b4 100644
--- a/drivers/leds/leds-gpio.c
+++ b/drivers/leds/leds-gpio.c
@@ -139,9 +140,12 @@ static int create_gpio_led(const struct gpio_led *template,
        if (!template->retain_state_suspended)
                led_dat->cdev.flags |= LED_CORE_SUSPENDRESUME;
 
+printk("%s\t%s, %s (%s,line%d)\n",__func__,led_dat->cdev.name, (gpiod_is_active_low(led_dat->gpiod)==1)?"ACTIVE_LOW":"ACTIVE_HIGH",__FILE__,__LI
+printk("%s\tset default state = %d,orig direction/value = %s/%d\n",__func__,state,(gpiod_get_direction(led_dat->gpiod)==1)?"Input":"Output",gpio
        ret = gpiod_direction_output(led_dat->gpiod, state);
        if (ret < 0)
                return ret;
+printk("%s\tcurrent direction/value after set = %s/%d\n",__func__,(gpiod_get_direction(led_dat->gpiod)==1)?"Input":"Output",gpiod_get_value(led_
 
        INIT_WORK(&led_dat->work, gpio_led_work);
````
````
create_gpio_led beep, ACTIVE_HIGH (/home/openbmc/build/tmp/work-shared/barreleye/kernel-source/drivers/leds/leds-gpio.c,line143)
create_gpio_led set default state = 1,orig direction/value = Output/1
        _gpiod_direction_output_raw     write value 1 to gpio 111
        _gpiod_direction_output_raw     read value from gpio 111 = 0
create_gpio_led current direction/value after set = Output/0
````

amboar, 2016-08-04:
@johnhcwang what kernel are you running? Perhaps adding some instrumentation to `drivers/gpio/gpio-aspeed.c` would provide more insight, specifically `aspeed_gpio_dir_out()`.

Also, can you provide the output for `devmem 0x1e6e2088` and `devmem 0x1e6e2090`?

johnhcwang, 2016-08-05:
The kernel I'm running is 4.4 and the aspeed_gpio_dir_out() only configure the GPIO direction.
The variable _val_ is unused.

The multi-function setting should be OK, the pin is GPIO function.
````
devmem 0x1e6e2088
0x01C00000
root@barreleye:~# devmem 0x1e6e2090
0x003FA008
````

amboar, 2016-08-05:
@johnhcwang right! Can you apply 5a41f2aa5f2573874f4098a300bebabf5758de20 and give it a test?

johnhcwang, 2016-08-05:
@amboar ,cool! It's working. Thanks.

shenki, 2016-08-19:
This has been merged into the dev-4.4 tree as aa65fa6cf3c3a558e009fb2c934b85974c2733db. It is part of the tag openbmc-4.4-20160819-1.

closed 2016-08-19 by shenki