openbmc/linux #100: aspeed-smc driver support for AST2500 SPI flash controller

GHF, 2016-08-23:
We need to support the register layout for the AST2500 SPI flash controller, selected by compatible = "aspeed,ast2500-smc";

mdmillerii, 2016-08-23:
Yes we do ... its on my list ...

For the time being selecting the 2400-fmc will provide basic support (for all 3 copies).

legoater, 2016-10-14:
Just pushed a tree with the support for AST2500 : 

https://github.com/legoater/linux/commits/openbmc-4.7

mdmillerii, 2016-10-17:
For others watching, the mdmillerii/linux@023a77be1712912a59db90cc6b949cd1978e38d3 in the spi-nor-4.7-wip branch has had support for the ast2500 variants and the spi-nor-4.8-wip branch has support on an early 4.8 base.   The merge process into openbmc/linux is taking additional time.

shenki, 2016-10-17:
Patch has been posted to the openbmc list by @legoater and a first review has been completed by myself.

Following review, this patch has been sent upstream for review:

 http://lists.infradead.org/pipermail/linux-mtd/2016-October/069726.html

legoater, 2017-02-13:
I think this issue can be closed now.

shenki, 2017-02-13:
Congratulations on your hard work in getting the driver upstream in 4.11 @legoater. Great work.

closed 2017-02-13 by shenki