openbmc/linux #103: cgroup_pidlist_start: Unable to handle kernel paging request at virtual address 1200544e

amboar, 2016-08-26:
Same kernel/qemu configuration as #101, but different backtrace:
```
Unable to handle kernel paging request at virtual address 1200544e
pgd = c8c3c000
[1200544e] *pgd=00000000
Internal error: Oops: 5 [#1] ARM
Modules linked in:
CPU: 0 PID: 1 Comm: systemd Not tainted 4.7.1-00053-g6d93f625315a #581
Hardware name: ASpeed SoC
task: df443460 ti: df444000 task.ti: df444000
PC is at cgroup_pidlist_start+0xc0/0x44c
LR is at cgroup_pidlist_start+0xa4/0x44c
pc : [<c0166868>]    lr : [<c016684c>]    psr: 00000093
sp : df445e18  ip : 00000000  fp : 00000000
r10: 00000002  r9 : df445f78  r8 : 00000000
r7 : c8f5d1a0  r6 : c8ddc0e0  r5 : c0702028  r4 : c8d6ee00
r3 : 1200544a  r2 : c8d6eea0  r1 : 12005452  r0 : c070d058
Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 00093177  Table: 48c3c000  DAC: 00000051
Process systemd (pid: 1, stack limit = 0xdf444190)
Stack: (0xdf445e18 to 0xdf446000)
5e00:                                                       00000001 ffffe000
5e20: 00000000 60000013 df445ea8 c0702028 df445f78 c8cacea0 df445ea8 c013628c
5e40: 00000001 00000000 00000000 c020abc0 00000000 00000000 ffffe000 1aadfd66
5e60: 60000013 df445ea8 c8ddc0e0 c8f5d1a0 00000000 810fa130 df445f78 c8cacea0
5e80: df445ea8 c020abe8 c8ddc0e0 00001000 00000000 c01d38dc 00000001 c8ddc110
5ea0: c0702028 c0702028 00000000 00000000 df444000 00000000 bec887ec 1aadfd66
5ec0: c8cacea0 c020a9b4 c8cacea0 c0702028 df445f78 810fa130 00001000 00000000
5ee0: bec887ec c01b1fd0 00000000 00000000 00000000 00000000 00001000 00000000
5f00: 00000000 00000000 57bfbcd2 1443fd00 57bfbcd2 1443fd00 57bfbcd2 1443fd00
5f20: 0000005f 00000000 00000000 c8cacea0 df445f78 1aadfd66 00001000 c8cacea0
5f40: 810fa130 df445f78 810fa130 c01b2c1c c8cacea0 810fa130 00001000 c8cacea0
5f60: c0702028 c0702028 c8cacea0 810fa130 00001000 c01b38d0 00000000 00000000
5f80: 00001000 1aadfd66 00000000 8109ac70 8109ac70 00000000 00000003 c01025a4
5fa0: df444000 c0102400 8109ac70 8109ac70 00000010 810fa130 00001000 00000040
5fc0: 8109ac70 8109ac70 00000000 00000003 7f74c8a7 00000000 b6dec020 bec887ec
5fe0: 00000000 bec881c0 b6d8e0b4 b6d700b4 80000010 00000010 00000000 00000000
[<c0166868>] (cgroup_pidlist_start) from [<c020abe8>] (kernfs_seq_start+0x4c/0x8c)
[<c020abe8>] (kernfs_seq_start) from [<c01d38dc>] (seq_read+0x194/0x428)
[<c01d38dc>] (seq_read) from [<c01b1fd0>] (__vfs_read+0x28/0x128)
[<c01b1fd0>] (__vfs_read) from [<c01b2c1c>] (vfs_read+0x90/0xfc)
[<c01b2c1c>] (vfs_read) from [<c01b38d0>] (SyS_read+0x50/0xa0)
[<c01b38d0>] (SyS_read) from [<c0102400>] (ret_fast_syscall+0x0/0x38)
Code: e2433008 e2831008 e1510002 0a000005 (e5931004) 
---[ end trace 05cbde247dfcab60 ]---
```

amboar, 2016-08-26:
Immediately followed by:

```
BUG: spinlock lockup suspected on CPU#0, systemd-cgroups/703
 lock: css_set_lock+0x0/0x20, .magic: dead4ead, .owner: systemd/1, .owner_cpu: 0
CPU: 0 PID: 703 Comm: systemd-cgroups Tainted: G      D         4.7.1-00053-g6d93f625315a #581
Hardware name: ASpeed SoC
[<c01077e8>] (unwind_backtrace) from [<c01053d0>] (show_stack+0x10/0x14)
[<c01053d0>] (show_stack) from [<c0138650>] (do_raw_spin_lock+0xd8/0x118)
[<c0138650>] (do_raw_spin_lock) from [<c01671ec>] (cgroup_exit+0x28/0x6c)
[<c01671ec>] (cgroup_exit) from [<c0110458>] (do_exit+0x3b8/0x848)
[<c0110458>] (do_exit) from [<c01118f8>] (do_group_exit+0x48/0xbc)
[<c01118f8>] (do_group_exit) from [<c011197c>] (SyS_exit_group+0x10/0x14)
[<c011197c>] (SyS_exit_group) from [<c0102400>] (ret_fast_syscall+0x0/0x38)
Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b

---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
```

amboar, 2017-03-14:
Just saw this on Witherspoon:

```
ucd9000: probe of 11-0064 failed with error -74
Alignment trap: not handling instruction e1902f9f at [<8043baf4>]
Unhandled fault: alignment exception (0x001) at 0x9e7c349f
pgd = 9d000000
[9e7c349f] *pgd=9e60041e(bad)
Internal error: : 1 [#1] ARM
CPU: 0 PID: 1 Comm: systemd Not tainted 4.7.10-00295-g1f70c8c051a8-dirty #548
Hardware name: ASpeed SoC
task: 9e4388a0 ti: 9e43a000 task.ti: 9e43a000
PC is at mutex_lock+0x8/0x24
LR is at cgroup_pidlist_start+0x4c/0x39c
pc : [<8043baf8>]    lr : [<801548a4>]    psr: 60000013
sp : 9e43be18  ip : 00000001  fp : 00000000
r10: 9d76c320  r9 : 00000000  r8 : 9d9914c0
r7 : 00000000  r6 : 9d9cca40  r5 : 9e7c33ff  r4 : 8070200c
r3 : 9e43bea8  r2 : 9e7a6d70  r1 : 9e43bea8  r0 : 9e7c349f
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 00c5387d  Table: 9d000008  DAC: 00000051
Process systemd (pid: 1, stack limit = 0x9e43a188)
Stack: (0x9e43be18 to 0x9e43c000)
be00:                                                       00000018 9d76c320
be20: 9d0ae000 9e43bea8 9e43bec0 801ac004 9e43bec0 00000000 00000000 801ad250
be40: 9d553000 9d494d20 8070bf30 00000000 9e43bf70 00000000 00020000 61fea8ae
be60: 00000000 9e43bea8 9d9cca40 9d9914c0 00000000 5543d828 9e43bf78 9d76c320
be80: 9e43bea8 801f6ea4 9d9cca40 00001000 00000000 801bfaf0 9e4046a0 9d9cca70
bea0: 8070200c 7eec3f78 00000000 00000000 9e43a000 00000000 7eec462c 61fea8ae
bec0: 9d76c320 801f6c84 00001000 8070200c 5543d828 9e43bf78 00001000 00000000
bee0: 7eec462c 801a2884 00000000 00000000 00000000 00000000 00001000 00000000
bf00: 00000000 00000000 58ad2cfa 160dc104 58ad2cfa 160dc104 58ad2cfa 160dc104
bf20: 00000117 00000000 00000000 9d76c320 9e43bf78 61fea8ae 9d76c320 00001000
bf40: 00000000 5543d828 9e43bf78 801a2a14 9d76c320 5543d828 00001000 9d76c320
bf60: 54c0f912 8070200c 9d76c320 5543d828 00001000 801a2df8 00000000 00000000
bf80: 00001000 61fea8ae 00000000 553bffb0 54c0f912 00000075 00000003 80102384
bfa0: 9e43a000 801021c0 553bffb0 54c0f912 00000048 5543d828 00001000 00000040
bfc0: 553bffb0 54c0f912 00000075 00000003 54c0f913 00000000 76e6c020 7eec462c
bfe0: 00000000 7eec3ffc 76d9a398 76df104c 60000010 00000048 e22ef1db 6b45edfe
[<8043baf8>] (mutex_lock) from [<801548a4>] (cgroup_pidlist_start+0x4c/0x39c)
[<801548a4>] (cgroup_pidlist_start) from [<801f6ea4>] (kernfs_seq_start+0x4c/0x80)
[<801f6ea4>] (kernfs_seq_start) from [<801bfaf0>] (seq_read+0x104/0x424)
[<801bfaf0>] (seq_read) from [<801a2884>] (__vfs_read+0x28/0x128)
[<801a2884>] (__vfs_read) from [<801a2a14>] (vfs_read+0x90/0xfc)
[<801a2a14>] (vfs_read) from [<801a2df8>] (SyS_read+0x4c/0x9c)
[<801a2df8>] (SyS_read) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
Code: e8bd83f0 8070200c f5d0f000 e1902f9f (e2422001) 
---[ end trace f354de3abd1d87d0 ]---
-sh: echo: write error: No such device
-sh: echo: write error: No such device
ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b

Rebooting in 1 seconds..

U-Boot 2016.07 (Feb 22 2017 - 17:20:20 +0000)
```

legoater, 2017-07-07:
as this issue occurred on a Witherspoon system just after the  ucd9000 was probed, can we consider it was fixed in the recent kernel changes ?