openbmc/linux #108: Kernel panic when running qemu palmetto-bmc machine

mjturek, 2016-10-28:
I'm hitting a kernel panic when running the qemu palmetto-bmc machine.

I'm building qemu from the upstream openbmc/qemu tree, using steps described [here](https://github.com/openbmc/docs/blob/master/cheatsheet.md#building-qemu).

```
$ git clone https://github.com/openbmc/qemu.git
$ cd qemu
$ git submodule update --init dtc
$ mkdir build
$ cd build
$ ../configure --target-list=arm-softmmu
$ make
```

Version looks correct for the newly built QEMU
```
ubuntu@mjturek-openbmc:~$ qemu/build/arm-softmmu/qemu-system-arm -version
QEMU emulator version 2.6.90 (v2.5.0-4924-g8b4e141-dirty), Copyright (c) 2003-2008 Fabrice Bellard
```

I build the image with the steps described [here](https://github.com/openbmc/docs/blob/master/cheatsheet.md#building-for-palmetto).
```
$ git clone https://github.com/openbmc/openbmc.git
$ cd openbmc
$ TEMPLATECONF=meta-openbmc-machines/meta-openpower/meta-ibm/meta-palmetto/conf . oe-init-build-env
$ bitbake obmc-phosphor-image
```
This creates the image `openbmc/build/tmp/deploy/images/palmetto/flash-palmetto-<datetime>`

Finally I run qemu as described [here](https://github.com/openbmc/docs/blob/master/cheatsheet.md#using-qemu):
```
qemu-system-arm -m 256 -M palmetto-bmc -nographic \
-drive file=openbmc/build/tmp/deploy/images/palmetto/flash-palmetto-<datetime>,format=raw,if=mtd \
-net nic \
-net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostname=qemu \
```

Which eventually fails with this kernel panic
```
U-Boot 2016.07 (Oct 27 2016 - 14:59:03 +0000)

DRAM:  240 MiB
WARNING: Caches not enabled
Flash: 32 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Error: aspeednic#0 address not set.

Hit any key to stop autoboot:  0 
## Booting kernel from Legacy Image at 20080000 ...
   Image Name:   Phosphor OpenBMC (Phosphor OpenB
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1531599 Bytes = 1.5 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Verifying Checksum ... OK
## Loading init Ramdisk from Legacy Image at 20300000 ...
   Image Name:   obmc-phosphor-image
   Image Type:   ARM Linux RAMDisk Image (lzma compressed)
   Data Size:    1567532 Bytes = 1.5 MiB
   Load Address: 40800000
   Entry Point:  40800000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK

Starting kernel ...
[... 127 lines cut ...]
9fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[<c02d0fb4>] (aspeed_smc_to_fifo) from [<c02d1aa0>] (aspeed_smc_read_reg+0x40/0x78)
[<c02d1aa0>] (aspeed_smc_read_reg) from [<c02cf694>] (spi_nor_read_id+0x2c/0xf8)
[<c02cf694>] (spi_nor_read_id) from [<c02d048c>] (spi_nor_scan+0xa8/0x7ec)
[<c02d048c>] (spi_nor_scan) from [<c02d18e4>] (aspeed_smc_probe+0x4d4/0x650)
[<c02d18e4>] (aspeed_smc_probe) from [<c02b58f0>] (platform_drv_probe+0x38/0x6c)
[<c02b58f0>] (platform_drv_probe) from [<c02b4510>] (driver_probe_device+0x1ac/0x3d8)
[<c02b4510>] (driver_probe_device) from [<c02b4804>] (__driver_attach+0xc8/0x108)
[<c02b4804>] (__driver_attach) from [<c02b29a8>] (bus_for_each_dev+0x78/0xb4)
[<c02b29a8>] (bus_for_each_dev) from [<c02b321c>] (bus_add_driver+0x110/0x230)
[<c02b321c>] (bus_add_driver) from [<c02b4ed0>] (driver_register+0x9c/0xe0)
[<c02b4ed0>] (driver_register) from [<c0500e4c>] (do_one_initcall+0xb8/0x17c)
[<c0500e4c>] (do_one_initcall) from [<c0501010>] (kernel_init_freeable+0x100/0x1c4)
[<c0501010>] (kernel_init_freeable) from [<c03e4ef8>] (kernel_init+0x8/0x10c)
[<c03e4ef8>] (kernel_init) from [<c0102250>] (ret_from_fork+0x14/0x24)
Code: e12fff1e e3110001 0a000002 e4d13001 (e5c03000) 
---[ end trace e616f81fa9521a14 ]---
Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b

---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
```

I've tried in both Ubuntu 14.04 and Ubuntu 16.04 with the same result.

As far as I can tell, I've followed the steps properly. Am I missing something or am I hitting a bug?

legoater, 2016-10-28:
Hello, thanks for the report, it clearly identifies the issue. 

The newer kernels make use of a feature of the SMC controller which is not yet supported in the openbmc/qemu.  The code is ready in mainline but we still need to do the merge. 

I will chat with @amboar next week to schedule that.

mjturek, 2016-11-01:
Thanks! Looks like it's working!