openbmc/linux #119: dev-4.7+ segfault in qemu

shenki, 2016-12-01:
```
[    3.290000] ftgmac100 1e660000.ethernet eth0: NCSI interface up
[    3.290000] Unable to handle kernel paging request at virtual address 12005456
[    3.290000] pgd = cb890000
[    3.290000] [12005456] *pgd=00000000
[    3.290000] Internal error: Oops: 805 [#1] ARM
[    3.290000] Modules linked in:
[    3.290000] CPU: 0 PID: 382 Comm: default.script Not tainted 4.7.10-00195-ge77a87158717 #116
[    3.290000] Hardware name: ASpeed SoC
[    3.290000] task: cf996aa0 ti: cb4de000 task.ti: cb4de000
[    3.290000] PC is at unlink_anon_vmas+0x118/0x19c
[    3.290000] LR is at 0x1000
[    3.290000] pc : [<c019eac8>]    lr : [<00001000>]    psr: a0000013
[    3.290000] sp : cb4dfdc0  ip : 000000a8  fp : 00000001
[    3.290000] r10: c0722259  r9 : cb4b1614  r8 : cb4b15d8
[    3.290000] r7 : cfaf3eb0  r6 : cfaf3eb0  r5 : 1200544a  r4 : cb48d5a0
[    3.290000] r3 : cb805634  r2 : 12005452  r1 : cfaf3efc  r0 : cb48d5a0
[    3.290000] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[    3.290000] Control: 00093177  Table: 4b890000  DAC: 00000051
[    3.290000] Process default.script (pid: 382, stack limit = 0xcb4de190)
[    3.290000] Stack: (0xcb4dfdc0 to 0xcb4e0000)
[    3.290000] fdc0: cb4b15d8 b7179000 b6f79000 cb4b1580 cb4b15d8 cb4dfe1c 00002000 00000000
[    3.290000] fde0: b6ed1000 cb4b1630 cb4b6654 c019439c b6ed1000 cb4b1738 cb4b1738 cb4b6200
[    3.290000] fe00: c0702028 cb40a300 cb4b6200 cf996dc4 00000001 c019a6d8 00000000 cb4b6200
[    3.290000] fe20: 00000001 00000000 00000000 ffffffff cf996dc4 00000001 0000009d 00000400
[    3.290000] fe40: cb5cc000 00000000 c01b7f70 60000013 cf996e28 cf996ae0 cb4b6600 cb4b6254
[    3.290000] fe60: cb4b6600 e61e5297 cb4b6200 00000000 cf996aa0 c010d248 cb4b6200 cb4b6600
[    3.290000] fe80: cf996aa0 c01b7e74 cb4b0334 cb4b0300 00000001 cb7b75a0 00000000 cb530000
[    3.290000] fea0: 00000001 c01f55f8 cb4b0334 00000034 00000000 cb40a300 00000000 cb530000
[    3.290000] fec0: cb41b300 cb40ac00 00000001 c01361e4 c07104c8 00000001 cb4b6654 c0136474
[    3.290000] fee0: 00000000 cb4dfea8 c01b75ac 60000013 cf996e28 cf996ae0 cf996e28 e61e5297
[    3.290000] ff00: c0710e5c cb40a300 c0710e5c fffffff8 c07104ac c04b9ad7 c07104c8 00000001
[    3.290000] ff20: cb4b6654 c01b75a0 cb40a300 00000000 ffffe000 00000000 cfa41000 0000017e
[    3.290000] ff40: cb4b6600 c01b8d5c 00000000 00000000 c01b8cf0 cfa41000 0058a294 0058a224
[    3.290000] ff60: 0058a294 c0702028 00000000 e61e5297 000a9014 000d9604 0058a2ec 000b80ad
[    3.290000] ff80: 0000000b c01025a4 cb4de000 00000000 000a9014 c01b8ea8 00000000 0058a224
[    3.290000] ffa0: 0058a294 c0102400 000d9604 0058a2ec 0058a2ec 0058a224 0058a294 000d9604
[    3.290000] ffc0: 000d9604 0058a2ec 000b80ad 0000000b 000c2d18 0058a224 00000007 000a9014
[    3.290000] ffe0: 2e87f9fa be882028 0004f3e0 b6f238a0 60000010 0058a2ec 4fffd871 4fffdc71
[    3.290000] [<c019eac8>] (unlink_anon_vmas) from [<c019439c>] (free_pgtables+0x80/0x9c)
[    3.290000] [<c019439c>] (free_pgtables) from [<c019a6d8>] (exit_mmap+0xf8/0x200)
[    3.290000] [<c019a6d8>] (exit_mmap) from [<c010d248>] (mmput+0x30/0xc4)
[    3.290000] [<c010d248>] (mmput) from [<c01b7e74>] (flush_old_exec+0x380/0x554)
[    3.290000] [<c01b7e74>] (flush_old_exec) from [<c01f55f8>] (load_elf_binary+0x318/0x101c)
[    3.290000] [<c01f55f8>] (load_elf_binary) from [<c01b75a0>] (search_binary_handler+0x100/0x18c)
[    3.290000] [<c01b75a0>] (search_binary_handler) from [<c01b8d5c>] (do_execveat_common+0x510/0x63c)
[    3.290000] [<c01b8d5c>] (do_execveat_common) from [<c01b8ea8>] (do_execve+0x20/0x28)
[    3.290000] [<c01b8ea8>] (do_execve) from [<c0102400>] (ret_fast_syscall+0x0/0x38)
[    3.290000] Code: eaffffc6 e594300c e5942008 e1a00004 (e5823004) 
[    3.290000] ---[ end trace f9dcb80df0a41235 ]---
Segmentation fault
```

amboar, 2016-12-01:
Odd spot for the crash... maybe memory corruption?

legoater, 2016-12-02:
This is similar to #102.  Not obvious at all.

closed 2017-02-13 by shenki