openbmc/linux #123: Crash at login on Qemu

shenki, 2017-01-30:
```Linux version 4.7.10-e62c7f11175525c028d9a6abaf6a2a5c275664d1 (jenkins@hudson) (gcc version 5.3.0 (GCC) ) #1 Wed Nov 16 04:25:06 UTC 2016```

```
romulus login: ftgmac100 1e660000.ethernet eth0: NCSI interface up
root
Unable to handle kernel NULL pointer dereference at virtual address 00000094
pgd = 9e754000
[00000094] *pgd=00000000
Internal error: Oops: 5 [#1] ARM
CPU: 0 PID: 682 Comm: systemd-network Not tainted 4.7.10-e62c7f11175525c028d9a6abaf6a2a5c275664d1 #1
Hardware name: ASpeed SoC
task: 9d4e4520 ti: 9e722000 task.ti: 9e722000
PC is at fib6_prune_clone+0x0/0x10
LR is at fib6_clean_node+0xac/0x15c
pc : [<803a005c>]    lr : [<803a1e1c>]    psr: 20000013
sp : 9e723c48  ip : 9ec19000  fp : 00000400
r10: 00000008  r9 : 9cfdcb18  r8 : 9e723dc8
r7 : 806217b0  r6 : 8060200c  r5 : 00000001  r4 : 9e723c98
r3 : 803a005c  r2 : 8060200c  r1 : 00000000  r0 : 00000001
Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 00c5387d  Table: 9e754008  DAC: 00000051
Process systemd-network (pid: 682, stack limit = 0x9e722188)
Stack: (0x9e723c48 to 0x9e724000)
3c40:                   9e479000 00000001 00000000 80617380 00000000 fa8e0beb
3c60: 02080020 9e723c98 00000002 80617380 9e6e5700 803a01e8 9e723c98 9cfdcb0c
3c80: 80617380 803a032c 8060200c 9cfdcb0c 00000000 803a0398 8061766c 8061766c
3ca0: 9cfdcb0c 9d0e09e0 9d055634 00000002 8061ac01 00000000 00000001 803a1d70
3cc0: 9cfdcb0c 80617380 803a005c 00000000 00000000 fa8e0beb 9d057020 803a164c
3ce0: 00000003 00000001 00000000 9e6e577c 9e723d2c 00000000 00000000 9cfdcb00
3d00: 9d057020 9e723d54 8060200c 9e740c00 9e723e40 00000000 8060200c 8039baf0
3d20: 00000000 8039e8dc 00000000 00000000 00000000 fa8e0beb 00000000 8060200c
3d40: 9d5ad000 0000000a 8060200c 8039e984 8042faae 000000fe 00000400 00000000
3d60: 00000000 00000002 00000003 00000009 00000001 00000000 00000000 00000000
3d80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
3da0: 00000000 000080fe 00000000 00000000 02000000 00000000 00000000 00000000
3dc0: 00000000 00000000 9d5ad000 80617380 00000001 00000000 00000000 fa8e0beb
3de0: 00000060 80329730 8060200c 803379d4 9e6fe000 9e723df4 9e723df4 80337c28
3e00: 9e740c00 fa8e0beb 9e723e3b 9d5ad000 9e740c00 80329590 8060200c 9e6fe000
3e20: 9e723e40 80339084 9e740c00 00000048 9e740c00 80329578 9e479000 80337b88
3e40: 7fffffff fa8e0beb 9e740c00 00000000 9e6fe000 9e723ed8 00000048 00000000
3e60: 00000008 80338398 7ee73a24 558d77d8 00000050 803a9a48 9d4c07c0 00000000
3e80: 000002aa 000003e0 000003e0 fa8e0beb 8060200c 9e723f04 00000000 8060200c
3ea0: 9e04f7c0 00000010 7ee7381c 00000000 558d7828 803012ac 9e723f04 80302488
3ec0: 9e723ee0 00000002 00000000 00000001 558d6f30 00000048 9e723f04 00000010
3ee0: 00000001 00000000 00000000 9e723ed8 00000000 00000000 00000000 00000000
3f00: 9d5b6d21 00000010 00000000 00000000 00000000 1efb5dd8 39af39bf 00000000
3f20: ffffffff 00000000 136f51a4 00b2d05e 00000000 8062a700 7ee739b8 8060200c
3f40: 9e723f8c 00000001 00000107 80102384 9e722000 00000051 00000001 00000107
3f60: 80102384 7ee73824 00000008 00000000 00000000 80276aac 7ee73824 8060200c
3f80: 00000001 fa8e0beb 00000001 7ee7381c 00000010 76f93ce8 00000122 80102384
3fa0: 9e722000 801021c0 7ee7381c 00000010 00000003 558d6f30 00000048 00000000
3fc0: 7ee7381c 00000010 76f93ce8 00000122 000002aa 558d3530 00000000 558d7828
3fe0: 00000000 7ee73800 7ee737dc 76eeceac 80000010 00000003 e5941008 e594300c
[<803a005c>] (fib6_prune_clone) from [<803a1e1c>] (fib6_clean_node+0xac/0x15c)
[<803a1e1c>] (fib6_clean_node) from [<803a01e8>] (fib6_walk_continue+0xe0/0x164)
[<803a01e8>] (fib6_walk_continue) from [<803a032c>] (fib6_walk+0x28/0x44)
[<803a032c>] (fib6_walk) from [<803a0398>] (fib6_prune_clones+0x50/0x78)
[<803a0398>] (fib6_prune_clones) from [<803a164c>] (fib6_add+0x788/0x8c4)
[<803a164c>] (fib6_add) from [<8039baf0>] (__ip6_ins_rt+0x34/0x50)
[<8039baf0>] (__ip6_ins_rt) from [<8039e8dc>] (ip6_route_add+0x5c/0xc4)
[<8039e8dc>] (ip6_route_add) from [<8039e984>] (inet6_rtm_newroute+0x40/0x60)
[<8039e984>] (inet6_rtm_newroute) from [<80329730>] (rtnetlink_rcv_msg+0x1a0/0x1d0)
[<80329730>] (rtnetlink_rcv_msg) from [<80339084>] (netlink_rcv_skb+0x58/0xb4)
[<80339084>] (netlink_rcv_skb) from [<80329578>] (rtnetlink_rcv+0x18/0x24)
[<80329578>] (rtnetlink_rcv) from [<80337b88>] (netlink_unicast+0x13c/0x210)
[<80337b88>] (netlink_unicast) from [<80338398>] (netlink_sendmsg+0x314/0x338)
[<80338398>] (netlink_sendmsg) from [<803012ac>] (sock_sendmsg+0x14/0x24)
[<803012ac>] (sock_sendmsg) from [<80302488>] (SyS_sendto+0xc4/0x100)
[<80302488>] (SyS_sendto) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
Code: e201101f e3a00001 e0030110 e12fff1e (e5d00093) 
---[ end trace ff0108a6246179b5 ]---
Kernel panic - not syncing: Fatal exception in interrupt
---[ end Kernel panic - not syncing: Fatal exception in interrupt
```

amboar, 2017-01-30:
This is in similar territory to #101

legoater, 2017-07-07:
moved to openbmc/qemu https://github.com/openbmc/qemu/issues/11

closed 2017-07-07 by legoater