openbmc/linux #128: 4.10: lockdep warning when rebooting

shenki, 2017-03-06:
```
[  404.970000] INFO: trying to register non-static key.
[  404.970000] the code is fine but needs lockdep annotation.
[  404.970000] turning off the locking correctness validator.
[  404.970000] CPU: 0 PID: 930 Comm: systemd Not tainted 4.10.0-00049-g94a27692999d #265
[  404.970000] Hardware name: ASpeed SoC
[  404.970000] [<c0107964>] (unwind_backtrace) from [<c0105488>] (show_stack+0x10/0x14)
[  404.970000] [<c0105488>] (show_stack) from [<c013699c>] (register_lock_class+0xb8/0x4ac)
[  404.970000] [<c013699c>] (register_lock_class) from [<c0138004>] (__lock_acquire.constprop.7+0x440/0x7f0)
[  404.970000] [<c0138004>] (__lock_acquire.constprop.7) from [<c0138740>] (lock_acquire+0x68/0x88)
[  404.970000] [<c0138740>] (lock_acquire) from [<c0122d34>] (flush_work+0x1f0/0x268)
[  404.970000] [<c0122d34>] (flush_work) from [<c0124ca0>] (__cancel_work_timer+0x11c/0x1bc)
[  404.970000] [<c0124ca0>] (__cancel_work_timer) from [<c0226934>] (jffs2_sync_fs+0x14/0x38)
[  404.970000] [<c0226934>] (jffs2_sync_fs) from [<c01bcd70>] (iterate_supers+0x98/0xc8)
[  404.970000] [<c01bcd70>] (iterate_supers) from [<c01e3c38>] (sys_sync+0x48/0xb8)
[  404.970000] [<c01e3c38>] (sys_sync) from [<c0102400>] (ret_fast_syscall+0x0/0x38)
```

mdmillerii, 2017-03-06:
dup of #70 

The work is never initialized because its only used for nand, but the work is flushed.

mdmillerii, 2017-03-06:
You can hide this by turning off `JFFS2_FS_WRITEBUFFER`.  The issue is when its configured but unused for any mount.