openbmc/linux #129: jffs2 errors after booting newer kernel

shenki, 2017-03-06:
I was using a system running openbmc master. I then did some testing with a 4.10 kernel, before going back to openbmc master. The 4.7 kernel produced a lot of warnings at boot:

```
jffs2: notice: (382) jffs2_get_inode_nodes: Node header CRC failed at 0x203cf0. {1985,c002,00000044,98f7fb1d}
jffs2: notice: (382) jffs2_get_inode_nodes: Node header CRC failed at 0x203c44. {1985,c002,000000ab,16ebaa73}
jffs2: notice: (382) jffs2_get_inode_nodes: Node header CRC failed at 0x203c00. {1985,c002,00000044,98f7fb1d}
jffs2: notice: (382) jffs2_get_inode_nodes: Node header CRC failed at 0x20200c. {1985,c002,000000ab,16ebaa73}
jffs2: notice: (382) jffs2_get_inode_nodes: Node header CRC failed at 0x1ff2dc. {1985,c002,00000044,98f7fb1d}
jffs2: notice: (382) jffs2_get_inode_nodes: Node header CRC failed at 0x1ff230. {1985,c002,000000ab,16ebaa73}
```

shenki, 2017-03-06:
I was using https://github.com/shenki/linux/commits/dev-4.10 (warning: frequent rebasing) which contains a number of @legoater's mtd changes

```
 ARM: dts: aspeed: add a fastread property
 mtd: spi-nor: aspeed: remove dummies from keep mask
 mtd: spi-nor: add SPI_NOR_DUAL_READ to some flash devices
 mtd: spi-nor: Add support for Macronix mx66l1g45g spi flash
 mtd: spi-nor: aspeed: add support for SPI dual IO read mode
 mtd: spi-nor: aspeed: add DMA support
 ARM: dts: aspeed: update spi controller bindings
 mtd: spi-nor: remove WARN_ONCE() message in spi_nor_write()
```

legoater, 2017-03-06:
I am seeing these warnings when booting on a 4.10 but not when booting back on a 4.7, and only on the ast2500. I will take a look at the kernel config.

shenki, 2017-03-06:
I've only tested on ast2400 so far.

Here are some builds if you want to do some testing:

 https://openpower.xyz/job/openbmc-build-gerrit-trigger-multi/2373/
 https://gerrit.openbmc-project.xyz/#/c/2846/

legoater, 2017-03-06:
I think I found this issue. You should make sure that your kernel config is exactly the same for jffs2. For instance : 
```
CONFIG_JFFS2_SUMMARY=y
CONFIG_JFFS2_FS_XATTR=y
CONFIG_JFFS2_FS_POSIX_ACL=y
CONFIG_JFFS2_FS_SECURITY=y
```

should be the same, else the CRC will differ. 

With matching 4.7 and 4.10 kernel configs, I am not seeing these warnings anymore on ast2400 and ast2500.

legoater, 2017-03-06:
as for the tests, I use 20170303* flash images with a custom fit file including the new kernel which I netboot.

geissonator, 2017-03-06:
Sounds like it's just a config thing that @shenki needs to incorporate?