openbmc/linux #131: Support for Aspeed USB controllers

kc8apf, 2017-03-28:
- USB 2.0 and 1.1 host controllers are normal EHCI and UHCI respectively
- USB2.0 and 1.1 have separate clock and reset controls in SCU
  - Clocks are in SCU0C
  - Resets are in SCU04
- USB2 phy has a separate reset in the USB2.0 Virtual Hub controller (register HUB00)
- Defaults for all of the above are to have clocks disabled and the devices held in reset
- Linux ehci-platform driver (compatible=generic-ehci) should work for USB2 after clock, reset, and phy drivers are written
- Linux uhci-platform driver doesn't have bindings for clocks or resets so separate aspeed HCD driver based on uhci-platform will be needed

mdmillerii, 2017-03-28:
- EHCI driver has clock and reset bindings but orders clock vs reset the opposite of aspeed documentation

- UHCI registers are 4 byte spacing and reordered.

I did a close review of the UHCI registers and believe the UHCI driver can be used with some header surgery to implement custom read/write register methods, along with something to limit the probing to 3 or 4 buses.  I did the work somewhere but never got to testing it on hardware.

ozbenh, 2017-03-29:
I had a look too. The clock & resets aren't a huge issue at this point. Almost all clock in that SoC require the corresponding reset to be held when toggling. I think when we add to the clock driver on/off methods we should simply do the reset along with turning the clock on. At least for the main IP clock. For the PHY, we could use the "port" binding if needed.

I'll experiment today with Joel. There's one other subtle difference with the EHCI: in the SDK code, they have a custom init that avoids the "ehci_halt" call. I don't know yet whether that's really necessary. I'll let you know how things go.

For UHCI, we would need some flags to enable different MMIO ops. I'm tempted to keep that low on the priority list as UHCI sucks in so many other ways unless you have a real need for it.

As for "separate platform driver", I don't think so. It's better to just add support for clocks/resets to uhci-platform I think.

ozbenh, 2017-03-29:
Actually ... it looks like the EHCI may not have a transaction translator, thus requiring the use of the UHCI for USB 1.x devices. I'll play around more.

ozbenh, 2017-03-29:
Is USB working on your EVB ? Even with Aspeed SDK 4.0 kernel I can't get it to react to any connected device in any port...

kc8apf, 2017-03-29:
I've never tried it.

ozbenh, 2017-03-29:
My board is broken. It works on Joel's. EHCI appears to work out with the driver unmodified as long as we get the right pinmux and reset/clk, we'll work on patches for that.
UHCI is a bit more tricky, I'll look into it.

shenki, 2017-03-29:
EHCI:
```
[    6.610000] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    6.610000] ehci-platform: EHCI generic platform driver
[    6.610000] ehci-platform 1e6a1000.usb: EHCI Host Controller
[    6.610000] ehci-platform 1e6a1000.usb: new USB bus registered, assigned bus number 1
[    6.610000] ehci-platform 1e6a1000.usb: irq 20, io mem 0x1e6a1000
[    6.740000] ehci-platform 1e6a1000.usb: USB 2.0 started, EHCI 1.00
[    6.760000] hub 1-0:1.0: USB hub found
[    6.760000] hub 1-0:1.0: 1 port detected
[    6.780000] ehci-platform 1e6a3000.usb: EHCI Host Controller
[    6.810000] ehci-platform 1e6a3000.usb: new USB bus registered, assigned bus number 2
[    6.810000] ehci-platform 1e6a3000.usb: irq 21, io mem 0x1e6a3000
[    6.840000] ehci-platform 1e6a3000.usb: USB 2.0 started, EHCI 1.00
[    6.850000] hub 2-0:1.0: USB hub found
[    6.870000] hub 2-0:1.0: 1 port detected

```
I was able to plug a USB2 device (a phone) into both ports on the EVB and have them enumerate:
```
# [   84.100000] usb 1-1: new high-speed USB device number 2 using ehci-platform
```

UCHI:
```
[    6.880000] uhci_hcd: USB Universal Host Controller Interface driver
[    6.880000] platform-uhci 1e6b0000.usb: register-map setup
[    6.910000] platform-uhci 1e6b0000.usb: Generic UHCI Host Controller
[    6.910000] platform-uhci 1e6b0000.usb: new USB bus registered, assigned bus number 3
[    6.910000] platform-uhci 1e6b0000.usb: detected 2 ports
[    6.910000] platform-uhci 1e6b0000.usb: irq 22, io mem 0x1e6b0000
[    6.950000] hub 3-0:1.0: USB hub found
[    6.950000] hub 3-0:1.0: 2 ports detected
```
I had less luck here with attaching a mouse.

mdmillerii, 2017-03-29:
@shenki was that our kernel or the Aspeed SDK ?

shenki, 2017-03-31:
@mdmillerii this is our kernel with some hacks in early_init to reset the EHCI controller. The UHCI hacks were more invasive.

I am working on a drivers to do the resets properly.

amboar, 2017-07-28:
I've sent out the pinctrl patches: https://lkml.org/lkml/2017/7/18/76

shenki, 2018-08-23:
The USB driver is are now upstream (4.18) and in openbmc (dev-4.17).

closed 2018-08-23 by shenki