openbmc/linux #132: Write driver for AST2400/2500 Hash and Crypto Engine

kc8apf, 2017-04-04:
Notes from my experiments:
- HACE can _only_ access RAM
  - HACE address registers assume offset into RAM rather than full AHB address
  - FIT SHA verification is done before copying to RAM(?) so HACE doesn't help
- Writes to the hash, crypto, and RSA "engine control" registers start the operation
- Hash, crypto, and RSA can run in parallel
- Command queue is just queue of register writes
  - Commands are tag, register offset within HACE register map, and data
  - It looks like tag changes get written to the tag register automagically
  - Writes to tag register can block(?) based on whether the engines are idle or busy
  - Useful operation seems to be:
    - Enable interrupts on tag register write _and_ completion
    - Queue a "transaction" (series of register writes) with the same "tag": write data src, write hash dest, write data len, write engine start
    - When tag register write interrupt fires, that transaction is starting
    - When completion interrupt fires, the transaction is done.
    - Using only tag register write interrupt would miss completion of last transaction in queue.

mdmillerii, 2017-04-04:
I've been spending much of my free time working on this driver and have one about 90% ready to touch hardware bringup.  I need to work on completions and probably clean up and chain the multi-stage hmac hashing.

MaxPower2005, 2017-07-27:
@mdmillerii Where can I download this driver/patch? I would like to give it a try on my AST2500 hardware.

mdmillerii, 2017-07-28:
I haven't had a chance to work on this in several weeks, but I can try to post my WIP.