openbmc/linux #136: VUART drops blocks of data on large host-side TX

jk-ozlabs, 2017-07-21:
[I'm not yet convinced that this is specifically a BMC kernel issue, but tracking under openbmc/linux for now]

If I write 132kB of data to the host-side of the VUART, the BMC side only receives around 69-75kB of the written data. 

on the host:

```
[jk@spoon2 ~]$ cat data | sudo dd of=/dev/hvc0
256+0 records in
256+0 records out
131072 bytes (131 kB, 128 KiB) copied, 0.172273 s, 761 kB/s
```

on the BMC:

```
root@spoon2-bmc:~# hexdump -v -C /dev/ttyVUART0
00000000  00 00 02 03 04 05 06 07  08 09 0a 0b 0c 0d 0e 0f  |................|
00000010  10 11 12 13 14 15 16 17  18 19 1a 1b 1c 1d 1e 1f  |................|
00000020  20 21 22 23 24 25 26 27  28 29 2a 2b 2c 2d 2e 2f  | !"#$%&'()*+,-./|
[…]
000115c0  c0 c1 c2 c3 c4 c5 c6 c7  c8 c9 ca cb cc cd ce cf  |................|
000115d0  d0 d1 d2 d3 d4 d5 d6 d7  d8 d9 da db dc dd de df  |................|
000115e0  e0 e1 e2 e3 e4 e5 e6 e7  e8 e9 ea eb ec ed ee ef  |................|
```

\- with `hexdump` waiting there for further data.

[For those reproducing elsewhere, you'll need to first:
 * on the BMC: `systemctl stop obmc-console@ttyVUART0`
 * on the BMC: `stty -F /dev/ttyVUART0 raw`
 * on the host: `stty -F /dev/hvc0 raw`

]

Given that we should be seeing 0x20000 bytes output, we're about 0xea10 bytes short there. This delta is a little variable (more detail below) but always seems to end on an 0xf0 boundary. With the specific data pattern that I'm using, the data always seems to be truncated (no missing data occurs before that point).

Over 20 runs, here are the lengths that I see:

0x115f0
0x114f0
0x116f0
0x115f0
0x115f0
0x114f0
0x116f0
0x115f0
0x114f0
0x115f0
0x115f0
0x118f0
0x114f0
0x114f0
0x113f0
0x124f0
0x114f0
0x114f0
0x115f0
0x115f0

jk-ozlabs, 2017-07-21:
Interestingly, the UART stats show the complete transfer (the two sets of stats are before and after a 128kB transfer):

```
root@spoon2-bmc:~# grep 1E787000 /proc/tty/driver/serial 
2: uart:16550A mmio:0x1E787000 irq:22 tx:42071 rx:14794301 RTS|CTS|DTR|DSR|CD
root@spoon2-bmc:~# grep 1E787000 /proc/tty/driver/serial 
2: uart:16550A mmio:0x1E787000 irq:22 tx:42071 rx:14925373 RTS|CTS|DTR|DSR|CD
```

where 14925373 - 14794301 = 131072.

So, perhaps the 8250 core is dropping chars for some reason...

mdmillerii, 2017-07-21:
There is code to limit the number of tty flip buffers in the tty layer.  I remember it exist but haven't searched for a reference.

jk-ozlabs, 2017-07-24:
@mdmillerii : even if we're hitting the flip buffer limit, I'd expect that the overrun would be counted in the tty stats...

jk-ozlabs, 2017-07-24:
(ie, I'm not seeing `tty_insert_flip_char()` failing during this test, as the `buf_overrun` stats stay at zero)

jk-ozlabs, 2017-07-25:
Wait, looks like I *am* seeing buf_overruns, accounting for the missing chars.

jk-ozlabs, 2017-07-25:
This seems to help:

  https://lkml.org/lkml/2017/7/25/149

.. in that I can now push 128kB of data through the VUART without problems. However, we still see the same issue with larger amounts of data.

Investigating whether we can increase the flip buffer sizes for VUART.

jk-ozlabs, 2018-05-08:
This should be fixed by https://github.com/openbmc/linux/commit/158c8e15d577cebdb047f5d553e12054b15b1b84, which is now in the tty-next tree too.

closed 2018-05-08 by jk-ozlabs