openbmc/linux #140: iio: aspeed-adc: wait for initial sequence return -110 means connection timeout

rhung2, 2017-10-08:
Hi
I tested the last aspeed-adc.c in AST2500 EVB board and saw that there was a error code:
[    1.852000] aspeed_adc: probe of 1e6e9000.adc failed with error -110
It just returned error code here:
/* Wait for initial sequence complete. */
		ret = readl_poll_timeout(data->base + ASPEED_REG_ENGINE_CONTROL,
					 adc_engine_control_reg_val,
					 adc_engine_control_reg_val &
					 ASPEED_ADC_CTRL_INIT_RDY,
					 ASPEED_ADC_INIT_POLLING_TIME,
					 ASPEED_ADC_INIT_TIMEOUT);
		if (ret)
			goto scaler_error;

it seems readl_poll_timeout gets some errors. How do I fix it? any idea?

amboar, 2017-10-08:
ping @aspeedtech @kc8apf

kc8apf, 2017-10-09:
The code seems to match the initialization process described in the AST2500 datasheet.  Getting -ETIMEDOUT means the ready bit never got set.  I'm assuming the prior write to enable the engine and change it to normal mode was successful.  Only thing I can guess is that the clock coming into the ADC was disabled prior to probe.  Currently, the call to enable the clocks (we enable the scaler directly which would indirectly enable the prescaler and then the input clock) comes after the ADC init sequence.  I don't know if the ADC hardware requires the sample clock to perform initialization.  Either way, moving 'clk_prepare_enable(data->clk_scaler->clk);' before the wait_init_sequence block seems like a good things to do.

rhung2, 2017-10-10:
Thanks, kc8apf.
I moved clk_prepare_enable in front of wait_init_sequence, but it still returned -110:aspeed_adc: probe of 1e6e9000.adc failed with error -110

        /* Start all channels in normal mode. */
	printk(KERN_INFO "[DEBUG] aspeed_adc: driver Start all channels in normal mode.\n");
	clk_prepare_enable(data->clk_scaler->clk);
		
	model_data = of_device_get_match_data(&pdev->dev);

	if (model_data->wait_init_sequence) {
		/* Enable engine in normal mode. */

kc8apf, 2017-10-10:
Interesting.  After the polling fails, what value is being read back from ASPEED_REG_ENGINE_CONTROL?

rhung2, 2017-10-11:
The ASPEED_REG_ENGINE_CONTROL val is 0, it seems driver code cannot init ADC somehow.
it should be failed on writel(ASPEED_OPERATION_MODE_NORMAL | ASPEED_ENGINE_ENABLE, data->base + ASPEED_REG_ENGINE_CONTROL);

rhung2, 2017-10-11:
I found the root cause, there was no init step in system control untit (0x1e6e2000) before writel(...).
This is my example code /* SCU INIT ADC */:
static int aspeed_adc_probe(struct platform_device *pdev)
{
        struct iio_dev *indio_dev;
        struct aspeed_adc_data *data;
        const struct aspeed_adc_model_data *model_data;
        struct resource *res;
        const char *clk_parent_name;
        int ret;
        u32 adc_engine_control_reg_val;

        /* SCU INIT ADC */
        writel(readl((void *)(ast_scu_base + AST_SCU_RESET)) | SCU_RESET_ADC, (void *) (ast_scu_base + AST_SCU_RESET));
        writel(readl((void *)(ast_scu_base + AST_SCU_RESET)) & ~SCU_RESET_ADC, (void *) (ast_scu_base + AST_SCU_RESET));

        indio_dev = devm_iio_device_alloc(&pdev->dev, sizeof(*data));
        if (!indio_dev)
                return -ENOMEM;

        data = iio_priv(indio_dev);
        data->dev = &pdev->dev;

kc8apf, 2017-10-11:
@shenki DId the reset controller get added?  I thought we had early init
code that enabled all peripherals.

amboar, 2017-10-12:
@rhung2 can you point us to the kernel tree that you're testing on the EVB?

shenki, 2017-10-13:
@kc8apf the under-review clk driver has a reset controller in it. Once we have that merged we can add the reset release to the adc driver.

rhung2, 2017-10-13:
I was using,
adc: adc@1e6e9000 {
				compatible = "aspeed,ast2500-adc";
				reg = <0x1e6e9000 0xC8>;
				clocks = <&clk_apb>;
				#io-channel-cells = <0 1 2 3>;
		};

amboar, 2017-10-13:
@rhung2 I meant can you link us to the source code of the kernel you are testing with?

rhung2, 2017-10-13:
I used linux-4.9.46.

amboar, 2017-10-13:
@rhung2 right, any reason you're not using openbmc/linux dev-4.10? Regardless, upstream won't have the board file, but in dev-4.10 we poke the reset register with the intent to reset the AHB bridges. As it happens the write also releases the rest of the IP blocks from reset: https://github.com/openbmc/linux/blob/dev-4.10/arch/arm/mach-aspeed/aspeed.c#L232

rhung2, 2017-10-13:
Thanks, Amboar. linux-4.9.46 is our current development base version. I will merge 4.10 and 4.9 and close this issue.

closed 2017-10-13 by rhung2