openbmc/linux #142: Monotonic timers can run backward

madscientist159, 2018-01-22:
@amboar 

Found an interesting bug somewhere in the OpenBMC system, presumed to be the kernel: monotonic timers appear to be able to run **backward**!

For the Talos systems we have a GPIO wired up to a piezo buzzer to be able to produce short beeps.  When trying to create a software beep generation utility, the timers kept acting weird (massive jitter, on the order of seconds), but this was chalked up to the underpowered BMC CPU.

However, busy waiting on the output of CLOCK_MONOTONIC_RAW yielded something far more interesting:

> OVERRAN timer (wanted 500000, got 6513841645600526421, start: 6513841646057535338.4828071968854116287 end: 2046806223074218103.2130774092136435831)
> OVERRAN timer (wanted 500000, got 6513841645640195121, start: 6513841646131186691.4828071968967436340 end: 2363136375515369591.2617480013752422519)
> OVERRAN timer (wanted 500000, got 6513841645630465387, start: 6513841646231423169.4828071969057943084 end: 2793648770391593079.3006203519299866743)
> OVERRAN timer (wanted 500000, got 6513841645626057624, start: 6513841646291207849.4828071969113320001 end: 3050422015793418359.3244045566768173175)

Note how the current times jump all over the place.

Code is here:

https://git.raptorcs.com/git/talos-skeleton/tree/softbeep/beep_obj.c

shenki, 2018-01-30:
Are you sure this is a kernel issue? I did notice a bunch of compiler warnings when building the tool.

Once fixing them I wasn't able to reproduce. Here's some example output:

```
./beep_obj 
OVERRAN timer (wanted 500000, got 1324283, start: 5965383.929866907 end: 5965383.931191190)
OVERRAN timer (wanted 500000, got 41248000, start: 5965383.959487069 end: 5965384.735069)
OVERRAN timer (wanted 500000, got 16393535, start: 5965384.21594463 end: 5965384.37987998)
```

closed 2018-02-05 by shenki

madscientist159, 2018-02-05:
@shenki Do you mind posting the fixes you applied?  I haven't had any time to look into this again but I do remember the timer problem showed up after the system was online for a while and it took a number of retries of the `beep` program to trigger the problem.