openbmc/linux #144: ubsan: pincrtrl warning

shenki, 2018-03-22:
e156398bfcad44943ea4881a390b8b816c854593
v4.16-rc6-119-ge156398bfcad

from Joel's experimental 4.16 tree, on a qemu romulus machine

```
[   26.624869] UBSAN: Undefined behaviour in ./arch/arm/include/asm/bitops.h:297:17
[   26.627188] negation of -2147483648 cannot be represented in type 'int':
[   26.628844] CPU: 0 PID: 1 Comm: swapper Not tainted 4.16.0-rc6-00118-g671c39af8e7d-dirty #269
[   26.629965] Hardware name: Generic DT based system
[   26.634039] [<80016978>] (unwind_backtrace) from [<80012af8>] (show_stack+0x20/0x24)
[   26.635105] [<80012af8>] (show_stack) from [<80929cfc>] (dump_stack+0x20/0x28)
[   26.636571] [<80929cfc>] (dump_stack) from [<8049a178>] (ubsan_epilogue+0x14/0x60)
[   26.637785] [<8049a178>] (ubsan_epilogue) from [<8049a694>] (__ubsan_handle_negate_overflow+0xd0/0xf4)
[   26.638634] [<8049a694>] (__ubsan_handle_negate_overflow) from [<804a87dc>] (aspeed_sig_expr_eval+0x2fc/0x43c)
[   26.639900] [<804a87dc>] (aspeed_sig_expr_eval) from [<804a8df0>] (aspeed_disable_sig.part.1+0x54/0x148)
[   26.640854] [<804a8df0>] (aspeed_disable_sig.part.1) from [<804a9854>] (aspeed_gpio_request_enable+0x120/0x218)
[   26.641794] [<804a9854>] (aspeed_gpio_request_enable) from [<804a3250>] (pin_request+0x100/0x50c)
[   26.642536] [<804a3250>] (pin_request) from [<804a3d08>] (pinmux_request_gpio+0x60/0xa0)
[   26.643188] [<804a3d08>] (pinmux_request_gpio) from [<804a0adc>] (pinctrl_gpio_request+0x1d4/0x2ec)
[   26.643881] [<804a0adc>] (pinctrl_gpio_request) from [<804baff0>] (aspeed_gpio_request+0x128/0x1ec)
[   26.644685] [<804baff0>] (aspeed_gpio_request) from [<804ad414>] (gpiod_request_commit+0x10c/0x224)
[   26.645570] [<804ad414>] (gpiod_request_commit) from [<804b00d4>] (gpiod_request+0x58/0x180)
[   26.646226] [<804b00d4>] (gpiod_request) from [<804b5714>] (gpiod_get_index+0xbc/0x400)
[   26.646863] [<804b5714>] (gpiod_get_index) from [<804aa938>] (devm_gpiod_get_index+0x58/0x94)
[   26.647507] [<804aa938>] (devm_gpiod_get_index) from [<804aa994>] (devm_gpiod_get+0x20/0x24)
[   26.648159] [<804aa994>] (devm_gpiod_get) from [<806ac594>] (fsi_master_gpio_probe+0x7c/0x2b0)
[   26.648838] [<806ac594>] (fsi_master_gpio_probe) from [<8051bb50>] (platform_drv_probe+0x88/0x170)
[   26.649498] [<8051bb50>] (platform_drv_probe) from [<80518418>] (driver_probe_device+0x484/0x908)
[   26.650136] [<80518418>] (driver_probe_device) from [<805189b8>] (__driver_attach+0x11c/0x180)
[   26.650833] [<805189b8>] (__driver_attach) from [<805145c8>] (bus_for_each_dev+0xb0/0x104)
[   26.651642] [<805145c8>] (bus_for_each_dev) from [<80518cd8>] (driver_attach+0x38/0x54)
[   26.652345] [<80518cd8>] (driver_attach) from [<8051572c>] (bus_add_driver+0x1ec/0x3f8)
[   26.652929] [<8051572c>] (bus_add_driver) from [<8051a0bc>] (driver_register+0xe4/0x1c8)
[   26.653543] [<8051a0bc>] (driver_register) from [<8051ccc4>] (__platform_driver_register+0x74/0x9c)
[   26.654265] [<8051ccc4>] (__platform_driver_register) from [<80b4b03c>] (fsi_master_gpio_driver_init+0x20/0x28)
[   26.655330] [<80b4b03c>] (fsi_master_gpio_driver_init) from [<80b18574>] (do_one_initcall+0x15c/0x260)
[   26.656156] [<80b18574>] (do_one_initcall) from [<80b1891c>] (kernel_init_freeable+0x2a4/0x388)
[   26.656817] [<80b1891c>] (kernel_init_freeable) from [<8094feb4>] (kernel_init+0x1c/0x124)
[   26.657436] [<8094feb4>] (kernel_init) from [<800090f0>] (ret_from_fork+0x14/0x24)
[   26.657965] Exception stack(0x97d03fb0 to 0x97d03ff8)
[   26.658604] 3fa0:                                     00000000 00000000 00000000 00000000
[   26.659406] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   26.660149] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
```

```
#0  __ubsan_handle_negate_overflow (data=0x80000000, old_val=2147483648) at lib/ubsan.c:220
        flags = 2546226464
        old_val_str = " Uė\034:З\b:З0\313S\200d\307\"\200 Uė\000\000\000\000D:З :ЗpqT\200"
#1  0x804a87dc in ffs (x=<optimized out>) at ./arch/arm/include/asm/bitops.h:297
No locals.
#2  __ffs (x=<optimized out>) at ./arch/arm/include/asm/bitops.h:306
No locals.
#3  aspeed_sig_desc_eval (map=<optimized out>, enabled=<optimized out>, desc=<optimized out>) at drivers/pinctrl/aspeed/pinctrl-aspeed.c:127
        ret = -2133780044
        raw = 134193152
#4  aspeed_sig_expr_eval (expr=0x80981f84 <sig_expr_NORD2_PNOR>, enabled=false, maps=0x0) at drivers/pinctrl/aspeed/pinctrl-aspeed.c:160
        desc = 0x80981f94 <sig_descs_NORD2_PNOR>
        i = 0
#5  0x804a8df0 in aspeed_sig_expr_disable (maps=<optimized out>, expr=<optimized out>) at drivers/pinctrl/aspeed/pinctrl-aspeed.c:274
        ret = -2133770768
#6  aspeed_disable_sig (exprs=0x80d13dec <sig_exprs_NORD2>, maps=0x80d130fc <aspeed_g5_pinctrl_data>)
    at drivers/pinctrl/aspeed/pinctrl-aspeed.c:301
        ret = -2133770768
#7  0x804a9854 in aspeed_disable_sig (maps=<optimized out>, exprs=<optimized out>) at drivers/pinctrl/aspeed/pinctrl-aspeed.c:539
No locals.
#8  aspeed_gpio_request_enable (pctldev=0x80d119b4, range=0x80000000, offset=0) at drivers/pinctrl/aspeed/pinctrl-aspeed.c:507
        pdata = 0x80d130fc <aspeed_g5_pinctrl_data>
        prios = 0x80d13dd8 <pin_exprs_210+8>
[... 60 lines cut ...]
No locals.
#23 0x805145c8 in bus_for_each_dev (bus=0x80d35f68, start=0x97e57ab0, data=0x80dafe64 <fsi_master_gpio_driver+20>, 
    fn=0x8051889c <__driver_attach>) at drivers/base/bus.c:311
        i = {i_klist = 0x97c60f4c, i_cur = 0x97e57ab0}
        error = -1746568528
#24 0x80518cd8 in driver_attach (drv=0x80dafe64 <fsi_master_gpio_driver+20>) at drivers/base/dd.c:844
---Type <return> to continue, or q <return> to quit---
No locals.
#25 0x8051572c in bus_add_driver (drv=0x80dafe64 <fsi_master_gpio_driver+20>) at drivers/base/bus.c:667
        __func__ = "bus_add_driver"
#26 0x8051a0bc in driver_register (drv=0x80dafe64 <fsi_master_gpio_driver+20>) at drivers/base/driver.c:166
        ret = -2133620400
#27 0x8051ccc4 in __platform_driver_register (drv=0x80dafe50 <fsi_master_gpio_driver>, owner=0x0) at drivers/base/platform.c:635
No locals.
#28 0x80b4b03c in fsi_master_gpio_driver_init () at drivers/fsi/fsi-master-gpio.c:732
No locals.
#29 0x80b18574 in do_one_initcall (fn=0x80b4b01c <fsi_master_gpio_driver_init>) at init/main.c:833
        ret = 0
        msgbuf = "\000>З\024^\006\200|\305\"\200,?З\200e\272\200|g\272\200\000\000\000\000\006\000\000\000\000\000\000\000\006\000\000\000\270֨\200\354x\261\200,Ѫ\200D?ЗU\252\277\237\000\000\000"
#30 0x80b1891c in do_initcall_level (level=<optimized out>) at init/main.c:899
        fn = 0x80b7ec50 <__initcall_scom_init6>
#31 do_initcalls () at init/main.c:907
        level = 7
#32 do_basic_setup () at init/main.c:925
```

amboar, 2018-04-09:
This appears to be a problem with `ffs()` using a signed value (`int`) as its parameter type. Shifting through the top bit of signed values gives undefined behaviour. As such this is less a problem with the ASPEED pinctrl driver and more a kernel-wide issue.

amboar, 2019-03-19:
Closing as a false positive based on the above.

closed 2019-03-19 by amboar