openbmc/linux #154: Disadvantages of DTS files?

ouyangxx10, 2018-05-07:
When I use the DTS file in the linux kernel, the linux kernel almost implementation all the driver of device, such as pca95xx, gpio-led, some i2c devices. But there is a downside to this approach. Sometimes, when BMC bootup and load device driver and  the driver need communicate with the device, need read or write the register of the device, that means the device need init before the BMC load the device driver. But the device may be init after the OS power up. There is a problem: the BMC bootup before the OS, and will occur the error of loading device driver. Do we have a way to solve this problem and then continue to use DTS to configure the device? such as the device of ina219.
```
[    3.334714] ina2xx 1-0040: error configuring the device: -6
[    3.348700] ina2xx 1-0041: error configuring the device: -6
```

ouyangxx10, 2018-05-08:
@ya-mouse  The patch just order the i2c device driver loading in all the driver. I think the patch can not resolve my problem. We cannot cconfirm the time od "ina219" device driver need to load, not sure it will wait one min or one day or even more long time. I think the DST file lack a condition  to judge when the driver need loading. thanks.

ya-mouse, 2018-05-08:
Regarding to the message above ("No such device or address"), you don't have i2c subsystem at the time of ina2xx loading: [devm_regmap_init_i2c](https://github.com/linux4sam/linux-at91/blob/master/drivers/iio/adc/ina2xx-adc.c#L647)
The driver will not wait, but moving i2c upper might add some time between ina2xx load and i2c bus init.
There is nothing to do with DTS as far as with old fashion way with platform's definition. If re-order wouldn't help (you may try anyway), I would recommend to add init script to load this driver.
We use overlay-stuff (backported from RaspberryPi) to load/unload [additional](https://github.com/ya-mouse/meta-openbmc-yandex/blob/master/meta-yandex/meta-openrack/meta-shaosi/recipes-kernel/overlays/files/shaosi-CB.dts) DTS on-the-fly when needed.

ouyangxx10, 2018-05-15:
maybe need load the driver when it need it, thanks.

closed 2018-05-15 by ouyangxx10