openbmc/linux #155: Issue with the w1 driver

akashgj, 2018-05-16:
I build Linux image using https://github.com/openbmc/openbmc/ for Ziaus machine and tried to boot the Linux using QEMU. Booted successfully, but showing the following messages periodically.

 [  125.350000] w1_master_driver w1_bus_master1: Family 0 for 00.400000000000.46 is not registered.
[  125.930000] w1_master_driver w1_bus_master3: Family 0 for 00.400000000000.46 is not registered.
[  125.930000] ------------[ cut here ]------------
[  125.930000] WARNING: CPU: 0 PID: 399 at /kernel-source//fs/sysfs/dir.c:31 sysfs_warn_dup+0x68/0x88
[  125.930000] sysfs: cannot create duplicate filename '/bus/w1/devices/00-400000000000'
[  125.930000] CPU: 0 PID: 399 Comm: w1_bus_master3 Tainted: G        W       4.7.10-2499942f10b161f6c980e558ffa3f99402b1710e 
[  125.930000] Hardware name: ASpeed SoC

Please someone help to resolve the issue.

mdmillerii, 2018-05-16:
My understanding of w1 (which is from ancient memory seeing technology demos, white papers, etc,  and reading Linux kernel driver code) is that the id address, `00.400000000000.46`, is supposed to be a unique serial number of the device, so the Linux kernel does not include the master bus when creating the device name.  Having the same device serial number appear on two buses causes sysfs to attempt to use the same name for the `/sys/bus/w1/devices` symlink and results in the warning you listed, with the second device creation being failed back to the caller.  I have seen similar issues reported (with address 800..0 iirc) when the hardware pullup was missing).

My advice is to teach your qemu machine that each w1 device has a unique serial number.

legoater, 2018-05-17:
what type of w1 device are you using ? I don't know any under QEMU. 

Also we don't have a zaius-bmc QEMU machine. Could you paste your QEMU command line please ?