openbmc/linux #161: Remote Logging config not updated

geissonator, 2018-09-14:
@gkeishin commented on [Fri Sep 14 2018](https://github.com/openbmc/phosphor-logging/issues/12)

On the latest build, configure the remote logging via REST interface and verify the /etc/rsyslog.d/server.conf config setting
```
$ ./rest_cmd -i wsbmcxxx-o GET -u /xyz/openbmc_project/logging/config/remote
Executing: GET  /xyz/openbmc_project/logging/config/remote

/xyz/openbmc_project/logging/config/remote
{
    "Address": "xx.xx.xx.xx",
    "Port": 514
}
Login to BMC and verify
```
```
root@wsbmcxxx:~# cat /etc/rsyslog.d/server.conf
# Remote Logging (we use TCP for reliable delivery)
#*.* @@remote-host:port
````
Don't see any failure or suspicious activity on journald to tell what went wrong there.

---

@dkodihal commented on [Fri Sep 14 2018](https://github.com/openbmc/phosphor-logging/issues/12#issuecomment-421236303)

Something weird with /etc. Existing files are not writeable. Thoughts @anoo1?

```
root@wsbmc014:/etc# cat synclist 
/etc/dropbear/
/etc/group
/etc/gshadow
/etc/hostname
/etc/machine-id
/etc/passwd
/etc/shadow
/etc/ssl/
/etc/ssl/certs/nginx/
/etc/ssl/private/
/etc/systemd/network/
root@wsbmc014:/etc# echo /bad/path > synclist 
-sh: synclist: No such file or directory
root@wsbmc014:/etc# echo /bad/path > foo      
root@wsbmc014:/etc# cat foo 
/bad/path
```

---

@dkodihal commented on [Fri Sep 14 2018](https://github.com/openbmc/phosphor-logging/issues/12#issuecomment-421236429)

Build info :

```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.3-329"
VERSION_ID="v2.3-329-gdfb7e65"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.3-329"
BUILD_ID="v2.3"
```

geissonator, 2018-09-14:
We're not sure how or why but we tracked this down to this commit:
f0ca94d: linux-aspeed: Update to 4.18.7 stable tree

To summarize the issue, an existing file in /etc/ can not be edited. We get a error like this when trying to save the change:
```
'/etc/nginx/nginx.conf' No such file or directory
```

This is only on /etc/ which is an overlay filesystem.  You can do some magic where you create the file in /tmp/ and then copy over the file in /etc/.  And once you "fix" the file you can then edit it fine. New files can be created fine in /etc/.

This is a significant issue because of files like our BMC network settings that sit out in /etc/ so we'll need to revert the commit until we can get this figured out.

geissonator, 2018-09-14:
A few overlayfs mentions in https://cdn.kernel.org/pub/linux/kernel/v4.x/ChangeLog-4.18.7 but nothing obvious to me.

anoo1, 2018-09-14:
Issue is not seen on romulus or palmetto, so seems unique to witherspoon which has a ubifs filesystem.

eddiejames, 2018-09-19:
https://github.com/openbmc/linux/commit/cef393ce9aea9e0a0804da9cbabb4bec78bd29fa#diff-86a8be662de25cbab39ff847f1e826af

This is the bad commit.

anoo1, 2018-09-19:
Eddie found the bad commit has just been reverted - https://lore.kernel.org/patchwork/patch/986563/

shenki, 2018-09-20:
I tested the upstream revert fixes the problem. It's been pulled into dev-4.18 and bumped in meta-aspeed.

closed 2019-04-30 by shenki