openbmc/linux #168: ASPEED Chip I2C DAM Implementation

wiwJackHsin, 2019-01-25:
Hi there,
Is there a plan to add I2C DMA mode support in ASPEED i2c driver(linux/drivers/i2c/busses/i2c-aspeed.c)?

tomrepo, 2019-01-25:
I'm actually having the same question. If no one is working on it, I may spend sometime looking into the area sometime this year (probably second half of 2019).

bjh83, 2019-02-04:
jae.hyun.yoo@linux.intel.com (@yoojae) has been doing a lot of work on the I2C driver lately (mostly multi-master support); I think he may have mentioned working on DMA support, but I am not sure. In anycase, you guys might want to talk to make sure you don't step on each other's toes.

tomrepo, 2019-02-05:
> [jae.hyun.yoo@linux.intel.com](mailto:jae.hyun.yoo@linux.intel.com) (@yoojae) has been doing a lot of work on the I2C driver lately (mostly multi-master support); I think he may have mentioned working on DMA support, but I am not sure. In anycase, you guys might want to talk to make sure you don't step on each other's toes.

That's great. Thank you Brendan for the message.