openbmc/linux #169: AST2400 support for aspeed_gfx

Kitsok, 2019-02-01:
Hi all!

There is a gfx driver for embedded graphics display controller that is used for KVM to host.

It's declared as ast2400 and ast2500 compatible as stated in the documentation and in the driver itself:
`$ cat drivers/gpu/drm/aspeed/aspeed_gfx_drv.c | grep '\.compatible'
        { .compatible = "aspeed,ast2400-gfx" },
        { .compatible = "aspeed,ast2500-gfx" },`

But if I try to include it into kernel and enable it in the DTS, I've got an error:
`[    5.050501] aspeed_gfx 1e6e6000.display: failed to find SCU regmap`

This is because aspeed_gfx depends on aspeed SCU, but in the driver itself it's searched by the following line:
`priv->scu = syscon_regmap_lookup_by_compatible("aspeed,ast2500-scu");`

Looks like the scu lookup must be implemented somehow with respect to the current compatibility (ast2400 or ast2500).

shenki, 2019-02-05:
The driver hasn't been tested on AST2400 due to a lack of development time. I would welcome work to enable it if you have time to test.

amboar, 2019-03-19:
The graphics bindings could do with a `syscon` property that's a phandle to the SCU node. This way we don't need the compatible in the lookup (just use a different syscon API to resolve the syscon instance).

The alternative might be to include some platform data, or perform a SCU compatible lookup based on the graphics compatible value, but both of those seem a bit hacky.

shenki, 2022-05-03:
As of v5.11 (bce724fa58e67d24fe9eddb40ebb665fbe4bd7e8) the driver has support for the 2400.

closed 2022-05-03 by shenki