openbmc/linux #171: aspeed-g5.dtsi: Incorrect GPIO range prevents access to AC5-AC6 pins

operezmuena, 2019-03-18:
According to The AST2500/AST2520 specs, these chips support up to 228 GPIO pins.  However, 'gpio-ranges' value in 'aspeed-g5.dtsi' file is currently setting the upper limit to 220 which won't allow access to pins AC5 and AC6 on my system.

I believe the correct upper limit value is 232 after I found the below snip in the 'gpio-aspeed.c' driver:

```
static const struct aspeed_gpio_config ast2500_config =
	/* 232 for simplicity, actual number is 228 (4-GPIO hole in GPIOAB) */
	{ .nr_gpios = 232, .props = ast2500_bank_props, };
```

Only after I changed the 'gpio-ranges' upper limit to 232 in 'aspeed-g5.dtsi' file, my system was able to access GPIO pins AC5 and AC6.   Please see below patch that I have successfully tested on my AST2520 system.

```
From f580f57f9f0fa0bfe15f699af833d4270d0c99e3 Mon Sep 17 00:00:00 2001
From: Oscar A Perez <oscar.perez@ztsystems.com>
Date: Thu, 14 Mar 2019 19:42:39 +0000
Subject: [PATCH] ARM: dts: aspeed-g5: fixed gpio-ranges upper value

---
 arch/arm/boot/dts/aspeed-g5.dtsi | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/arch/arm/boot/dts/aspeed-g5.dtsi b/arch/arm/boot/dts/aspeed-g5.dtsi
index d8a9d20cc4e0..c0991e066b0d 100644
--- a/arch/arm/boot/dts/aspeed-g5.dtsi
+++ b/arch/arm/boot/dts/aspeed-g5.dtsi
@@ -263,7 +263,7 @@
 				compatible = "aspeed,ast2500-gpio";
 				reg = <0x1e780000 0x1000>;
 				interrupts = <20>;
-				gpio-ranges = <&pinctrl 0 0 220>;
+				gpio-ranges = <&pinctrl 0 0 232>;
 				clocks = <&syscon ASPEED_CLK_APB>;
 				interrupt-controller;
 				#interrupt-cells = <2>;
-- 
2.17.1

```

amboar, 2019-03-19:
Thanks for the patch. Can you please follow the usual kernel development process to send it upstream? Once it's upstream you can ping @shenki on the OpenBMC mailing list to get it backported into the OpenBMC kernel tree.

operezmuena, 2019-09-10:
Patch was added to Linux-Aspeed v5.2.11.  Closing.

closed 2019-09-10 by operezmuena