openbmc/linux #175: Upstream miscellaneous control interfaces patch

amboar, 2019-03-20:
We need a way to expose random bits in random registers in random devices in a unified way. This ran aground upstream previously due to the proposed devicetree bindings. Avoid that mess, implement support as a library, then implement basic drivers for the devices and expose the bits using the library.

bradbishop, 2019-03-20:
@osenft just fyi I know you are interested in this work.

osenft, 2019-03-20:
@amboar, can you explain more about the proposed solution?

I still feel its beneficial to have a clean Kernel<->userspace boundary that clearly defines which bits a userspace process can modify. I generally prefer a /sys based interface since the abstraction lives in the kernel. However, I appreciate that this means a lot of (potentially) duplicate work that each needs to be upstreamed separately ...

amboar, 2019-03-22:
@osenft I think this was the most recent set of patches I sent:

https://lists.openwall.net/linux-kernel/2018/07/11/179

They need to be reworked as I note above and sent out again. Hopefully they'll get a bit more traction this way.

shenki, 2021-05-10:
These patches were removed in favour of adding code to the relevant for the controls that were exposed. See 696029eb36c08d7b60097cae44a6269bfa68b51c for an example involving the display driver.