openbmc/linux #182: ftgmac100: mdio read timeout

ytmhub, 2019-07-09:
I'm using linux 4.18 with BCM54616S phy, and I'm getting the following errors when ftgmac100 driver tries to read from mdio bus:
```
    ftgmac100 1e660000.ethernet (unnamed net_device) (uninitialized): mdio read timed out
    ftgmac100 1e660000.ethernet (unnamed net_device) (uninitialized): mdio read timed out
    ...
    ftgmac100 1e660000.ethernet (unnamed net_device) (uninitialized): eth%d: no PHY found
    ftgmac100 1e660000.ethernet: MII Probe failed!
```
This issue seems the go away if  'ftgmac100_setup_clk' is called before 'ftgmac100_setup_mdio'.

I failed to find a mention of this in the ast2500 datasheet, did anyone else encounter this issue?

BTW, only mac0 is enabled in the dts, and the pinctrl for is:
pinctrl-0 = <&pinctrl_rgmii1_default &pinctrl_mdio1_default>;

amboar, 2019-07-11:
Can you try with the dev-5.2 branch instead? Just so we're not debugging issues that are already resolved.

ytmhub, 2019-11-16:
This seems to have been fixed since then, as can be seen [here](https://github.com/openbmc/linux/commit/de501f8417a42f645d41691e466cf9ff46dd7486).

closed 2019-11-16 by ytmhub