openbmc/linux #183: Describe upstream regulators for Witherspoon's BMP280

amboar, 2019-07-23:
Currently the following warnings are issued for Witherspoon's devicetree:

```
arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dt.yaml: bmp280@77: 'vddd-supply' is a required property
arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dt.yaml: bmp280@77: 'vdda-supply' is a required property
```

spinler, 2019-07-23:
FYI, that device only existed on some early lab versions of that system, and didn't ship to the field.

amboar, 2019-07-24:
Can we drop it from the devicetree then?

msbarth, 2019-07-24:
@amboar, I'd prefer that it does not get dropped as that would then cause these lab systems still in use to no longer have an available ambient sensor.

bjwyman, 2019-07-25:
Not that I am a kernel guru or anything, but BMP280 issue caught my eye.

I was curious what those were, and why they would be required.

Is it just me, or did those start out as optional and move to required?
https://patchwork.ozlabs.org/patch/637124/

amboar, 2019-07-30:
@bjwyman hah! Indeed they did, but not in the change you pointed to, but rather this one:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v5.3-rc2&id=88aa7ae661284accd4c058429d6d0a3b7397e081

This was introduced in v5.2. I'll poke people.

bjwyman, 2019-07-30:
Okay, I was just noting that going back to the patch I mentioned, those appeared to be optional, but somehow they later ended up as required. The change you pointed to is obviously where that was introduced.