openbmc/linux #184: Oops when removing ftgmac100

shenki, 2019-09-06:
While debugging some NCSI issues I tried to remove the ftgmac100 driver, which blew up.

```
/sys/bus/platform/drivers/ftgmac100# echo 1e670000.ftgma
c > unbind
[  853.453236] ------------[ cut here ]------------
[  853.458475] kernel BUG at net/core/dev.c:8990!
[  853.463438] Internal error: Oops - BUG: 0 [#1] SMP ARM
[  853.469185] Modules linked in:
[  853.472620] CPU: 0 PID: 123 Comm: sh Not tainted 5.2.9-c008a29-dirty-a07bd94 #1
[  853.480776] Hardware name: Generic DT based system
[  853.486164] PC is at netdev_run_todo+0x268/0x2ac
[  853.491327] LR is at cpumask_next+0x20/0x24
[  853.495993] pc : [<8061f054>]    lr : [<8075593c>]    psr: 20030013
[  853.502986] sp : b7915d48  ip : 00000002  fp : b7915d94
[  853.508816] r10: b7915d54  r9 : b7915f60  r8 : 80b45bc0
[  853.514646] r7 : 0000d831  r6 : 80b03d00  r5 : bdbf3000  r4 : bdbf327c
[  853.521932] r3 : bdbf3058  r2 : bdabc474  r1 : 00000002  r0 : 00000000
[  853.529223] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[  853.537187] Control: 10c5387d  Table: b81fc06a  DAC: 00000051
[  853.543616] Process sh (pid: 123, stack limit = 0x081a74c4)
[  853.549838] Stack: (0xb7915d48 to 0xb7916000)
[  853.554708] 5d40:                   809530a8 8095257c 00000000 b7915d54 b7915d54 b71c8280
[  853.563845] 5d60: b7915f60 80b05c88 b7915da4 bdbf3000 bdbf3500 80b39760 00000010 b71c8280
[  853.572979] 5d80: b7915f60 b71c8290 b7915da4 b7915d98 8062f218 8061edf8 b7915dbc b7915da8
[  853.582112] 5da0: 8061e634 8062f20c bdbf3000 bdbf3500 b7915ddc b7915dc0 80519388 8061e614
[  853.591247] 5dc0: bd91be10 bd91be10 80b39760 00000010 b7915df4 b7915de0 80497aec 80519370
[  853.600380] 5de0: bd91be10 bd91b410 b7915e14 b7915df8 80495da0 80497ac4 bd91be10 80b39760
[  853.609514] 5e00: 80b319b8 00000010 b7915e24 b7915e18 80495e9c 80495cbc b7915e44 b7915e28
[  853.618647] 5e20: 804939bc 80495e88 8049394c b9475e80 00000000 00000000 b7915e5c b7915e48
[  853.627782] 5e40: 80492cb4 80493958 80492c84 b9475e80 b7915e74 b7915e60 80312aa4 80492c90
[  853.636915] 5e60: 00000010 b9475e80 b7915eac b7915e78 80311f94 80312a68 00000000 00000000
[  853.646050] 5e80: 8016345c b78546c0 b7915f60 00000010 80311e8c 00000000 00000010 00000004
[  853.655183] 5ea0: b7915f2c b7915eb0 802924b8 80311e98 00000400 00000007 7eb409a4 00000000
[  853.664317] 5ec0: 00000004 00000000 b7914000 80127f2c b7915f3c b7915ee0 80127f2c 80142ff8
[  853.673450] 5ee0: b7914000 00000004 00000007 00000000 00000000 00000000 00000000 80294f4c
[  853.682583] 5f00: bda03300 80b05c88 00000010 b78546c0 011ea6a8 b7915f60 00000000 00000010
[  853.691716] 5f20: b7915f5c b7915f30 80294f8c 80292474 b7915fa4 802b6ad4 b7915f5c b78546c0
[  853.700851] 5f40: b78546c0 00000000 00000000 011ea6a8 b7915f94 b7915f60 8029524c 80294ee4
[  853.709984] 5f60: 00000000 00000000 b7915f94 80b05c88 00000001 000b1bd4 76f94fe0 00000004
[  853.719119] 5f80: 80101204 b7914000 b7915fa4 b7915f98 802952ec 802951e4 00000000 b7915fa8
[  853.728253] 5fa0: 80101000 802952e0 00000001 000b1bd4 00000001 011ea6a8 00000010 00000000
[  853.737386] 5fc0: 00000001 000b1bd4 76f94fe0 00000004 011ea6a8 011ea6a8 011e64c5 0009656c
[  853.746517] 5fe0: 000b1354 7eb40818 00019c14 4d967170 60000010 00000001 00000000 00000000
[  853.755642] Backtrace: 
[  853.758402] [<8061edec>] (netdev_run_todo) from [<8062f218>] (rtnl_unlock+0x18/0x1c)
[  853.767057]  r10:b71c8290 r9:b7915f60 r8:b71c8280 r7:00000010 r6:80b39760 r5:bdbf3500
[  853.775794]  r4:bdbf3000
[  853.778626] [<8062f200>] (rtnl_unlock) from [<8061e634>] (unregister_netdev+0x2c/0x30)
[  853.787481] [<8061e608>] (unregister_netdev) from [<80519388>] (ftgmac100_remove+0x24/0x70)
[  853.796802]  r5:bdbf3500 r4:bdbf3000
[  853.800803] [<80519364>] (ftgmac100_remove) from [<80497aec>] (platform_drv_remove+0x34/0x4c)
[  853.810323]  r7:00000010 r6:80b39760 r5:bd91be10 r4:bd91be10
[  853.816646] [<80497ab8>] (platform_drv_remove) from [<80495da0>] (device_release_driver_internal+0xf0/0x1a8)
[  853.827616]  r5:bd91b410 r4:bd91be10
[  853.831609] [<80495cb0>] (device_release_driver_internal) from [<80495e9c>] (device_driver_detach+0x20/0x24)
[  853.842584]  r7:00000010 r6:80b319b8 r5:80b39760 r4:bd91be10
[  853.848904] [<80495e7c>] (device_driver_detach) from [<804939bc>] (unbind_store+0x70/0xe4)
[  853.858145] [<8049394c>] (unbind_store) from [<80492cb4>] (drv_attr_store+0x30/0x3c)
[  853.866790]  r7:00000000 r6:00000000 r5:b9475e80 r4:8049394c
[  853.873112] [<80492c84>] (drv_attr_store) from [<80312aa4>] (sysfs_kf_write+0x48/0x54)
[  853.881949]  r5:b9475e80 r4:80492c84
[  853.885942] [<80312a5c>] (sysfs_kf_write) from [<80311f94>] (kernfs_fop_write+0x108/0x1ec)
[  853.895165]  r5:b9475e80 r4:00000010
[  853.899165] [<80311e8c>] (kernfs_fop_write) from [<802924b8>] (__vfs_write+0x50/0x1dc)
[  853.908007]  r10:00000004 r9:00000010 r8:00000000 r7:80311e8c r6:00000010 r5:b7915f60
[  853.916744]  r4:b78546c0
[  853.919573] [<80292468>] (__vfs_write) from [<80294f8c>] (vfs_write+0xb4/0x194)
[  853.927736]  r9:00000010 r8:00000000 r7:b7915f60 r6:011ea6a8 r5:b78546c0 r4:00000010
[  853.936387] [<80294ed8>] (vfs_write) from [<8029524c>] (ksys_write+0x74/0xfc)
[  853.944355]  r8:011ea6a8 r7:00000000 r6:00000000 r5:b78546c0 r4:b78546c0
[  853.951839] [<802951d8>] (ksys_write) from [<802952ec>] (sys_write+0x18/0x1c)
[  853.959809]  r9:b7914000 r8:80101204 r7:00000004 r6:76f94fe0 r5:000b1bd4 r4:00000001
[  853.968463] [<802952d4>] (sys_write) from [<80101000>] (ret_fast_syscall+0x0/0x54)
[  853.976912] Exception stack(0xb7915fa8 to 0xb7915ff0)
[  853.982551] 5fa0:                   00000001 000b1bd4 00000001 011ea6a8 00000010 00000000
[  853.991686] 5fc0: 00000001 000b1bd4 76f94fe0 00000004 011ea6a8 011ea6a8 011e64c5 0009656c
[  854.000815] 5fe0: 000b1354 7eb40818 00019c14 4d967170
[  854.006464] Code: ebffd525 eaffffd3 e7f001f2 e7f001f2 (e7f001f2) 
[  854.013276] ---[ end trace 6082dffde6ac5516 ]---
[  854.018437] Kernel panic - not syncing: Fatal exception
[  854.024294] CPU1: stopping
[  854.027330] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D           5.2.9-c008a29-dirty-a07bd94 #1
[  854.037523] Hardware name: Generic DT based system
[  854.042867] Backtrace: 
[  854.045619] [<8010f04c>] (dump_backtrace) from [<8010f2d0>] (show_stack+0x20/0x24)
[  854.054071]  r7:00000001 r6:60060193 r5:00000000 r4:80b4e7cc
[  854.060386] [<8010f2b0>] (show_stack) from [<80755e0c>] (dump_stack+0x8c/0xa0)
[  854.068456] [<80755d80>] (dump_stack) from [<801114d4>] (handle_IPI+0x340/0x374)
[  854.076714]  r7:00000001 r6:00000000 r5:00000004 r4:80b6b538
[  854.083028] [<80111194>] (handle_IPI) from [<80102250>] (gic_handle_irq+0x84/0x88)
[  854.091482]  r10:00000000 r9:bd8f8000 r8:bd8f9f38 r7:c0003000 r6:c0002000 r5:c000200c
[  854.100220]  r4:80b0631c r3:8010a5b0
[  854.104212] [<801021cc>] (gic_handle_irq) from [<80101a8c>] (__irq_svc+0x6c/0x90)
[  854.112563] Exception stack(0xbd8f9f38 to 0xbd8f9f80)
[  854.118201] 9f20:                                                       00000000 000a6464
[  854.127336] 9f40: bdfdeb60 8011ca40 bd8f8000 00000001 80b05ca8 80b05ce0 80b51d2f 808df678
[  854.136468] 9f60: 00000000 bd8f9f94 bd8f9f98 bd8f9f88 8010a5b0 8010a5b4 60060013 ffffffff
[  854.145603]  r9:bd8f8000 r8:80b51d2f r7:bd8f9f6c r6:ffffffff r5:60060013 r4:8010a5b4
[  854.154265] [<8010a56c>] (arch_cpu_idle) from [<8077586c>] (default_idle_call+0x30/0x3c)
[  854.163301] [<8077583c>] (default_idle_call) from [<80154ef4>] (do_idle+0xe8/0x168)
[  854.171852] [<80154e0c>] (do_idle) from [<80155240>] (cpu_startup_entry+0x28/0x2c)
[  854.180306]  r9:410fc075 r8:8000406a r7:80b6b540 r6:10c0387d r5:00000001 r4:00000089
[  854.188956] [<80155218>] (cpu_startup_entry) from [<80110edc>] (secondary_start_kernel+0x164/0x188)
[  854.199059] [<80110d78>] (secondary_start_kernel) from [<801026ec>] (__enable_mmu+0x0/0x14)
[  854.208379]  r5:00000051 r4:bd8ec06a
[  854.212384] ---[ end Kernel panic - not syncing: Fatal exception ]---
```

shenki, 2019-09-06:
If this can be reproduced with an upstream kernel we should report to the upstream list.

amboar, 2019-09-07:
Looks the same as #180

shenki, 2020-10-12:
```
# echo 1e670000.ftgmac > unbind
[   95.813231] ------------[ cut here ]------------
[   95.814015] kernel BUG at net/core/dev.c:9780!
[   95.814593] Internal error: Oops - BUG: 0 [#1] SMP ARM
[   95.815868] Modules linked in:
[   95.817500] CPU: 0 PID: 129 Comm: sh Not tainted 5.8.14-00189-g45b85c8ea9c5 #212
[   95.818331] Hardware name: Generic DT based system
[   95.820333] PC is at netdev_run_todo+0x2ec/0x36c
[   95.820874] LR is at cpumask_next+0x20/0x24
[   95.821310] pc : [<807b07d8>]    lr : [<804ff7b8>]    psr: 20000013
[   95.821945] sp : b4d1ddb8  ip : 00000002  fp : b4d1de04
[   95.822485] r10: 80e03d00  r9 : 00000000  r8 : 00000000
[   95.823041] r7 : 80e06260  r6 : 80e05e14  r5 : 80e05d30  r4 : b4dc6a7c
[   95.823660] r3 : b4dc6854  r2 : bd31c470  r1 : 00000002  r0 : 00000002
[   95.824505] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[   95.825457] Control: 10c5387d  Table: b52cc06a  DAC: 00000051
[   95.826356] Process sh (pid: 129, stack limit = 0x8f069ac2)
[   95.827323] Stack: (0xb4d1ddb8 to 0xb4d1e000)
[   95.828402] dda0:                                                       b4dc6800 80e62a20
[   95.830160] ddc0: b4dc6800 b4d1ddc4 b4d1ddc4 b4d1ddd8 807b02a0 80e05d08 b4d1de2c b4dc6800
[   95.831831] dde0: bd140810 bd11b410 80e527cc bd11b454 bd2a5210 00000051 b4d1de14 b4d1de08
[   95.832960] de00: 807ba8f0 807b04f8 b4d1de2c b4d1de18 807b0344 807ba8e4 b4dc6800 bd140810
[   95.833992] de20: b4d1de4c b4d1de30 8064f5f8 807b0324 bd140810 bd140810 bd11b410 80e527cc
[   95.835040] de40: b4d1de64 b4d1de50 805d4d20 8064f5e4 bd140810 80e527cc b4d1de8c b4d1de68
[   95.836344] de60: 805d1db8 805d4cf8 bd140810 bd11b410 bd140854 80e4c2a8 b525e780 bd2a5210
[   95.837425] de80: b4d1deac b4d1de90 805d3050 805d1c64 bd140810 00000010 80e527cc 80e4c2a8
[   95.838328] dea0: b4d1decc b4d1deb0 805d0458 805d3010 805d03ec 00000000 bd2a5200 b4d1df68
[   95.839472] dec0: b4d1dee4 b4d1ded0 805cf920 805d03f8 805cf8f0 00000000 b4d1defc b4d1dee8
[   95.840429] dee0: 80374390 805cf8fc 00000000 00000000 b4d1df34 b4d1df00 803731d8 80374354
[   95.841216] df00: 00000000 00000000 b4d1dfb0 b527ab40 00000010 803730c0 012f6d78 b4d1df68
[   95.841904] df20: b4d1c000 00000004 b4d1df64 b4d1df38 802d6ccc 803730cc 80129e8c 802fbdc0
[   95.842524] df40: b527ab40 b527ab40 00000000 00000000 80100284 b4d1c000 b4d1df94 b4d1df68
[   95.843245] df60: 802d6f70 802d6c0c 00000000 00000000 80116c38 80e05d08 00000000 00000001
[   95.843992] df80: 012f6d78 00000004 b4d1dfa4 b4d1df98 802d7014 802d6f0c 00000000 b4d1dfa8
[   95.844619] dfa0: 80100080 802d7008 00000000 00000001 00000001 012f6d78 00000010 00000000
[   95.845227] dfc0: 00000000 00000001 012f6d78 00000004 000deb5c 00000020 012f529c 012f52b8
[   95.845717] dfe0: 000de24c 7edd1988 00019954 76f59c6c 60000010 00000001 00000000 00000000
[   95.846215] Backtrace: 
[   95.846590] [<807b04ec>] (netdev_run_todo) from [<807ba8f0>] (rtnl_unlock+0x18/0x1c)
[   95.847100]  r10:00000051 r9:bd2a5210 r8:bd11b454 r7:80e527cc r6:bd11b410 r5:bd140810
[   95.847472]  r4:b4dc6800
[   95.847678] [<807ba8d8>] (rtnl_unlock) from [<807b0344>] (unregister_netdev+0x2c/0x30)
[   95.848051] [<807b0318>] (unregister_netdev) from [<8064f5f8>] (ftgmac100_remove+0x20/0x9c)
[   95.848425]  r5:bd140810 r4:b4dc6800
[   95.848619] [<8064f5d8>] (ftgmac100_remove) from [<805d4d20>] (platform_drv_remove+0x34/0x4c)
[   95.849001]  r7:80e527cc r6:bd11b410 r5:bd140810 r4:bd140810
[   95.849295] [<805d4cec>] (platform_drv_remove) from [<805d1db8>] (__device_release_driver+0x160/0x210)
[   95.849707]  r5:80e527cc r4:bd140810
[   95.849869] [<805d1c58>] (__device_release_driver) from [<805d3050>] (device_driver_detach+0x4c/0x9c)
[   95.850222]  r9:bd2a5210 r8:b525e780 r7:80e4c2a8 r6:bd140854 r5:bd11b410 r4:bd140810
[   95.850524] [<805d3004>] (device_driver_detach) from [<805d0458>] (unbind_store+0x6c/0xe0)
[   95.850841]  r7:80e4c2a8 r6:80e527cc r5:00000010 r4:bd140810
[   95.851077] [<805d03ec>] (unbind_store) from [<805cf920>] (drv_attr_store+0x30/0x3c)
[   95.851377]  r7:b4d1df68 r6:bd2a5200 r5:00000000 r4:805d03ec
[   95.851614] [<805cf8f0>] (drv_attr_store) from [<80374390>] (sysfs_kf_write+0x48/0x54)
[   95.851918]  r5:00000000 r4:805cf8f0
[   95.852076] [<80374348>] (sysfs_kf_write) from [<803731d8>] (kernfs_fop_write+0x118/0x23c)
[   95.852389]  r5:00000000 r4:00000000
[   95.852552] [<803730c0>] (kernfs_fop_write) from [<802d6ccc>] (vfs_write+0xcc/0x1e8)
[   95.852859]  r10:00000004 r9:b4d1c000 r8:b4d1df68 r7:012f6d78 r6:803730c0 r5:00000010
[   95.853157]  r4:b527ab40
[   95.853286] [<802d6c00>] (vfs_write) from [<802d6f70>] (ksys_write+0x70/0xfc)
[   95.853571]  r9:b4d1c000 r8:80100284 r7:00000000 r6:00000000 r5:b527ab40 r4:b527ab40
[   95.853873] [<802d6f00>] (ksys_write) from [<802d7014>] (sys_write+0x18/0x1c)
[   95.854181]  r7:00000004 r6:012f6d78 r5:00000001 r4:00000000
[   95.854390] [<802d6ffc>] (sys_write) from [<80100080>] (ret_fast_syscall+0x0/0x54)
[   95.854680] Exception stack(0xb4d1dfa8 to 0xb4d1dff0)
[   95.854939] dfa0:                   00000000 00000001 00000001 012f6d78 00000010 00000000
[   95.855298] dfc0: 00000000 00000001 012f6d78 00000004 000deb5c 00000020 012f529c 012f52b8
[... 4 lines cut ...]
[   95.857760] CPU1: stopping
[   95.858106] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D           5.8.14-00189-g45b85c8ea9c5 #212
[   95.858500] Hardware name: Generic DT based system
[   95.858760] Backtrace: 
[... cut ...]
[   95.863055]  r9:bd100000 r8:80e70eeb r7:bd101f7c r6:ffffffff r5:60000013 r4:80109440
[   95.863391] [<80109400>] (arch_cpu_idle) from [<8015b48c>] (do_idle+0xdc/0x180)
[   95.863725] [<8015b3b0>] (do_idle) from [<8015b800>] (cpu_startup_entry+0x28/0x2c)
[   95.864069]  r9:410fc075 r8:8000406a r7:80e8f280 r6:10c0387d r5:00000001 r4:0000008f
[   95.864431] [<8015b7d8>] (cpu_startup_entry) from [<8010ff4c>] (secondary_start_kernel+0x148/0x18c)
[   95.864823] [<8010fe04>] (secondary_start_kernel) from [<801018ac>] (__enable_mmu+0x0/0x14)
[   95.865135]  r5:00000051 r4:bd0f406a
```

shenki, 2020-10-12:
A different crash when running in mdio mode:

```
[  237.784574] [<80649e78>] (mdiobus_unregister) from [<8064f644>] (ftgmac100_remove+0x6c/0x94)
[  237.785023]  r7:80e527cc r6:bd11e810 r5:b8da0a80 r4:b8d92000
[  237.785322] [<8064f5d8>] (ftgmac100_remove) from [<805d4d20>] (platform_drv_remove+0x34/0x4c)
[  237.785659]  r5:bd1c2c10 r4:bd1c2c10
[  237.785821] [<805d4cec>] (platform_drv_remove) from [<805d1db8>] (__device_release_driver+0x160/0x210)
[  237.786169]  r5:80e527cc r4:bd1c2c10
[  237.786329] [<805d1c58>] (__device_release_driver) from [<805d3050>] (device_driver_detach+0x4c/0x9c)
[  237.786693]  r9:b8dfc890 r8:b9039cc0 r7:80e4c2a8 r6:bd1c2c54 r5:bd11e810 r4:bd1c2c10
[  237.787004] [<805d3004>] (device_driver_detach) from [<805d0458>] (unbind_store+0x6c/0xe0)
[  237.787321]  r7:80e4c2a8 r6:80e527cc r5:00000010 r4:bd1c2c10
[  237.787566] [<805d03ec>] (unbind_store) from [<805cf920>] (drv_attr_store+0x30/0x3c)
[  237.787813]  r7:b8dd7f68 r6:b8dfc880 r5:00000000 r4:805d03ec
[  237.788008] [<805cf8f0>] (drv_attr_store) from [<80374390>] (sysfs_kf_write+0x48/0x54)
[  237.788274]  r5:00000000 r4:805cf8f0
[  237.788440] [<80374348>] (sysfs_kf_write) from [<803731d8>] (kernfs_fop_write+0x118/0x23c)
```

mdmillerii, 2020-10-12:
So ncsi mode is an ipv4 reference. 

The mdiobus_unregister bug is for not registered or already unregistered.

philenotfound, 2021-03-12:
I had the same issue..
If using NCSI more clean up is required.

```
diff --git a/drivers/net/ethernet/faraday/ftgmac100.c b/drivers/net/ethernet/faraday/ftgmac100.c
index b62da80de753..d77600abca9d 100644
--- a/drivers/net/ethernet/faraday/ftgmac100.c
+++ b/drivers/net/ethernet/faraday/ftgmac100.c
@@ -1995,6 +1995,10 @@ static int ftgmac100_remove(struct platform_device *pdev)
        netdev = platform_get_drvdata(pdev);
        priv = netdev_priv(netdev);

+       /* NCSI still registered will result in kernel panic when unbinding */
+       if (priv->ndev)
+               ncsi_unregister_dev(priv->ndev);
+
        unregister_netdev(netdev);

        if (priv->rclk)
```

shenki, 2021-03-12:
@philenotfound This issue is fixed in mainline linux v5.10, and therefore in the dev-5.10 openbmc branch.

closed 2021-03-12 by shenki