openbmc/linux #193: ast2400 w83795 no effect

wws66, 2020-05-21:
Hello everyone, here I add a w83795adg and two temperature sensors on it, but they have no effect.I checked the experience of other people in the list and found that my / sys / class / hwmon did not generate files similar to hwmon10. I found the following problems through udevadm, but I don't know why he made mistakes. Please help me to see why.

~~~
root@palmetto:~#  udevadm test /sys/class/hwmon/hwmon0
This program is for debugging only, it does not run any program
specified by a RUN key. It may show incorrect results, because
some values may be different, or not available at a simulation run.

Parsed configuration file /lib/systemd/network/99-default.link
Created link configuration context.
Reading rules file: /lib/udev/rules.d/50-udev-default.rules
Reading rules file: /lib/udev/rules.d/60-block.rules
Reading rules file: /lib/udev/rules.d/60-cdrom_id.rules
Reading rules file: /lib/udev/rules.d/60-drm.rules
Reading rules file: /lib/udev/rules.d/60-evdev.rules
Reading rules file: /lib/udev/rules.d/60-input-id.rules
Reading rules file: /lib/udev/rules.d/60-persistent-alsa.rules
Reading rules file: /lib/udev/rules.d/60-persistent-input.rules
Reading rules file: /lib/udev/rules.d/60-persistent-storage-tape.rules
Reading rules file: /lib/udev/rules.d/60-persistent-storage.rules
Reading rules file: /lib/udev/rules.d/60-persistent-v4l.rules
Reading rules file: /lib/udev/rules.d/60-sensor.rules
Reading rules file: /lib/udev/rules.d/60-serial.rules
Reading rules file: /lib/udev/rules.d/61-aspeed-vuart.rules
Reading rules file: /lib/udev/rules.d/64-btrfs.rules
Reading rules file: /lib/udev/rules.d/70-hwmon.rules
Reading rules file: /lib/udev/rules.d/70-iio.rules
Reading rules file: /lib/udev/rules.d/70-joystick.rules
Reading rules file: /lib/udev/rules.d/70-leds.rules
Reading rules file: /lib/udev/rules.d/70-mouse.rules
Reading rules file: /lib/udev/rules.d/70-touchpad.rules
Reading rules file: /lib/udev/rules.d/75-net-description.rules
Reading rules file: /lib/udev/rules.d/75-probe_mtd.rules
Reading rules file: /lib/udev/rules.d/76-aspeed-mtd-partitions.rules
Reading rules file: /lib/udev/rules.d/78-sound-card.rules
Reading rules file: /lib/udev/rules.d/80-drivers.rules
/lib/udev/rules.d/80-drivers.rules:5: RUN{builtin}: 'kmod load $env{MODALIAS}' unknown, ignoring
/lib/udev/rules.d/80-drivers.rules:6: RUN{builtin}: 'kmod load tifm_sd' unknown, ignoring
/lib/udev/rules.d/80-drivers.rules:7: RUN{builtin}: 'kmod load tifm_ms' unknown, ignoring
/lib/udev/rules.d/80-drivers.rules:8: RUN{builtin}: 'kmod load ms_block mspro_block' unknown, ignoring
/lib/udev/rules.d/80-drivers.rules:9: RUN{builtin}: 'kmod load i2o_block' unknown, ignoring
/lib/udev/rules.d/80-drivers.rules:10: RUN{builtin}: 'kmod load ppdev' unknown, ignoring
/lib/udev/rules.d/80-drivers.rules:11: RUN{builtin}: 'kmod load sm_ftl' unknown, ignoring
Reading rules file: /lib/udev/rules.d/80-net-setup-link.rules
Reading rules file: /lib/udev/rules.d/99-aspeed-lpc-ctrl.rules
Reading rules file: /lib/udev/rules.d/99-systemd.rules
Reading rules file: /etc/udev/rules.d/touchscreen.rules
Rules contain 24576 bytes tokens (2048 * 12 bytes), 9735 bytes strings
1349 strings (17099 bytes), 880 de-duplicated (7834 bytes), 470 trie nodes used
Failed to open device '/sys/class/hwmon/hwmon0': No such device
Unloaded link configuration context.
~~~

mdmillerii, 2020-05-21:
No device driver found the desired hardware.    Look at the kernel log for error message.  Look under `/sys/bus/` by connection type if your device is not present.  Check the device tree for the description of your device.

wws66, 2020-05-22:
Hello, I checked the kernel log. There is no description of 83795. The watchdog error has always existed. It is not caused by adding sensor operation. You can see w83795adg under / sys / bus. I checked the related problems, for example: [#3519](https://github.com/openbmc/openbmc/issues/3519) I added the description of w83795adg in the device tree, and added the related .conf and .Bbappend file, but I didn't generate hwmon related things. Would you like to know whether it is automatically instantiated or what other configuration I need to add to generate it.

kernel log
~~~
root@palmetto:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.3.15-c008a29 (oe-user@oe-host) (gcc version 9.2.0 (GCC)) #1 Tue Apr 21 07:41:48 UTC 2020
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] OF: fdt: Machine model: Palmetto BMC
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 57344
[    0.000000]   Normal zone: 448 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 57344 pages, LIFO batch:15
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 56896
[    0.000000] Kernel command line: console=ttyS4,115200n8 root=/dev/ram rw
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 215988K/229376K available (6144K kernel code, 363K rwdata, 1196K rodata, 1024K init, 143K bss, 13388K reserved, 0K cma-reserved)
[    0.000000] random: get_random_u32 called from cache_alloc_refill+0x3ec/0x8ec with crng_init=0
[    0.000000] ftrace: allocating 18705 entries in 37 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] i2c controller registered, irq 16
[    0.000000] clocksource: FTTMR010-TIMER2: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 39817925974 ns
[    0.000024] sched_clock: 32 bits at 48MHz, resolution 20ns, wraps every 44739242997ns
[    0.000124] Switching to timer-based delay loop, resolution 20ns
[    0.002401] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.002472] pid_max: default: 32768 minimum: 301
[    0.003614] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.003674] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.006279] CPU: Testing write buffer coherency: ok
[    0.008652] Setting up static identity map for 0x40100000 - 0x40100058
[    0.014935] devtmpfs: initialized
[    0.039859] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.039930] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.040361] pinctrl core: initialized pinctrl subsystem
[    0.041756] NET: Registered protocol family 16
[    0.042848] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.104081] clocksource: Switched to clocksource FTTMR010-TIMER2
[    0.177140] NET: Registered protocol family 2
[    0.179083] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.179195] TCP established hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.179295] TCP bind hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.179381] TCP: Hash tables configured (established 2048 bind 2048)
[    0.179667] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.179744] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.180441] NET: Registered protocol family 1
[    0.181531] Unpacking initramfs...
[    2.554143] random: fast init done
[    3.924655] Freeing initrd memory: 1496K
[    3.928423] workingset: timestamp_bits=30 max_order=16 bucket_order=0
[    3.929110] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    3.929155] jffs2: version 2.2. (SUMMARY)  漏 2001-2006 Red Hat, Inc.
[    3.930254] NET: Registered protocol family 38
[    3.933235] ipmi-bt-host 1e789140.ibt: Found bt bmc device
[    3.934427] ipmi-bt-host 1e789140.ibt: Using IRQ 30
[    3.942598] Serial: 8250/16550 driver, 6 ports, IRQ sharing enabled
[    3.949555] 1e787000.serial: ttyS5 at MMIO 0x1e787000 (irq = 30, base_baud = 3000000) is a 16550A
[    3.953455] 1e783000.serial: ttyS0 at MMIO 0x1e783000 (irq = 28, base_baud = 1500000) is a 16550A
[    3.955892] 1e784000.serial: ttyS4 at MMIO 0x1e784000 (irq = 29, base_baud = 1500000) is a 16550A
[    4.302484] printk: console [ttyS4] enabled
[    4.310049] timeriomem_rng 1e6e2078.hwrng: 32bits from 0x(ptrval) @ 1us
[    4.323332] random: crng init done
[    4.359639] loop: module loaded
[    4.372541] aspeed-smc 1e620000.spi: Using 50 MHz SPI frequency
[    4.378770] aspeed-smc 1e620000.spi: mx25l25635e (32768 Kbytes)
[... 179 lines cut ...]
~~~

conf file in /etc/default/obmc/
~~~
root@palmetto:~# cat /etc/default/obmc/hwmon/ahb/apb/bus\@1e78a000/i2c-bus\@c0/w83795adg\@2c.conf 
MODE_temp4="label"
LABEL_temp4=aaa
WARNLO_temp4=10000
WARNHI_temp4=40000

MODE_temp2="label"
LABEL_temp2=bbb
WARNLO_temp2=10000
WARNHI_temp2=40000
~~~

/sys/bus
~~~
root@palmetto:~# cat /sys/bus/platform/devices/1e78a0c0.i2c-bus/i2c-2/2-002c/of_node/compatible 
nuvoton,w83795adg
root@palmetto:~# cat /sys/bus/platform/devices/1e78a0c0.i2c-bus/i2c-2/2-002c/of_node/name       
w83795adg
root@palmetto:~# cat /sys/bus/platform/devices/1e78a0c0.i2c-bus/i2c-2/2-002c/of_node/reg 
,
~~~

mdmillerii, 2020-06-03:
Do you have `CONIFG_SENSORS_W83795=y` in your kernel config?
Yocto uses small cfg snippets to enable drivers that exist on your hardware.

wws66, 2020-06-03:
Hello, I wrote the configuration in the following file.

/meta-ibm/meta-palmetto/recipes-kernel/linux/linux-aspeed/palmetto.cfg
~~~
CONFIG_SENSORS_NCT7904=y
CONFIG_EEPROM_AT24=y
CONFIG_SENSORS_TMP421=y
CONFIG_SENSORS_W83795=y
~~~

mdmillerii, 2020-06-03:
All I can say is look at sysfs and dmesg again.
You found the device; if the driver is bound you should find a `driver` symlink.  If not look for errors or why the compatible string doesn't match.

229024527, 2020-07-28:
my w83795adg but can't display on my web,/sys/class/hwmon/hwmon0/device can read data

229024527, 2020-07-28:
I also have w83795,  but webui can't display value ,under /sys/class/hwmon/hwmon/device can read value

mdmillerii, 2020-07-28:
https://github.com/openbmc/docs/blob/master/development/add-new-system.md#hwmon-sensors