openbmc/linux #198: WARNING: CPU: 0 PID: 142 at kernel/dma/mapping.c:148 dma_map_page_attrs with v5.9

shenki, 2020-11-20:
dev-5.9 (wip) on witherspoon:

```
[   53.724209] ------------[ cut here ]------------
[   53.728913] WARNING: CPU: 0 PID: 142 at kernel/dma/mapping.c:148 dma_map_page_attrs+0x17c/0x228
[   53.737616] Modules linked in:
[   53.740696] CPU: 0 PID: 142 Comm: systemctl Not tainted 5.9.9-00137-g6adb2e629a5b #274
[   53.748602] Hardware name: Generic DT based system
[   53.753385] Backtrace: 
[   53.755873] [<8091a910>] (dump_backtrace) from [<8091abb4>] (show_stack+0x20/0x24)
[   53.763451]  r7:00000009 r6:60000193 r5:00000000 r4:80e40efc
[   53.769132] [<8091ab94>] (show_stack) from [<8091fc8c>] (dump_stack+0xc8/0xe4)
[   53.776390] [<8091fbc4>] (dump_stack) from [<80124f04>] (__warn+0xd0/0x148)
[   53.783356]  r7:00000009 r6:00000094 r5:8019aeb4 r4:80b1da6c
[   53.789037] [<80124e34>] (__warn) from [<8091b370>] (warn_slowpath_fmt+0x6c/0xe4)
[   53.796519]  r7:8019aeb4 r6:00000094 r5:80b1da6c r4:00000000
[   53.802195] [<8091b308>] (warn_slowpath_fmt) from [<8019aeb4>] (dma_map_page_attrs+0x17c/0x228)
[   53.810900]  r8:00000002 r7:9ee6c9a0 r6:00000800 r5:00000008 r4:9e55c400
[   53.817640] [<8019ad38>] (dma_map_page_attrs) from [<8069243c>] (usb_gadget_map_request_by_dev.part.0+0xe8/0x1d8)
[   53.827904]  r8:00000008 r7:000139cd r6:9e55c400 r5:00000000 r4:939c4180
[   53.834626] [<80692354>] (usb_gadget_map_request_by_dev.part.0) from [<80692584>] (usb_gadget_map_request+0x28/0x34)
[   53.845153]  r9:8069e868 r8:939c4180 r7:93960000 r6:95e38e40 r5:95e4a0e4 r4:939c4180
[   53.852908] [<8069255c>] (usb_gadget_map_request) from [<806958c0>] (ast_vhub_epn_queue+0xa4/0x208)
[   53.861963] [<8069581c>] (ast_vhub_epn_queue) from [<8069151c>] (usb_ep_queue+0x48/0x110)
[   53.870146]  r10:951f6268 r9:8069e868 r8:939c4180 r7:93960000 r6:00000001 r5:939c4180
[   53.877971]  r4:95e4a0e4
[   53.880527] [<806914d4>] (usb_ep_queue) from [<8069ed18>] (hidg_set_alt+0x11c/0x2ac)
[   53.888276]  r9:8069e868 r8:938fab00 r7:93960000 r6:00000001 r5:939c4180 r4:93960084
[   53.896032] [<8069ebfc>] (hidg_set_alt) from [<8068ba3c>] (composite_setup+0x7a0/0x1cc4)
[   53.904129]  r10:939600e4 r9:951f6268 r8:95068cdc r7:93960084 r6:00000000 r5:95e41060
[   53.911954]  r4:8069ebfc r3:00000004
[   53.915549] [<8068b29c>] (composite_setup) from [<8068e930>] (configfs_composite_setup+0x74/0x8c)
[   53.924425]  r10:9528e000 r9:00000000 r8:951f62dc r7:a0000193 r6:9528fdbc r5:95e41060
[   53.932249]  r4:951f6268
[   53.934803] [<8068e8bc>] (configfs_composite_setup) from [<80694980>] (ast_vhub_ep0_handle_setup+0xc8/0x1f8)
[   53.944635]  r9:00000000 r8:00000015 r7:0000000e r6:00000009 r5:00000002 r4:95e41288
[   53.952392] [<806948b8>] (ast_vhub_ep0_handle_setup) from [<80696614>] (ast_vhub_dev_irq+0x58/0x84)
[   53.961433]  r5:00000001 r4:95e41048
[   53.965024] [<806965bc>] (ast_vhub_dev_irq) from [<80693b78>] (ast_vhub_irq+0x148/0x210)
[   53.973117]  r5:00000200 r4:95e38e40
[   53.976724] [<80693a30>] (ast_vhub_irq) from [<8017fab0>] (__handle_irq_event_percpu+0x54/0x224)
[   53.985518]  r10:9528e000 r9:00000000 r8:00000015 r7:00000001 r6:9e563e68 r5:80d6ad58
[   53.993340]  r4:95e2ea80
[   53.995890] [<8017fa5c>] (__handle_irq_event_percpu) from [<8017fd90>] (handle_irq_event+0x74/0xec)
[   54.004939]  r10:00000129 r9:9528e000 r8:9e40ea00 r7:00000001 r6:9e563e68 r5:80d6ad58
[   54.012762]  r4:9e563e00
[   54.015313] [<8017fd1c>] (handle_irq_event) from [<80184ad8>] (handle_level_irq+0xc0/0x1c0)
[   54.023663]  r6:00000000 r5:80d6ad58 r4:9e563e00
[   54.028294] [<80184a18>] (handle_level_irq) from [<8017f2e4>] (__handle_domain_irq+0x84/0xe4)
[   54.036816]  r5:80d6ad58 r4:00000000
[   54.040414] [<8017f260>] (__handle_domain_irq) from [<8010146c>] (avic_handle_irq+0x68/0x70)
[   54.048852]  r9:9528e000 r8:80100284 r7:9528ff24 r6:ffffffff r5:9528fef0 r4:9e483280
[   54.056606] [<80101404>] (avic_handle_irq) from [<80100b6c>] (__irq_svc+0x6c/0x90)
[   54.064176] Exception stack(0x9528fef0 to 0x9528ff38)
[   54.069231] fee0:                                     00000003 7ef4c86c 40000040 00000000
[   54.077408] ff00: 7ef4c898 01b89928 7ef4c864 00000129 80100284 9528e000 00000129 9528ffa4
[   54.085576] ff20: 9528ffa8 9528ff40 80100080 80786570 80000013 ffffffff
[   54.092185]  r5:80000013 r4:80786570
[   54.095785] [<80786560>] (sys_recvmsg) from [<80100080>] (ret_fast_syscall+0x0/0x54)
[   54.103523] Exception stack(0x9528ffa8 to 0x9528fff0)
[   54.108581] ffa0:                   7ef4c898 01b89928 00000003 7ef4c86c 40000040 00000000
[   54.116754] ffc0: 7ef4c898 01b89928 7ef4c864 00000129 7ef4c860 00000001 00000000 7ef4daa4
[   54.124922] ffe0: 470499e0 7ef4c820 46f5c4e8 46c231b4
[   54.129975]  r6:7ef4c864 r5:01b89928 r4:7ef4c898
[   54.134592] ---[ end trace 552986e33242a1ea ]---
[   54.139219]  1e6a0000.usb-vhub:p1: failed to map buffer
[   54.144456] aspeed_vhub 1e6a0000.usb-vhub: Request mapping failure -14
[   54.150986] configfs-gadget gadget: ep2 queue req --> -14
```

mdmillerii, 2020-11-20:
https://github.com/shenki/linux/blob/dev-5.9/kernel/dma/mapping.c#L148

shenki, 2021-02-11:
@geissonator saw this with v5.10.

When attempting to debug this the issue goes away, so there is some aspect of timing or build configuration that affects reproducibility.

shenki, 2021-02-24:
69c9e825e812ec6d663e64ebf14bd3bc7f37e2c7 fixes a similar looking warning. I suspect we're setting this on the wrong device. Reliably reproducing is still difficult which makes it hard to test.

vkrk-user, 2021-03-22:
@shenki  USB-DMA issue observed in 5.10 based kernel

I could able to recreate this issue with AST2600 eval kit i.e.., whenever I connect the USB cable to my host setup (laptop/server) such dma triggers were observed through USB interrupt which in turn lead to a crash.
Also in our setup we've enabled the network over usb (USB_ETH) during our testing. Whereas this issue is not seen when I swithc back to 5.8 kernel. Looks like there seems to be a compatiblitiy issue with Aspeed USB gadget driver on latest kernel.

The errors logs observed in my setup is shown below:

[  228.370578] IPv6: ADDRCONF(NETDEV_CHANGE): usb0: link becomes ready
[  228.377631] ------------[ cut here ]------------
[  228.377655] WARNING: CPU: 0 PID: 14 at kernel/dma/mapping.c:149 dma_map_page_attrs+0x2ac/0x39c
[  228.377666] CPU: 0 PID: 14 Comm: kworker/0:1 Not tainted 5.10.20-74acb73 #1
[  228.377669] Hardware name: Generic DT based system
[  228.377680] Workqueue: events linkwatch_event
[  228.377687] Backtrace:
[  228.377701] [<80976638>] (dump_backtrace) from [<8097688c>] (show_stack+0x20/0x24)
[  228.377708]  r7:00000095 r6:600b0193 r5:00000000 r4:80da2358
[  228.377718] [<8097686c>] (show_stack) from [<80982584>] (dump_stack+0x9c/0xb0)
[  228.377729] [<809824e8>] (dump_stack) from [<80121888>] (__warn+0xf8/0x154)
[  228.377735]  r7:00000095 r6:00000009 r5:80199d3c r4:80b21528
[  228.377744] [<80121790>] (__warn) from [<80977014>] (warn_slowpath_fmt+0x8c/0xc0)
[  228.377750]  r7:00000095 r6:80b21528 r5:00000000 r4:81116000
[  228.377759] [<80976f8c>] (warn_slowpath_fmt) from [<80199d3c>] (dma_map_page_attrs+0x2ac/0x39c)
[  228.377768]  r9:00000001 r8:00000200 r7:80a01764 r6:00000000 r5:81116000 r4:811cf400
[  228.377780] [<80199a90>] (dma_map_page_attrs) from [<806d0560>] (usb_gadget_map_request_by_dev.part.0+0xe4/0x1d0)
[  228.377788]  r10:80d05d10 r9:00000001 r8:00000008 r7:00005540 r6:811cf400 r5:00000000
[  228.377792]  r4:8553d900
[  228.377802] [<806d047c>] (usb_gadget_map_request_by_dev.part.0) from [<806d1ac0>] (usb_gadget_map_request+0x28/0x34)
[  228.377810]  r9:81116000 r8:855406f4 r7:855406f4 r6:811cfa40 r5:856b0188 r4:8553d900
[  228.377820] [<806d1a98>] (usb_gadget_map_request) from [<806d5938>] (ast_vhub_epn_queue+0xa4/0x208)
[  228.377828] [<806d5894>] (ast_vhub_epn_queue) from [<806d09e4>] (usb_ep_queue+0x48/0x100)
[  228.377837]  r10:80d05d10 r9:81116000 r8:855406f4 r7:855406f4 r6:8553d900 r5:85586d7c
[  228.377840]  r4:856b0188
[  228.377850] [<806d099c>] (usb_ep_queue) from [<806dc18c>] (rndis_response_available+0x6c/0xb4)
[  228.377857]  r7:855406f4 r6:81143e80 r5:85586d7c r4:85586cc0
[  228.377866] [<806dc120>] (rndis_response_available) from [<806dd2ac>] (rndis_msg_parser+0x190/0x630)
[  228.377872]  r7:855406f4 r6:00000000 r5:855406c0 r4:8553b000
[  228.377881] [<806dd11c>] (rndis_msg_parser) from [<806dafe0>] (rndis_command_complete+0x28/0x48)
[  228.377888]  r8:00000000 r7:81143bb8 r6:00000000 r5:85582288 r4:81143b80
[  228.377898] [<806dafb8>] (rndis_command_complete) from [<806d1c38>] (usb_gadget_giveback_request+0x30/0x9c)
[  228.377902]  r5:85582288 r4:81143b80
[  228.377913] [<806d1c08>] (usb_gadget_giveback_request) from [<806d3ed8>] (ast_vhub_done+0xbc/0xe4)
[  228.377918]  r5:85582288 r4:81143b80
[  228.377928] [<806d3e1c>] (ast_vhub_done) from [<806d4d94>] (ast_vhub_ep0_handle_ack+0x1a0/0x288)
[  228.377936]  r9:81116000 r8:00000001 r7:00000000 r6:00000018 r5:81143b80 r4:85582288
[  228.377945] [<806d4bf4>] (ast_vhub_ep0_handle_ack) from [<806d665c>] (ast_vhub_dev_irq+0x60/0x7c)
[  228.377953]  r9:81116000 r8:00000200 r7:000002f8 r6:00000001 r5:00000012 r4:85582048
[  228.377962] [<806d65fc>] (ast_vhub_dev_irq) from [<806d3d58>] (ast_vhub_irq+0x110/0x1d4)
[  228.377966]  r5:80000200 r4:811cfa40
[  228.377978] [<806d3c48>] (ast_vhub_irq) from [<8017d830>] (__handle_irq_event_percpu+0x68/0x204)
[  228.377986]  r10:80d05d10 r9:81116000 r8:00000027 r7:81117990 r6:00000000 r5:81116000
[  228.377990]  r4:8567f140 r3:806d3c48
[  228.377999] [<8017d7c8>] (__handle_irq_event_percpu) from [<8017dad8>] (handle_irq_event+0x78/0xe4)
[  228.378008]  r10:80df0fa0 r9:81116000 r8:81043800 r7:811e056c r6:81117bd0 r5:81116000
[  228.378011]  r4:811e0500
[  228.378020] [<8017da60>] (handle_irq_event) from [<80182b4c>] (handle_fasteoi_irq+0xa8/0x218)
[  228.378026]  r7:00000001 r6:81117bd0 r5:80d063f4 r4:811e0500
[  228.378034] [<80182aa4>] (handle_fasteoi_irq) from [<8017cf3c>] (__handle_domain_irq+0x90/0xe4)
[  228.378041]  r7:00000001 r6:81117bd0 r5:00000000 r4:80c764cc
[  228.378050] [<8017ceac>] (__handle_domain_irq) from [<80101348>] (gic_handle_irq+0x7c/0x90)
[  228.378058]  r9:81116000 r8:bf80200c r7:80c75bc8 r6:bf802000 r5:81117a28 r4:80d063f4
[  228.378066] [<801012cc>] (gic_handle_irq) from [<80100b0c>] (__irq_svc+0x6c/0x90)
[  228.378070] Exception stack(0x81117a28 to 0x81117a70)
[  228.378078] 7a20:                   bd7d3dc0 00000000 ffffe406 000009e3 bd7d3dc0 00200000
[  228.378086] 7a40: 81117a90 ffffe3f3 81116000 00000000 80df0fa0 81117ae4 ffffe450 81117a78
[  228.378092] 7a60: 801a11e0 801a113c a00b0113 ffffffff
[  228.378100]  r9:81116000 r8:81116000 r7:81117a5c r6:ffffffff r5:a00b0113 r4:801a113c
[  228.378109] [<801a0fdc>] (__run_timers.part.0) from [<801a139c>] (run_timer_softirq+0x70/0x78)
[  228.378117]  r10:81116000 r9:81116000 r8:00000082 r7:00000100 r6:00000001 r5:00000002
[  228.378121]  r4:80d03d00
[... cut ...]
[  228.378488] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  228.378496]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80146c50
[  228.378499]  r4:810cd8c0
[  228.378503] ---[ end trace 0ed0f73ab1520d45 ]---
[  228.378511]  1e6a0000.usb-vhub:p1: failed to map buffer
[  228.378519] aspeed_vhub 1e6a0000.usb-vhub: Request mapping failure -14
[  233.386352]  1e6a0000.usb-vhub:p2: failed to map buffer
[  233.392210] aspeed_vhub 1e6a0000.usb-vhub: Request mapping failure -14
[  233.399496] configfs-gadget gadget: ep2 queue req --> -14

leiyu-bytedance, 2021-03-24:
We have exactly the same issue on 6687842ee681 on g220a.

zevweiss, 2021-03-25:
Also encountering this on my e3c246d4i port (100% reproducible so far).

tao-ren, 2021-03-25:
I'm facing the similar backtrace when moving to kernel 5.10, and I have applied below 1-line patch in my tree to fix it. So far it works for Facebook AST2400 and AST2500 platforms but I didn't get chance to try it on AST2600.

https://github.com/facebook/openbmc-linux/commit/83153bda736b26a07a376529941906f0a825862e#diff-d44e373511270207f6d3dc1f1081834d0b7cd91ffa086c86a999a3dd753cc744

shenki, 2021-03-25:
Thanks @tao-ren. I think this indicates the kernel is using the wrong device pointer somewhere along the lines. I've been reading through the code trying to work out where this is.

Can someone boot using CONFIG_DMA_API_DEBUG and reproduce the issue?

zevweiss, 2021-03-25:
Full dmesg with `CONFIG_DMA_API_DEBUG=y`:

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.23-6687842-dirty-12a1b95 (oe-user@oe-host) (arm-openbmc-linux-gnueabi-gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.36.1.20210209) #1 Thu Mar 25 05:17:00 UTC 2021
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: ASRock e3c246d4i v1.00
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 16 MiB at 0x9d800000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080000000-0x000000009effffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x000000009effffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000009effffff]
[    0.000000] On node 0 totalpages: 126976
[    0.000000]   Normal zone: 992 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 126976 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 125984
[    0.000000] Kernel command line: console=ttyS4,115200n8 root=/dev/ram rw
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 474636K/507904K available (7168K kernel code, 660K rwdata, 1496K rodata, 1024K init, 279K bss, 16884K reserved, 16384K cma-reserved)
[    0.000000] random: get_random_u32 called from cache_alloc_refill+0x420/0x960 with crng_init=0
[    0.000000] ftrace: allocating 22921 entries in 45 pages
[    0.000000] ftrace: allocated 45 pages with 4 groups
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] i2c controller registered, irq 17
[    0.000000] clocksource: FTTMR010-TIMER2: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 77222644334 ns
[    0.000024] sched_clock: 32 bits at 24MHz, resolution 40ns, wraps every 86767015915ns
[    0.000108] Switching to timer-based delay loop, resolution 40ns
[    0.001257] Calibrating delay loop (skipped), value calculated using timer frequency.. 49.50 BogoMIPS (lpj=247500)
[    0.001331] pid_max: default: 32768 minimum: 301
[    0.002109] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.002175] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.004051] CPU: Testing write buffer coherency: ok
[    0.006259] Setting up static identity map for 0x80100000 - 0x80100038
[    0.007529] ASPEED AST2500 rev A2 (04030303)
[    0.008500] devtmpfs: initialized
[    0.045581] DMA-API: preallocated 65550 debug entries
[    0.045630] DMA-API: debugging enabled by kernel config
[    0.045674] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.045731] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.048992] pinctrl core: initialized pinctrl subsystem
[    0.050509] NET: Registered protocol family 16
[    0.052919] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.055482] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.055533] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.112242] mc: Linux media interface: v0.10
[    0.112456] videodev: Linux video capture interface: v2.00
[    0.116043] clocksource: Switched to clocksource FTTMR010-TIMER2
[    0.171894] NET: Registered protocol family 2
[    0.173216] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.173333] TCP established hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.173447] TCP bind hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.173556] TCP: Hash tables configured (established 4096 bind 4096)
[    0.173838] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.173925] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.174421] NET: Registered protocol family 1
[    0.175343] Unpacking initramfs...
[    1.150251] Freeing initrd memory: 1128K
[    1.152878] workingset: timestamp_bits=30 max_order=17 bucket_order=0
[    1.153513] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    1.153568] jffs2: version 2.2. (SUMMARY)  © 2001-2006 Red Hat, Inc.
[    1.154282] NET: Registered protocol family 38
[... 260 lines cut ...]
[   40.966789]  r9:8533e000 r8:80100224 r7:8533fe54 r6:ffffffff r5:8533fe20 r4:80c04320
[   40.974542] [<801011a8>] (avic_handle_irq) from [<80100acc>] (__irq_svc+0x6c/0x90)
[   40.982112] Exception stack(0x8533fe20 to 0x8533fe68)
[   40.987174] fe20: 0000001c 8533fea0 8533fe9c 00000000 8533fea0 8533fe9c 019472a0 00000176
[   40.995360] fe40: 80100224 8533e000 00004040 8533fe84 8533fe88 8533fe70 805e2bc0 802a509c
[   41.003535] fe60: 80000013 ffffffff
[   41.007027]  r5:80000013 r4:802a509c
[   41.010644] [<805e2ba0>] (sockfd_lookup_light) from [<805e6478>] (__sys_sendmmsg+0x50/0x1a0)
[   41.019085]  r5:7eee679c r4:00000001
[   41.022687] [<805e6428>] (__sys_sendmmsg) from [<805e65ec>] (sys_sendmmsg+0x24/0x2c)
[... cut ...]
[   41.053720] ffa0:                   019730e0 76f8ee30 0000001c 7eee679c 00000001 00004040
[   41.061905] ffc0: 019730e0 76f8ee30 019472a0 00000176 7eee679c 019472b0 00000001 00000000
[   41.070079] ffe0: 00000000 7eee6770 0049caa8 76e49150
[   41.075133] ---[ end trace e1f25ad1cb8de2a1 ]---
[   41.079747] DMA-API: Mapped at:
[   41.082909]  debug_dma_map_page+0x58/0xf4
[   41.086949]  dma_map_page_attrs+0x1c0/0x3e0
[   41.091160]  ftgmac100_alloc_rx_buf.constprop.0+0xb8/0x1d8
[   41.096659]  ftgmac100_init_all+0xd8/0x2e0
[   41.100764]  ftgmac100_reset_task+0x134/0x1d0
```

tao-ren, 2021-03-25:
> Thanks @tao-ren. I think this indicates the kernel is using the wrong device pointer somewhere along the lines. I've been reading through the code trying to work out where this is.

My feeling was: the "dummy" device (one for each port, created in "ast_vhub_init_dev") was not fully initialized (no dma mask). The driver used be working, but it fails after commit f959dcd6ddfd ("dma-direct: Fix potential NULL pointer dereference").

zevweiss, 2021-03-25:
I can't speak to whether or not it's entirely the right fix, but experimentally I've just tested @tao-ren's patch, and with it applied I no longer see the aspeed-vhub WARN() and the openbmc iKVM's K and M components are functional.  (I still get the ftgmac100 `check_unmap()` warning, but I presume that's unrelated.)

shenki, 2021-03-25:
@zevweiss Thanks for testing. The ftgmac100 warning is https://github.com/openbmc/linux/issues/201.

tao-ren, 2021-03-25:
> Inspecting the code, this is our call stack:
> 
> ```
> ast_vhub_epn_queue(struct usb_ep* u_ep, ...)
>   struct ast_vhub_ep *ep = to_ast_ep(u_ep);
> 
>   usb_gadget_map_request(&ep->dev->gadget, ...)
>     usb_gadget_map_request_by_dev(gadget->dev.parent, ...)
>       dma_map_single(dev, ...)
>         dma_map_single_attrs(dev, ...)
>           dma_map_page_attrs(dev, ...)
> ```
> 
> The dma buffer is mapped with
> 
> ```
> drivers/usb/gadget/udc/aspeed-vhub/epn.c:       ep->buf = dma_alloc_coherent(&vhub->pdev->dev,
> ```
> 
> So we need to know, does `&vhub->pdev->dev` == `gadget->dev.parent`?
> 
> Following Tao's thread, the fake dummy device has it's parent set to `vhub->pdev->dev`. In usb_initialize_gadget `gadget->dev.parent = parent;`, which takes me to a dead end!
> 
> @tao-ren Can you send your patch to the upstream mailing list? cc the people in this thread, and those from the dma-direct patch you linked to. If it's not the right thing to do I'm sure we will be told!

I was about to learn more before sending out the patch but didn't get chance to do so (struggling with ast2600 these days). Sure I will send out the patch to upstream list soon.

leiyu-bytedance, 2021-03-25:
> I'm facing the similar backtrace when moving to kernel 5.10, and I have applied below 1-line patch in my tree to fix it. So far it works for Facebook AST2400 and AST2500 platforms but I didn't get chance to try it on AST2600.
> 
> [facebook/openbmc-linux@83153bd#diff-d44e373511270207f6d3dc1f1081834d0b7cd91ffa086c86a999a3dd753cc744](https://github.com/facebook/openbmc-linux/commit/83153bda736b26a07a376529941906f0a825862e#diff-d44e373511270207f6d3dc1f1081834d0b7cd91ffa086c86a999a3dd753cc744)

Tried the patch on g220a and it works great!

tao-ren, 2021-03-26:
FYI I just sent the patch to upstream list. I couldn't cc all the people in the thread because I don't have all the email addresses, but I added openbmc email alias, and you could search "[PATCH] usb: gadget: aspeed: set port_dev dma mask".

Cheers,
Tao

shenki, 2021-05-10:
This has been fixed in mainline with bd4d607044b961cecbf8c4c2f3bb5da4fb156993, and backported to the openbmc dev-5.10 kenrel with bbfc8e2ed94396cb0b5c914324680c47172a9762.

closed 2021-05-10 by shenki