openbmc/linux #202: kasan causes userspace to fail

shenki, 2021-02-25:
v5.12-rc0, tacoma, qemu

```
[   10.707901] Run /init as init process
[   10.731892] process '/bin/busybox' started with executable stack
[   10.745234] ==================================================================
[   10.745796] BUG: KASAN: user-memory-access in __clear_user_memset+0x258/0x3ac
[   10.747260] Write of size 2687 at addr 000de581 by task init/1
[   10.747683] 
[   10.748100] CPU: 0 PID: 1 Comm: init Not tainted 5.11.0-10867-g29c395c77a9a-dirty #24
[   10.748671] Hardware name: Generic DT based system
[   10.749255] Backtrace: 
[   10.749686] [<80bd4e44>] (dump_backtrace) from [<80bd5098>] (show_stack+0x20/0x24)
[   10.750428]  r7:81d50010 r6:80000093 r5:00000000 r4:81399560
[   10.750790] [<80bd5078>] (show_stack) from [<80be621c>] (dump_stack+0xd4/0xf0)
[   10.751316] [<80be6148>] (dump_stack) from [<80389174>] (kasan_report+0x100/0x1c0)
[   10.751832]  r7:8057d618 r6:00000001 r5:00000a7f r4:000de581
[   10.752122] [<80389074>] (kasan_report) from [<80389654>] (kasan_check_range+0xcc/0x1a4)
[   10.752647]  r7:81d5000c r6:00000000 r5:000df000 r4:00000a7f
[   10.752951] [<80389588>] (kasan_check_range) from [<8038a674>] (memset+0x28/0x44)
[   10.753575]  r10:81d53c80 r9:81d53c70 r8:00000000 r7:81d5000c r6:00000000 r5:000de581
[   10.754113]  r4:00000a7f r3:8057d618
[   10.754356] [<8038a64c>] (memset) from [<8057d618>] (__clear_user_memset+0x258/0x3ac)
[   10.755001]  r7:81d5000c r6:00000055 r5:000de581 r4:81d50000
[   10.755336] [<8057d3c0>] (__clear_user_memset) from [<8057dc90>] (arm_clear_user+0x54/0x58)
[   10.756047]  r10:00000001 r9:000deb94 r8:8294bc00 r7:00000051 r6:00000000 r5:000de581
[   10.756604]  r4:00000a7f
[   10.756847] [<8057dc3c>] (arm_clear_user) from [<8044f858>] (padzero+0x88/0xa0)
[   10.757421]  r5:000de581 r4:00000a7f
[   10.757646] [<8044f7d0>] (padzero) from [<8045194c>] (load_elf_binary+0xb80/0x1ccc)
[   10.758178]  r7:8649d400 r6:000de581 r5:00000000 r4:000de581
[   10.758483] [<80450dcc>] (load_elf_binary) from [<803a37a8>] (bprm_execve+0x3bc/0x890)
[   10.759049]  r10:fffffff8 r9:811cf060 r8:8649d400 r7:811bc8e0 r6:8649d414 r5:00000000
[   10.759564]  r4:813ea960
[   10.759784] [<803a33ec>] (bprm_execve) from [<803a5260>] (kernel_execve+0x1a0/0x1dc)
[   10.760474]  r10:8649d434 r9:8649d430 r8:8110d7a0 r7:8110d6a0 r6:81c0ed40 r5:00000000
[   10.760987]  r4:8649d400
[   10.761239] [<803a50c0>] (kernel_execve) from [<80bd49c0>] (run_init_process+0xc4/0xf0)
[   10.761918]  r10:00000000 r9:00000000 r8:00000000 r7:80c00500 r6:813a3b18 r5:8110d7ac
[   10.762440]  r4:80c014a0
[   10.762667] [<80bd48fc>] (run_init_process) from [<80be69e8>] (kernel_init+0x70/0x13c)
[   10.763299]  r7:00000000 r6:00000000 r5:80be6978 r4:8110d6a0
[   10.763698] [<80be6978>] (kernel_init) from [<80100170>] (ret_from_fork+0x14/0x24)
[   10.764371] Exception stack(0x81d53fb0 to 0x81d53ff8)
[   10.765046] 3fa0:                                     00000000 00000000 00000000 00000000
[   10.765854] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   10.766561] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   10.767066]  r5:80be6978 r4:00000000
[   10.767433] ==================================================================
```

amboar, 2021-02-25:
This might help: https://amboar.github.io/notes/2019/12/27/arm-kasan.html

zevweiss, 2024-02-14:
@amboar's patch is now in openbmc/linux (commit ceac10c83b330680cc01ceaaab86cd49f4f30d81), so I think this should be fixed now.

amboar, 2024-02-14:
I agree. Let's re-open if anyone discovers further, related issues.

closed 2024-02-14 by amboar