openbmc/linux #203: Unbinding the aspeed-vuart driver tries to free nonexistent resource

amboar, 2021-03-24:
```
# uname -a
Linux rain104bmc 5.8.17-0b37d6f #1 SMP Mon Mar 8 18:59:13 UTC 2021 armv7l GNU/Linux
# echo 1e788000.serial > unbind
[ 9471.114671] Trying to free nonexistent resource <000000001e788000-000000001e78803f>
```

zevweiss, 2021-05-09:
ftrace shows multiple calls to `__release_region()` from a single unbind:

```
              sh-300     [000] .... 25902.658668: __release_region <-serial8250_release_std_resource
              sh-300     [000] .... 25902.658692: <stack trace>
 => serial8250_release_std_resource
 => serial8250_release_port
 => uart_remove_one_port
 => serial8250_unregister_port
 => aspeed_vuart_remove
 => platform_drv_remove
 => __device_release_driver
 => device_driver_detach
 => unbind_store
 => drv_attr_store
 => sysfs_kf_write
 => kernfs_fop_write
 => vfs_write
 => ksys_write
 => sys_write
 => ret_fast_syscall
 ```

and

```
              sh-300     [000] .... 25902.666500: __release_region <-devm_region_release
              sh-300     [000] .... 25902.666519: <stack trace>
 => devm_region_release
 => release_nodes
 => devres_release_all
 => __device_release_driver
 => device_driver_detach
 => unbind_store
 => drv_attr_store
 => sysfs_kf_write
 => kernfs_fop_write
 => vfs_write
 => ksys_write
 => sys_write
 => ret_fast_syscall
```

So, perhaps some duplication between a manual release and a devres auto-release?

zevweiss, 2021-05-09:
With the following patch applied the warning no longer triggers on unbind, but it seems like a bit of a hack; I'm guessing there's a better way...

```diff
diff --git a/drivers/tty/serial/8250/8250_aspeed_vuart.c b/drivers/tty/serial/8250/8250_aspeed_vuart.c
index c33e02cbde93..2f8cbf5d1010 100644
--- a/drivers/tty/serial/8250/8250_aspeed_vuart.c
+++ b/drivers/tty/serial/8250/8250_aspeed_vuart.c
@@ -502,6 +502,7 @@ static int aspeed_vuart_remove(struct platform_device *pdev)
 
 	del_timer_sync(&vuart->unthrottle_timer);
 	aspeed_vuart_set_enabled(vuart, false);
+	serial8250_get_port(vuart->line)->port.mapbase = 0;
 	serial8250_unregister_port(vuart->line);
 	sysfs_remove_group(&vuart->dev->kobj, &aspeed_vuart_attr_group);
 	clk_disable_unprepare(vuart->clk);
```

zevweiss, 2021-05-09:
A slightly less clunky approach, letting the serial8250 code take care of the ioremap instead; still not sure if this is entirely "right" though.

```diff
diff --git a/drivers/tty/serial/8250/8250_aspeed_vuart.c b/drivers/tty/serial/8250/8250_aspeed_vuart.c
index c33e02cbde93..b16dd5f58723 100644
--- a/drivers/tty/serial/8250/8250_aspeed_vuart.c
+++ b/drivers/tty/serial/8250/8250_aspeed_vuart.c
@@ -387,13 +387,9 @@ static int aspeed_vuart_probe(struct platform_device *pdev)
 	timer_setup(&vuart->unthrottle_timer, aspeed_vuart_unthrottle_exp, 0);
 
 	res = platform_get_resource(pdev, IORESOURCE_MEM, 0);
-	vuart->regs = devm_ioremap_resource(&pdev->dev, res);
-	if (IS_ERR(vuart->regs))
-		return PTR_ERR(vuart->regs);
 
 	memset(&port, 0, sizeof(port));
 	port.port.private_data = vuart;
-	port.port.membase = vuart->regs;
 	port.port.mapbase = res->start;
 	port.port.mapsize = resource_size(res);
 	port.port.startup = aspeed_vuart_startup;
@@ -450,7 +446,7 @@ static int aspeed_vuart_probe(struct platform_device *pdev)
 	port.port.iotype = UPIO_MEM;
 	port.port.type = PORT_16550A;
 	port.port.uartclk = clk;
-	port.port.flags = UPF_SHARE_IRQ | UPF_BOOT_AUTOCONF
+	port.port.flags = UPF_SHARE_IRQ | UPF_BOOT_AUTOCONF | UPF_IOREMAP
 		| UPF_FIXED_PORT | UPF_FIXED_TYPE | UPF_NO_THRE_TEST;
 
 	if (of_property_read_bool(np, "no-loopback-test"))
@@ -467,6 +463,7 @@ static int aspeed_vuart_probe(struct platform_device *pdev)
 		goto err_clk_disable;
 
 	vuart->line = rc;
+	vuart->regs = serial8250_get_port(rc)->port.membase;
 
 	rc = of_parse_phandle_with_fixed_args(
 		np, "aspeed,sirq-polarity-sense", 2, 0,
```

amboar, 2021-05-09:
I think it might be better to eliminate `vuart->regs` and just use `port.membase` throughout (possibly with a helper for accessing it).

zevweiss, 2021-05-09:
I actually tried doing exactly that as my first fix attempt, but it panicked on boot for reasons that weren't immediately obvious to me -- looking back at it now, it looks like that was probably because `vuart->port` is only ever initialized in the throttling path and I had been blindly assuming it was set to something useful in the probe function.  Armed with that insight, lemme give that another shot...

zevweiss, 2021-05-10:
...alright, with the `vuart-port` initialization sorted out, that now works as expected -- guess I'll send that to the list shortly then.

amboar, 2021-05-10:
Thanks @zevweiss !

zevweiss, 2021-10-14:
This should be fixed as of upstream commit 54da3e381c2b55289b220601f403f17df7b20597.

shenki, 2021-12-01:
This hit mainline in v5.14, so it's part of dev-5.15. Thanks Zev.

closed 2021-12-01 by shenki