openbmc/linux #204: Nbd_X Unexpected reply issue.

edineshkumar, 2021-05-10:
Two media instances has been redirected by two different nbd (nbd0 and nb2)device. It can be visible in host by issuing the `lsblk` linux cmd. 

When rebooting the host, nbd2 is getting issue of unexpected replay. So that nbd2 socket is getting closing up normally.
redirected image is getting removed from host.

**Expected result:**
Nbd socket should not close and after host rebooted, redirected media instance should visible in host.

**BMC log:**
kernel: block nbd2: Unexpected reply (80) f2061c20
kernel: block nbd2: shutting down sockets
bmcweb[848]: nbd-proxy: wait: Interrupted system call

Kernel version:
5.4.85

Kindly help me to resolve this issue. Thanks for the advance.

williamspatrick, 2021-05-11:
I don't think the people you @'d are going to provide support here.  Maybe if you think those people would be useful you could send an email to the LKML, but I'm not sure why you think those people are relevant because they are not listed in the MAINTAINERS section for nbd:
```
NETWORK BLOCK DEVICE (NBD)
M:	Josef Bacik <josef@toxicpanda.com>
S:	Maintained
L:	linux-block@vger.kernel.org
L:	nbd@other.debian.org
F:	Documentation/admin-guide/blockdev/nbd.rst
F:	drivers/block/nbd.c
F:	include/trace/events/nbd.h
F:	include/uapi/linux/nbd.h
```

If you are using an OpenBMC kernel, that is quite old at this point, and we generally don't provide support outside the currently supported version.  But, you could send an email to our mailing list as well and see if anyone has ideas.

The code sending this dmesg is:
```
	if (!req || !blk_mq_request_started(req)) {
		dev_err(disk_to_dev(nbd->disk), "Unexpected reply (%d) %p\n",
			tag, req);
		return ERR_PTR(-ENOENT);
	}
```

Have you done any investigation into what is causing it to go down this path?

Based on the symptoms you are describing, I seem to recall that some NC-SI NIC cards go through an odd reset mode when the host is power cycled and the BMC can end up with garbled packets.  I'm basing this on about 4 year old recollection though so I don't know the cause or if there was any resolution.  I could conceive that if the BMC received a garbled packet due to a power cycle of the host then the `nbd` code would get upset like this.

edineshkumar, 2021-05-11:
I have been tested one test case, which is working case. It might be useful to the `nbd` experts to found the root cause.

Step :1   Redirected media instance by using /dev/nbd0 and it could be able to see in host.
step :2    Reboot the host, Media instance is still present in host.
Step :3   Redirected one more instance by using /dev/nbd2 and both instances could be visible in host.
Step :4   Rebooted the host, both media instances were in host.

These are my expected behaviour.

Instead of redirecting one after one, if I redirected both image at a time and then reboot the host, it will be the issue.
 
please suggest me the inputs.

mdmillerii, 2021-05-11:
Can you recreate with just fio or similar to two nbd devices ?  
What are you using for nbd server?   (Guessing two processes of some code).
 Trying to reduce the environment required to recreate, unless you can find an older kernel that hits it.

closed 2021-06-11 by edineshkumar