openbmc/linux #212: Query on intel PECI subsytem driver being removed

hemanthkumarm-ami, 2021-11-16:
can i know the reason why intel PECI subsytem is removed after kernel dev5.8 ? is there any plan to support this in future releases.

@yoojae

zevweiss, 2021-11-16:
The PECI patches had been living in the OpenBMC kernel tree (and not in the mainline kernel) for a long time and the efforts to get them merged upstream seemed to have stalled, so they were removed from the OpenBMC kernel (see [OpenBMC kernel docs](https://github.com/openbmc/docs/blob/master/kernel-development.md) and [Joel's email](https://lore.kernel.org/all/CACPK8XcCVmibRzOvrca87pzWzg0K5=ZBMeFQE7xorcLN8gxQdQ@mail.gmail.com/#r)).

However, there's recently been some renewed activity on that from (see https://lore.kernel.org/openbmc/20211115182552.3830849-1-iwona.winiarska@intel.com/T/#t), so hopefully we'll see a PECI subsystem merged in the mainline kernel (and OpenBMC's as well) in the not-too-distant future (though note that the new version is behaviorally a bit different than the old one and hence the OpenBMC userspace may require some adaptation to make use of it).

shenki, 2021-11-23:
Thanks for summarising Zev. We would like to see Intel support their hardware in the mainline tree.

Please take any follow up discussion to the mailing list.

closed 2021-11-23 by shenki