openbmc/linux #216: WARN in ncsi-cmd code

shenki, 2022-11-15:
next-20221114 + openbmc patches

```
[   44.627332] ------------[ cut here ]------------
[   44.632657] WARNING: CPU: 0 PID: 508 at net/ncsi/ncsi-cmd.c:231 ncsi_cmd_handler_oem+0xbc/0xd0
[   44.642387] memcpy: detected field-spanning write (size 7) of single field "&cmd->mfr_id" at net/ncsi/ncsi-cmd.c:231 (size 4)
[   44.655131] CPU: 0 PID: 508 Comm: ncsi-netlink Not tainted 6.1.0-rc5-14066-gefbdad8553d8 #17
[   44.664577] Hardware name: Generic DT based system
[   44.664599]  unwind_backtrace from show_stack+0x18/0x1c
[   44.675801]  show_stack from dump_stack_lvl+0x40/0x4c
[   44.681458]  dump_stack_lvl from __warn+0xb8/0x12c
[   44.686814]  __warn from warn_slowpath_fmt+0x9c/0xd8
[   44.692370]  warn_slowpath_fmt from ncsi_cmd_handler_oem+0xbc/0xd0
[   44.699285]  ncsi_cmd_handler_oem from ncsi_xmit_cmd+0x160/0x29c
[   44.706002]  ncsi_xmit_cmd from ncsi_send_cmd_nl+0x13c/0x1dc
[   44.712337]  ncsi_send_cmd_nl from genl_rcv_msg+0x1d0/0x440
[   44.718579]  genl_rcv_msg from netlink_rcv_skb+0xc0/0x120
[   44.724623]  netlink_rcv_skb from genl_rcv+0x28/0x3c
[   44.730182]  genl_rcv from netlink_unicast+0x208/0x370
[   44.735934]  netlink_unicast from netlink_sendmsg+0x1e4/0x450
[   44.742365]  netlink_sendmsg from ____sys_sendmsg+0x23c/0x2b8
[   44.748799]  ____sys_sendmsg from ___sys_sendmsg+0x9c/0xd0
[   44.754941]  ___sys_sendmsg from sys_sendmsg+0x78/0xbc
[   44.760695]  sys_sendmsg from ret_fast_syscall+0x0/0x54
[   44.766544] Exception stack(0xb57b1fa8 to 0xb57b1ff0)
[   44.772191] 1fa0:                   0244f330 0244f1e0 00000003 7ee36a60 00000000 00000000
[   44.781328] 1fc0: 0244f330 0244f1e0 76f35c60 00000128 76f91550 0244f387 0244f387 00498e7c
[   44.790462] 1fe0: 76f35d34 7ee36a10 76f1b510 76bba140
[   44.796186] ---[ end trace 0000000000000000 ]---
```

shenki, 2022-11-15:
```c
static int ncsi_cmd_handler_oem(struct sk_buff *skb,
                                struct ncsi_cmd_arg *nca)
{
        struct ncsi_cmd_oem_pkt *cmd;
        unsigned int len;
        int payload;
        /* NC-SI spec DSP_0222_1.2.0, section 8.2.2.2
         * requires payload to be padded with 0 to
         * 32-bit boundary before the checksum field.
         * Ensure the padding bytes are accounted for in
         * skb allocation
         */ 

        payload = ALIGN(nca->payload, 4);
        len = sizeof(struct ncsi_cmd_pkt_hdr) + 4; 
        len += max(payload, padding_bytes);

        cmd = skb_put_zero(skb, len);
        memcpy(&cmd->mfr_id, nca->data, nca->payload);
        ncsi_cmd_build_header(&cmd->cmd.common, nca);

        return 0;
}
```

shenki, 2022-11-17:
https://lore.kernel.org/netdev/CACPK8Xdfi=OJKP0x0D1w87fQeFZ4A2DP2qzGCRcuVbpU-9=4sQ@mail.gmail.com/

mdmillerii, 2023-03-14:
Upstream https://github.com/torvalds/linux/commit/b93884eea26f97a3dc4c1df8c64389cbb0673001